Apollon77/ioBroker.smartmeter #2: this.pause is not a function in SerialResponseTransport.js - Windows Umgebung

knatzkopp, 2017-04-01:
Hallo, 
ich hab mit dem Adapter unter Windows (Server 2016)/ iobroker auf Node.js 4.3.1 (x64) / serialport@4.0.7 Probleme mit dem Adapter (gehabt), sodass ich im Log folgende Fehler erhalten habe:

`terminating
Exception: TypeError: this.pause is not a function
at SerialPort.module.exports.raw (C:\ioBroker\node_modules\iobroker.smartmeter\node_modules\smartmeter-obis\node_modules\serialport\lib\parsers.js:7:13)
at SerialPort.emit (events.js:169:7)
at emitOne (events.js:77:13)
at SerialPort. (C:\ioBroker\node_modules\iobroker.smartmeter\node_modules\smartmeter-obis\lib\transports\SerialResponseTransport.js:53:18)
TypeError: this.pause is not a function
uncaught exception: this.pause is not a function
starting. Version 0.5.0 in C:/ioBroker/node_modules/iobroker.smartmeter, node: v4.3.1`

Ein ausklammern der Zeile bzw. ein Code-Schnippel hat kurzhand ausgeholfen:
`if (process.platform !== 'win32'){
				this.pause();
			}`

Vielleicht ein Problem nur bei mir - melde es trotzdem einmal :-)
Ein klasse Adapter, danke!! 

Gruß
Jakob

Apollon77, 2017-04-02:
Welche Adapter-Version benutzt Du? Bitte eueste von Github (bzw 0.5.6 aus latest repo) versuchen

knatzkopp, 2017-04-04:
Ist ein Argument... hab das Update erfolgreich über mein Update-Script nicht bekommen und im git nicht noch einmal genau hingeschaut.

Jetzt neu auf 0.5.7 -> aber same failure:
`terminating
Exception: TypeError: self.serialComm.pause is not a function
at SerialPort.module.exports.raw (C:\ioBroker\node_modules\iobroker.smartmeter\node_modules\smartmeter-obis\node_modules\serialport\lib\parsers.js:7:13)
at SerialPort.emit (events.js:169:7)
at emitOne (events.js:77:13)
at SerialPort. (C:\ioBroker\node_modules\iobroker.smartmeter\node_modules\smartmeter-obis\lib\transports\SerialResponseTransport.js:60:29)
TypeError: self.serialComm.pause is not a function
uncaught exception: self.serialComm.pause is not a function
starting. Version 0.5.7 in C:/ioBroker/node_modules/iobroker.smartmeter, node: v4.3.1`

Die Zeile mit `self.serialComm.pause();` ausgeklammert, läuft der Adapter hoch und rennt einwandfrei...

Gruß
Jakob

Apollon77, 2017-04-04:
Ja, Google sagt das gleiche :-((

Bitte Github Version versuchen ... 0.5.8

knatzkopp, 2017-04-05:
**0.5.8 läuft, top!! Danke!**

Kleiner Hinweis, was aber eher meiner Windows-Umgebung geschuldet ist... die Installation des Adapters über iobroker > github-url lief in einen Fehler, wo der Befehl "git" (obwohl vorhanden) nicht ausgeführt werden kann - hab den selben npm Befehl direkt auf dem Host ausgeführt und alles jut. 

Freilich läuft das Ganze unterm Raspberry geschmeidiger, aber jut... warum einfach wenn auch schwer geht :-)

Apollon77, 2017-04-06:
Alles klar, dann mache ich die 0.5.8. mal offiziell. 

Wegen git ... keine Ahnung, an sich sollte npm das alles selbst erledigen :-(
Das das überhaupt so ist das ich von Git installieren muss liegt nur daran das der Owner einer Library meinen Oatch nicht akzeptiert und auch nicht mal kommentiert ... wenn das wäre könnte man die offizielle Version nutzen und alles wäre super :-( Naja, muss weiter treten

closed 2017-04-06 by Apollon77