Apollon77/ioBroker.smartmeter #10: communication  with secure 3 phase meter

dhirajthaokar, 2018-03-27:
how to communicate secure apex meter.
i am using smartmeter-obis (npm )

my code is
///////////////////////
var SmartmeterObis = require('smartmeter-obis');

var options = {
    'protocol': "D0Protocol",
    'transport': "SerialRequestResponseTransport",
    'transportSerialPort': "/dev/ttyUSB0",
    'transportSerialBaudrate': 300,
    'protocolD0WakeupCharacters': 40,
    'protocolD0DeviceAddress': '',
    'requestInterval': 10,
    'obisNameLanguage': 'en',
    'obisFallbackMedium': 6
};

function displayData(err, obisResult) {
    if (err) {
        console.log(err);
        // handle error
        // if you want to cancel the processing because of this error call smTransport.stop() before returning
        // else processing continues
        return;
    }
    for (var obisId in obisResult) {
        console.log(
            obisResult[obisId].idToString() + ': ' +
            SmartmeterObis.ObisNames.resolveObisName(obisResult[obisId], options.obisNameLanguage).obisName + ' = ' +
            obisResult[obisId].valueToString()
        );
    }

}

var smTransport = SmartmeterObis.init(options, displayData);

smTransport.process();

setTimeout(smTransport.stop, 60000);
/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
getting error: 

 if (this.options.debug === 2) this.options.logger('STOP');
                    ^

TypeError: Cannot read property 'debug' of undefined
    at Timeout.stop [as _onTimeout]
 (node_modules/smartmeter-obis/lib/transports/SerialRequestResponseTransport.js:329:21)
    at ontimeout (timers.js:386:14)
    at tryOnTimeout (timers.js:250:5)
    at Timer.listOnTimeout (timers.js:214:5)

node red code 

[{"id":"df634da1.86f38","type":"smartmeter","z":"4fbf46c4.ca6da8","name":"","datasource":"32ad24b8.a54fcc","protocol":"D0Protocol","transport":"SerialRequestResponseTransport","requestInterval":"1","x":230,"y":60,"wires":[["76cc5800.fa78c8"]]},{"id":"32ad24b8.a54fcc","type":"smartmeter-connection","z":"","sourcetype":"serial","serialport":"/dev/ttyUSB0","serialbaud":"300","databits":"7","parity":"none","stopbits":"1","httphost":"localhost","httpport":"80","filepath":"/dev/null"}]

node -red gives fallowing error:
"smartmeter-obis error: Error: No or too long answer from Serial Device after last request."

Apollon77, 2018-03-27:
Try to set "debug: 2" in your options array then we will get more debug

Apollon77, 2018-03-27:
PS: Also a more detailed and exact name of the smart meter would help. "Apex" is a company only ... And it seems that I can not get any info from "non-relevant country" :-)

Apollon77, 2018-03-27:
PPS: your error comes from your line

```
setTimeout(smTransport.stop, 60000);
```

When using objects you need to use a function in your callback:

```
setTimeout(function() {smTransport.stop();}, 60000);
```

dhirajthaokar, 2018-03-27:
when i set debug : 2

o/p: 
CREATE SERIALPORT: 300 7 1 even
SERIALPORT OPEN
SERIALPORT RESET BAUDRATE TO 300
INITIAL MESSAGES TO SEND: 2
CURRENT PROCESS STEP 0 IN GETNEXTMESSAGE
TO SEND 2: 
DONE SEND 2
CURRENT PROCESS STEP 1 IN GETNEXTMESSAGE
TO SEND 1: /?!

DONE SEND 1
DONE SEND 0
SET MESSAGE TIMEOUT TIMER2: 120000
STOP
CLEAR MESSAGE TIMEOUT TIMER
SERIALPORT CLOSE
SERIALPORT REMOVE LISTENERS

reading not found..

Apollon77, 2018-03-27:
Then the basic question is what communication options you need to use to communicate with your smartmeter? What the docs/specs say?

dhirajthaokar, 2018-03-27:
i am using usb optical probe for read data/file from energy meter.

Apollon77, 2018-03-27:
Great, but you need to know how the energy meter is sending it's data ... only then you know what communication-settings youneed to use.

Please provide the EXACT Name/model ID whatever of the energy meter and/or try to contact your power company to find out tzhe details for communication.

dhirajthaokar, 2018-03-30:
i got manufacture ID from this site.
http://dlms.com/organization/flagmanufacturesids/index.html
my meter is "EHL" - secure 

how to apply in code in program ,
provide me sample code to retrieve data of hole month.

Apollon77, 2018-03-30:
So you found out the manufactorer. But still I have no clue about the concrete model. 
Is it that one? http://securemeters.com/index.php/products/generation/premier-300/ or something different?

I try to help you, but to be honest: you need to find out how to communicate with your power meter at all ... only after that the Library can come into play. 
So sorry I try to support a bit with google when I have time, but it is not as easy as that you will get all Infos from me simply out of the box. Is imply can not do that with the number of different power meters out there!!

dhirajthaokar, 2018-03-30:
i am working on  this meter : http://securemeters.com/index.php/products/generation/apex-100/

Apollon77, 2018-03-30:
So in General when I interpret the information I found ( which was not much) D0 as protocol and also 300baud to start should be correct. 

So It is still the Problem that it seems you are not getting data at all. 

So first step for you is (without the adapter/library) get the optical probe and the power meter to send you data at all. So start using raw  ways depending on your operating system. Google provides many results when it comes to that point. 
One example in german would be https://wiki.volkszaehler.org/hardware/controllers/ir-schreib-lesekopf-usb-ausgang ... when you are able to get data at all then we can go further with the library/adapter

dhirajthaokar, 2018-03-30:
now i GOT 

CREATE SERIALPORT: 300 7 1 even
SERIALPORT OPEN
SERIALPORT RESET BAUDRATE TO 300
INITIAL MESSAGES TO SEND: 2
CURRENT PROCESS STEP 0 IN GETNEXTMESSAGE
TO SEND 2: 
DONE SEND 2
CURRENT PROCESS STEP 1 IN GETNEXTMESSAGE
TO SEND 1: /?SECURE!

DONE SEND 1
DONE SEND 0
SET MESSAGE TIMEOUT TIMER2: 120000
NEW DATA
CURRENT PROCESS STEP 2 IN CHECKMESSAGE
null
MATCH-RESULT SIGNON: "K" -> null
NEW DATA
CURRENT PROCESS STEP 2 IN CHECKMESSAGE
null
MATCH-RESULT SIGNON: "KJ" -> null

dhirajthaokar, 2018-03-30:
what is this :...... 'protocolD0DeviceAddress': 'Bla0',
how to find device adress .

Apollon77, 2018-03-30:
Set that empty. Unless you need to provide the id in order to access the device. Depends on your energy company. Till now I never needed it. 

But you got data. Can you send a complete log with debug?!

Apollon77, 2018-03-30:
The sended „/?!“ is kind of the login, so looks better. Send full log please!

dhirajthaokar, 2018-03-30:
log :+1: 

CREATE SERIALPORT: 300 7 1 even
SERIALPORT OPEN
SERIALPORT RESET BAUDRATE TO 300
INITIAL MESSAGES TO SEND: 2
CURRENT PROCESS STEP 0 IN GETNEXTMESSAGE
TO SEND 2: 
DONE SEND 2
CURRENT PROCESS STEP 1 IN GETNEXTMESSAGE
TO SEND 1: /?!

DONE SEND 1
DONE SEND 0
SET MESSAGE TIMEOUT TIMER2: 120000

Apollon77, 2018-03-30:
Not more then that?

dhirajthaokar, 2018-03-31:
i send you complete log.....

dhirajthaokar, 2018-04-02:
[image]

Apollon77, 2018-04-02:
So it still seems that no dsta er send by your power meter ... so I can not do anything ... Sorry.
In https://github.com/Apollon77/ioBroker.smartmeter/issues/10#issuecomment-377490053 you received something ... There you had set a "Device ID" (SECURE) ... maybe try that again? Or ask your power company for details

dhirajthaokar, 2018-04-02:
plz forward me sample code

Apollon77, 2018-04-02:
In general your question is in the wrong repository! You use the smartmeter-obis Library when I understand correctly, but you opened the question in the Repo of a plugin for ioBroker smart home software.
If this is true and you want to use the library then ignore the adapter! Formally we would need to close here and reopen the topic in smartmeter-obis repository ...

dhirajthaokar, 2018-04-02:
ok :+1:

closed 2018-04-02 by dhirajthaokar