Apollon77/ioBroker.smartmeter #11: No or wrong information displayed after click at "?"

wendy2702, 2018-04-12:
Hi,

as allready shown here:

https://forum.iobroker.net/viewtopic.php?f=23&t=5047&p=141823#p141814

[image]

[image]

No or wrong information is shown for Adapter.

Br

Apollon77, 2018-04-18:
please update to latest version and retest

Apollon77, 2018-04-19:
Do you also updated the adapter on all machines where it is installed?

wendy2702, 2018-04-19:
Adapter is only installed at one Slave which is called: PI-IOBROKER as you can see from attached screenshots above.

wendy2702, 2018-04-19:
Sorry, my mistake. Did not select the correct HOST while installingg Smartmeter Adapter from GIT.

Now i have to figure out how to delete the Adapter from Host only.

closed 2018-04-19 by wendy2702

Apollon77, 2018-04-19:
Commandline 

npm uninstall iobroker.smartmeter