Apollon77/ioBroker.smartmeter #22: wrong ELF class: ELFCLASS32

backons, 2019-02-21:
Hallo,
ich bekomme folgendes im LOG wenn ich den adapter starte. Er startet danach immerwieder neu...
Als Zähler habe ich einen MT175.

host.rock64	2019-02-20 20:41:12.062	info	Restart adapter system.adapter.smartmeter.0 because enabled
host.rock64	2019-02-20 20:41:12.062	error	instance system.adapter.smartmeter.0 terminated with code 1 ()
Caught	2019-02-20 20:41:12.061	error	by controller[0]: at Object.Module._extensions..js (module.js:664:10)
Caught	2019-02-20 20:41:12.061	error	by controller[0]: at Module._compile (module.js:653:30)
Caught	2019-02-20 20:41:12.061	error	by controller[0]: at Object.<anonymous> (/opt/iobroker/node_modules/serialport/lib/bindings/linux.js:2:36)
Caught	2019-02-20 20:41:12.061	error	by controller[0]: at bindings (/opt/iobroker/node_modules/serialport/node_modules/bindings/bindings.js:81:44)
Caught	2019-02-20 20:41:12.061	error	by controller[0]: at require (internal/module.js:11:18)
Caught	2019-02-20 20:41:12.061	error	by controller[0]: at Module.require (module.js:597:17)
Caught	2019-02-20 20:41:12.060	error	by controller[0]: at Function.Module._load (module.js:498:3)
Caught	2019-02-20 20:41:12.060	error	by controller[0]: at tryModuleLoad (module.js:506:12)
Caught	2019-02-20 20:41:12.060	error	by controller[0]: at Module.load (module.js:566:32)
Caught	2019-02-20 20:41:12.060	error	by controller[0]: at Object.Module._extensions..node (module.js:682:18)
Caught	2019-02-20 20:41:12.060	error	by controller[0]: Error: /opt/iobroker/node_modules/serialport/build/Release/serialport.node: wrong ELF class: ELFCLASS32
Caught	2019-02-20 20:41:12.060	error	by controller[0]: ^
Caught	2019-02-20 20:41:12.059	error	by controller[0]: throw e
Caught	2019-02-20 20:41:12.059	error	by controller[0]: /opt/iobroker/node_modules/serialport/node_modules/bindings/bindings.js:88
host.rock64	2019-02-20 20:41:10.699	info	instance system.adapter.smartmeter.0 started with pid 9496

Apollon77, 2019-02-21:
Bitte neu installieren bzw im iobroker Verzeichnis bzw in node_modules/iobroker.smartmeter ein „npm rebuild“ ausführen.

closed 2019-02-21 by Apollon77

lschaeferdryden, 2020-08-28:
Hallo, aktuell habe ich das gleiche Problem. Der ioBroker läuft im Docker Container.
Gibt es da eventuell auch eine Lösung dafür?

Apollon77, 2020-08-28:
Naja siehe letzter Kommentar. Neu installieren, pot npm rebuild, pot /opt/iobroker/node_modules/serialport löschen und dann "npm install --production" in /opt/iobroker/node_modules/iobroker.smartmeter