Apollon77/ioBroker.smartmeter #32: Error while parsing D0 content

overfl0w72, 2019-09-22:
Hallo Zusammen,
ich hab mal eine Frage zu meinem kürzlich in Betrieb genommenen Smartmeter Adapter, dieser spuckt häufig warnings in der Log aus. Soweit ich das beurteilen kann läuft er aber.. Vielleicht kann mir jemand hier helfen?

Auszug aus der LOG:

`

smartmeter.0 | 2019-09-22 20:27:10.016 | warn | invalid
-- | -- | -- | --
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | !
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 0-0:96.8.0*255(003533BF)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:96.5.0*255(001C0104)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:76.7.0*255(000113.80*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:56.7.0*255(000014.83*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:36.7.0*255(000241.91*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:16.7.0*255(000370.54*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:1.8.0*255(000422.20196574*kWh)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:96.1.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:0.0.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn |  
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 0-@P`R+*B*U-L(RLx/EBZ5DD3BZ06ETA_107
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:96.5.0*255(001C0104)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:76.7.0*255(000113.33*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:56.7.0*255(000014.73*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:36.7.0*255(000244.68*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:16.7.0*255(000372.74*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:1.8.0*255(000422.20186281*kWh)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:96.1.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | 1-0:0.0.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | warn | Message
smartmeter.0 | 2019-09-22 20:27:10.016 | info | invalid
smartmeter.0 | 2019-09-22 20:27:10.016 | info | !
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 0-0:96.8.0*255(003533BF)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:96.5.0*255(001C0104)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:76.7.0*255(000113.80*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:56.7.0*255(000014.83*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:36.7.0*255(000241.91*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:16.7.0*255(000370.54*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:1.8.0*255(000422.20196574*kWh)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:96.1.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:0.0.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | info |  
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 0-@P`R+*B*U-L(RLx/EBZ5DD3BZ06ETA_107
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:96.5.0*255(001C0104)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:76.7.0*255(000113.33*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:56.7.0*255(000014.73*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:36.7.0*255(000244.68*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:16.7.0*255(000372.74*W)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:1.8.0*255(000422.20186281*kWh)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:96.1.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | 1-0:0.0.0*255(1EBZ010036XXXX)
smartmeter.0 | 2019-09-22 20:27:10.016 | info | Error: Message
smartmeter.0 | 2019-09-22 20:26:08.253 | info | ERROR CLOSING SERIALPORT
smartmeter.0 | 2019-09-22 20:26:08.252 | info | Received 8 values, 5 updated
smartmeter.0 | 2019-09-22 20:26:08.249 | warn | Error while parsing D0 content: ignore complete part Error: Invalid Obis String "6.8.0*255
smartmeter.0 | 2019-09-22 20:26:08.249 | info | Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String "6.8.0*255

`

(Ich verwende den Weidmann Optokopf.)
Danke im Voraus

overfl0w

Apollon77, 2019-09-22:
Da scheinen immer mal wider Müll Daten anzukommen. Checke mal die positionierung des Kopfes und das Kabel wegen Störeinstrahlungen und sowas. Ansonsten mal Adapter auf Debug log setzen und log senden (aus dem File nicht vom Admin)

closed 2019-11-26 by Apollon77

Quarksack, 2021-09-17:
Hallo Apollon,

leider muss ich den Fehler wieder aufmachen. Ich haben ein Ähnliches Problem.
Der Adapter bombt mir das Log mit Error while parsing D0 content: ignore complete part Error: Invalid Obis immer wieder zu.

Den Sitz des Lesekopfs (auch Wiedmann) habe ich geprüft. Der Sitzt soweit ich das sehen kann korrekt. 

Ausgelesen wird ein Siemens TD3511. Es kommen auch Daten an aber die Fehler machen mich schon stutzig.

Anbei ein Debuglog des Adapters und ein Screenshot der Adaptereinstellung.

Viele Grüße und Danke für deine Mühen.

Quarksack

[log.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/7183342/log.txt)
[settings](https://user-images.githubusercontent.com/90892930/133738805-b8b11ecd-cf5c-413d-88ae-963fc20251d7.JPG)

Apollon77, 2021-09-17:
@Quarksack really? "Wieder auf machen" durch posten in ein fast 2 Jahre altes issue ... kannst froh sein das ich sowas sehe. Bitte neues Issue erstellen - mal abgesehn davon das die Fehlermeldung in dem Issue ne ganz andere ist ... 

Sieht für mich eher (mal wieder) nach Zählern aus die Ungültige OBIS IDs in den Daten liefern diees laut spezifikation nicht gibt.

Quarksack, 2021-09-17:
@Apollon77 : Sorry. Sah für mich wie der gleiche Fehler aus. Mach ich. Danke für die schnelle Antwort.