Apollon77/ioBroker.smartmeter #34: [Holley DTZ541] Error while parsing SML message: Error: Wrong TL-Field 0x65 for Choice!

hacki11, 2019-10-11:
Ich habe einen Holley DTZ541 mit fehlerhaft implementiertem SML.
Man könnte versuchen den Zähler zu unterstützen indem man einen Hack in die verwendete SML Lib einbaut.

Details siehe:
https://github.com/volkszaehler/libsml/issues/75

Workaround für libsml:
https://github.com/volkszaehler/libsml/pull/76

Thread:
https://forum.iobroker.net/topic/25091/holley-dtz541-sml-1-04/7

Scheinbar gefixt in neueren Lieferungen:
https://www.loxforum.com/forum/projektforen/loxberry/plugins/85702-neues-plugin-smartmeter-stromzähler-auslesen?p=206977#post206977

Ich habe meinen Netzbetreiber darauf aufmerksam gemacht, aber wie geahnt interessiert oder versteht das dort hinter dem Kontaktformular niemand.

Gerhard1968, 2019-10-28:
Habe den gleichen Zähler. Mit der Weidmann/Emlog Testsoftware bekomme ich die Werte aber angezeigt (1.8.0/2.8.0 und Momentanleistung).

Apollon77, 2019-10-28:
Vllt aus einer Charge mit gefixter Formware? ;-)

Gerhard1968, 2019-10-30:
Kann sein. Aber mit deinem smartmeter geht's trotzdem leider nicht.Besteht die Aussicht das der holley irgendwann Mal geht ;-)GrußGerhardVon meinem Huawei-Mobiltelefon gesendet-------- Originalnachricht --------Betreff: Re: [Apollon77/ioBroker.smartmeter] [Holley DTZ541] Error while parsing SML message: Error: Wrong TL-Field 0x65 for Choice! (#34)Von: Ingo Fischer An: "Apollon77/ioBroker.smartmeter" Cc: Gerhard1968 ,Comment Vllt aus einer Charge mit gefixter Formware? ;-)

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Apollon77/ioBroker.smartmeter/issues/34?email_source=notifications\u0026email_token=ANSCP25LPQN4RKUITM445TLQQ3E7JA5CNFSM4I74ZR32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECMSHNA#issuecomment-546907060",
"url": "https://github.com/Apollon77/ioBroker.smartmeter/issues/34?email_source=notifications\u0026email_token=ANSCP25LPQN4RKUITM445TLQQ3E7JA5CNFSM4I74ZR32YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOECMSHNA#issuecomment-546907060",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

hacki11, 2019-10-31:
Wenn ich Zeit finde kann ich den Workaround welcher nun in der volkszähler libsml ist und die Sml lib des smartmeters portieren

Apollon77, 2019-10-31:
Das wäre interessant ... machs gern gegen meinen Fork https://github.com/Apollon77/OpenSML ... am ENde muss man es mit dem ENtwickler der Lib dann klären, aber der adapter kann auch meinen Fork nehmen

hacki11, 2019-10-31:
https://github.com/Apollon77/OpenSML/pull/1

hacki11, 2019-10-31:
@Apollon77, kannst du dann das aktualisieren der libs und die releases übernehmen?

Apollon77, 2019-10-31:
Checke es morgen  danke!!!

hacki11, 2019-11-05:
Bist du schon dazugekommen?

Apollon77, 2019-11-05:
Dauert noch paar Tage. Ich bin gerade noch tief in der Finalisierung des js-controller 2.0 ...

hacki11, 2019-11-19:
Falls ich irgendwie unterstützen oder etwas übernehmen soll, gib einfach Bescheid!
Grob würde ich mergen, opensml releasen, opensml in smartmeter einbinden, bei mir Testen und ggf. smartmeter releasen.

Apollon77, 2019-11-19:
Js-Controller 2 hat mich etwas länger gebunden als gedacht. Abe Hat sich gelohnt. 

Generell korrekt, aber ich bin nicht der opensml lib maintainer. Nur contributor. Ich kann also erstmal nur einen pr machen und warten. 

Meistens baue ich ne GitHub Variante vom Adapter das man es testen kann. 

Ich räume gerade alle meine todos auf und dann geht weiter.

hacki11, 2019-11-19:
Klaro, Prioritäten muss man setzen - eins nach dem anderen.

Den PR kann ich ja auch gegen sein Repo stellen, dann fällt das schonmal weg von dir.

Apollon77, 2019-11-19:
Gern!, erwähne mich mit bzw schreib das es mit mir abgestimmt ist ... vllt hilfts ;-)

Apollon77, 2019-11-19:
@hacki11 .... ... Danke ... weisst Du was ich gerade gesehen (und mich erinnert hab) ... ich bin inzwischen doch maintainer und hab sogar npm rechte ;-)) ich warte jetzt nen tag falls der andere OpenSML Dev was sagen will ... dann publishen wir :-)

... zu viele projekte ...

Apollon77, 2019-11-26:
will be fixed in upcoming v 3.0.0

closed 2019-11-26 by Apollon77

Apollon77, 2019-11-26:
v3.0.0 on GitHub ... https://forum.iobroker.net/topic/27116/test-adapter-smartmeter-3-0-x please report back in forum