Apollon77/ioBroker.smartmeter #37: All Datapoints are generated again

chriscrown, 2019-11-29:
After upgrading to 3.0.0 all my datapoints are generated again with double underscore in the ID (e.g. smartmeter.0.1-0:1_7_0_255 gets smartmeter.0.1-0:1_7_0__255).

I'm not really shure, but from viewing the code (I'm not a JS-guy) it could be this line:

https://github.com/Apollon77/ioBroker.smartmeter/blob/f4d9a0e288e192f7d19a724ccb359682c1b141da/smartmeter.js#L295

Greetz,
Chris

Apollon77, 2019-11-30:
Please see changelog and version news in Admin, Info on breaking change in (unreleased 2.0.0, so now in 3.0 relesed). Was needed because of the fact that the *->_ before lead to "not unique" keys and so ewas not really working

closed 2019-11-30 by Apollon77

fah, 2019-12-05:
Hallo @Apollon, is it possible to give a hint on how to solve this? I was recording the values for some time in MariaDB now. I switched ON the recording of the double underscore keys in addition, so I don't lose the new values. 

Do I have to change all scripts to the new double underscore names?
So then I would update the data tables in MariaDB so the old values fit to the new keys.

Is it possible to provide some general instructions for this? Or add some button to do this? :)

Apollon77, 2019-12-05:
Schau mal https://forum.iobroker.net/topic/27116/test-adapter-smartmeter-3-0-x/18

Apollon77, 2019-12-05:
Alternative: use the "alias" feature from sql adapter and enable history for the new datapoints and enter the "old" ID as "alias" in the config