Apollon77/ioBroker.smartmeter #95: SML Parsefehler

Trunks1982, 2020-03-26:
Hallo ich bekomme immer diesen Fehler:

smartmeter.0 | 2020-03-26 12:49:12.732 | debug | (12729) Error: Error while parsing SML message: Error: Wrong TL-Field 0x00 for Choice!: Message: 1b1b1b1b010101017607001603f8c7ff620062007263010176010107001605dded550b0901454d4800005d8352010163652d007
-- | -- | -- | --
smartmeter.0 | 2020-03-26 12:49:12.732 | warn | (12729) Error while parsing SML message: Error: Wrong TL-Field 0x00 for Choice!: Message: 1b1b1b1b010101017607001603f8c7ff620062007263010176010107001605dded550b0901454d4800005d8352010163652d0076070016
smartmeter.0 | 2020-03-26 12:49:12.732 | debug | (12729) Error: Error while parsing SML message: Error: Wrong TL-Field 0x00 for Choice!: Message: 1b1b1b1b010101017607001603f8c7ff620062007263010176010107001605dded550b0901454d4800005d8352010163652d007
smartmeter.0 | 2020-03-26 12:49:12.731 | debug | (12729) MSG IDX 2
smartmeter.0 | 2020-03-26 12:49:12.731 | debug | (12729) PAUSE READING SERIALPORT TO HANDLE MESSAGE
smartmeter.0 | 2020-03-26 12:49:12.731 | debug | (12729) MATCH-RESULT MESSAGE: "001b1b1b1b010101017607001603f8c7ff620062007263010176010107001605dded550b0901454d4800005d8352010163652d007607001603f8c80062006200007263070177010b0901454d4800005d835207010
smartmeter.0 | 2020-03-26 12:49:12.730 | debug | (12729) ADD NEW DATA (262 + NEW 14)
smartmeter.0 | 2020-03-26 12:49:12.090 | debug | (12729) MATCH-RESULT MESSAGE: "001b1b1b1b010101017607001603f8c7ff620062007263010176010107001605dded550b0901454d4800005d8352010163652d007607001603f8c80062006200007263070177010b0901454d4800005d835207010
smartmeter.0 | 2020-03-26 12:49:12.090 | debug | (12729) ADD NEW DATA (0 + NEW 262)
smartmeter.0 | 2020-03-26 12:49:12.076 | debug | (12729) SET MESSAGE TIMEOUT TIMER: 120000
smartmeter.0 | 2020-03-26 12:49:12.075 | debug | (12729) SERIALPORT OPEN
smartmeter.0 | 2020-03-26 12:49:10.123 | debug | (12729) connected set to true
smartmeter.0 | 2020-03-26 12:49:10.074 | debug | (12729) SERIALPORT CLOSE
smartmeter.0 | 2020-03-26 12:49:10.066 | debug | (12729) SCHEDULE NEXT RUN IN 2s
smartmeter.0 | 2020-03-26 12:49:10.065 | debug | (12729) RESUME READING SERIALPORT
smartmeter.0 | 2020-03-26 12:49:10.065 | debug | (12729) SET MESSAGE TIMEOUT TIMER: 120000

Apollon77, 2020-03-26:
mehr details wären super.
1.) Was ist es für ein Zähler?
2.) bitte vollständige log zeile aus dem log auf platte holen. admin schneidet ab
3.) Wirklich immer exakt gleiche meldung? Gleicher Fehler ?

Trunks1982, 2020-03-26:
Ja ist immer gleiche Meldung

Trunks1982, 2020-03-27:
Zweirichtungszähler EMH00 06128466
wo bekomme ich die log file her?

Trunks1982, 2020-03-27:
[iobroker.2020-03-27.log](https://github.com/Apollon77/ioBroker.smartmeter/files/4394749/iobroker.2020-03-27.log)

Apollon77, 2020-03-27:
Im Log sehe ich das es nicht "jedes mal" passiert.Es gibt erfolgreiche Läufe. Damit ist aus Erfahrung die Wahrscheinlichkeit von Kommunikationsstörungen am wahrscheinlichsten. Also bitte
- Sitz des Kopfes prüfen
- Kabel gut genug abgeschirmt? Weit weg vom Stromzähler verlegt?
- ggf mal einen aktiven USB Hub versuchen das die Spannung ggf höher ist

closed 2020-03-27 by Apollon77