Apollon77/ioBroker.smartmeter #116: Landis & Gyr E220 erfolgreich eingebunden

MMeinhardt1, 2020-05-20:
Hi,

ich habe einen Landis & Gyr E220 nach Eingabe des PIN und freischalten der erweiterten Info Schnittstelle erfolgreich in den Smartmeter Adapter einbinden können.
Das ganze funktioniert sehr gut und zuverlässig!

Danke

closed 2020-07-20 by Apollon77

snookerap, 2022-04-02:
Hi, mit welchen Einstellungen im Adapter hast du den E220 zum Laufen gebracht.
Habe eine E320, der sieht identisch wie der E220 aus.
Eventuell läuft der ja auch mit deinen Einstellungen.

Danke im voraus

MMeinhardt1, 2022-04-03:
Hi @snookerap 
hier ein Bild meiner Einstellungen, vielleicht hilft es dir. 

[image]

snookerap, 2022-04-03:
mhh.. danke ...

Leider funzt es damit auch nicht, denke meiner muss freigeschaltet sein.
Gestatte mir nch eine Frage... :-)

Wie hast du den freigeschaltet, muss das vom Netzbetreiber passieren?
Die PIN habe ich, ins Menü komme ich auch rein.

Dank für deine Mühe

MMeinhardt1, 2022-04-03:
Hi,
du musst den PIN eingeben und InF freischalten, dass sind die erweiterten Daten. Lies dir das Manual aus dem oberen Link mal durch, insbesondere ab Seite 27, dort wirst du fündig und dann wird es bestimmt auch klappen.