Apollon77/ioBroker.smartmeter #125: Adapter very talkative

PLCHome, 2020-06-19:
I use 3.1.2, it is possible to set the update and port message to "debug" or "silly".

I have the adapter read the data every second. The log runs fully, so that important messages from other adapters are lost.

```
smartmeter.0 | 2020-06-19 12:41:58.511 | info | (1894) Received 7 values, 3 v
-- | -- | -- | --
smartmeter.0 | 2020-06-19 12:41:57.044 | info | (1894) Received 7 values, 3 updated
smartmeter.0 | 2020-06-19 12:41:55.577 | info | (1894) Received 7 values, 3 updated
smartmeter.0 | 2020-06-19 12:41:54.072 | info | (1894) Received 7 values, 3 updated
smartmeter.0 | 2020-06-19 12:41:53.387 | info | (1894) List of port: [{"path":"/dev/ttyACM0"},{"path":"/dev/ttyUSB2"},{"path":"/dev/ttyUSB1"},{"path":"/dev/ttyACM1"},{"path":"/dev/ttyUSB0"},{"path":"/dev/ttyAMA0"}]
```

Update messages are usually grouped under silli.
I think the port message is a debug message.

Apollon77, 2020-06-19:
Yes, the default loglevel is "Info" and in my understanding this is woth being logged as "info". Ypu can change the instance loglevel to"Warning" to prevent these logs.

For this go to instances -> Expert mode -> change in column "log level"

closed 2020-06-19 by Apollon77

PLCHome, 2020-06-19:
Thank you, the views diverge. But I can live with it.