Apollon77/ioBroker.smartmeter #128: DWS7412.2 Zähler werte unter 650W werden als Minus angezeigt

Jetiman, 2020-07-04:
MoinMoin,

ich habe ein Problem bei einem Zähler, die anderen vorhandenen Zähler funktionieren. Einer wurde aber erneuert durch den DWS 7412.2.
Wenn der Wert am Zähler unter 650W geht werden im Broker Minus werte angezeigt.

Als Beispiel:
Zähler 582W
Broker -94.41W

Als Einstellungen nutze ich:
60S Abfrageintervall
Daten nur Lesen
SML 1.0.3
Baud 9600

Den Rest habe ich Standard gelassen.

Apollon77, 2020-07-07:
Bitte vollständige SML Nachricht (Debug log aktivieren und so eine holen) und idealerweise Logging bereitstellen. Dann muss man sehen. CHance auf nen Firmware Bug im gerät ist snicht schlecht ... wurde auch bei FHEM berichtet (https://forum.fhem.de/index.php/topic,51948.msg834274.html#msg834274)

Apollon77, 2020-07-07:
Und du bist sicher das negativ nicht irgendwie einspeisung oder so bedeutet ?!

Jetiman, 2020-07-08:
Log kommt noch. Einspeisung habe ich nicht.

Jetiman, 2020-07-19:
Ich hoffe damit kann man etwas anfangen.

`smartmeter.0	2020-07-19 20:26:23.757	debug	(26076) SERIALPORT CLOSE
smartmeter.0	2020-07-19 20:26:23.725	debug	(26076) SCHEDULE NEXT RUN IN 30s
smartmeter.0	2020-07-19 20:26:23.723	debug	(26076) RESUME READING SERIALPORT
smartmeter.0	2020-07-19 20:26:23.721	debug	(26076) SET MESSAGE TIMEOUT TIMER: 120000
smartmeter.0	2020-07-19 20:26:23.719	info	(26076) Received 5 values, 2 updated
smartmeter.0	2020-07-19 20:26:23.717	debug	(26076) Set State 1-0:16_7_0__255.value = 5226.49
smartmeter.0	2020-07-19 20:26:23.714	debug	(26076) Set State 1-0:16_7_0__255.rawvalue = undefined
smartmeter.0	2020-07-19 20:26:23.713	debug	(26076) 1-0:16.7.0*255: Momentanwert Gesamtwirkleistung (Total) = 5226.49 W
smartmeter.0	2020-07-19 20:26:23.710	debug	(26076) Data for 1-0:2_8_0__255 unchanged
smartmeter.0	2020-07-19 20:26:23.708	debug	(26076) 1-0:2.8.0*255: Zählerstand 1 Summe Wirkarbeit Abgabe - (Total) = 0 kWh
smartmeter.0	2020-07-19 20:26:23.706	debug	(26076) Set State 1-0:1_8_0__255.value = 16718.478
smartmeter.0	2020-07-19 20:26:23.703	debug	(26076) Set State 1-0:1_8_0__255.rawvalue = undefined
smartmeter.0	2020-07-19 20:26:23.702	debug	(26076) 1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 16718.478 kWh
smartmeter.0	2020-07-19 20:26:23.700	debug	(26076) Data for 1-0:96_1_0__255 unchanged
smartmeter.0	2020-07-19 20:26:23.699	debug	(26076) 1-0:96.1.0*255: Serialnumber = 0a01445a47000282526b
smartmeter.0	2020-07-19 20:26:23.697	debug	(26076) Data for 1-0:96_50_1__1 unchanged
smartmeter.0	2020-07-19 20:26:23.695	debug	(26076) 1-0:96.50.1*1: Wartungseintrag (T1) letztes Jahr = DZG
smartmeter.0	2020-07-19 20:26:23.694	debug	(26076) STORE DATA
smartmeter.0	2020-07-19 20:26:23.691	debug	(26076) SML MESSAGE: START SML-File START SmlMessage Transaction-ID: S� (0a53e00a) Group-No: 0 Abort On Error: 0 Message-Body: SmlPublicOpenResponse Codepage: undefined Client-ID: u
smartmeter.0	2020-07-19 20:26:23.686	debug	(26076) MSG IDX 2
smartmeter.0	2020-07-19 20:26:23.685	debug	(26076) PAUSE READING SERIALPORT TO HANDLE MESSAGE
smartmeter.0	2020-07-19 20:26:23.683	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.681	debug	(26076) ADD NEW DATA (250 + NEW 7)
smartmeter.0	2020-07-19 20:26:23.667	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.665	debug	(26076) ADD NEW DATA (236 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.651	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.649	debug	(26076) ADD NEW DATA (222 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.635	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.633	debug	(26076) ADD NEW DATA (208 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.619	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.617	debug	(26076) ADD NEW DATA (194 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.603	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.601	debug	(26076) ADD NEW DATA (181 + NEW 13)
smartmeter.0	2020-07-19 20:26:23.587	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.585	debug	(26076) ADD NEW DATA (167 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.571	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.569	debug	(26076) ADD NEW DATA (153 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.555	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.553	debug	(26076) ADD NEW DATA (139 + NEW 14)
smartmeter.0	2020-07-19 20:26:23.539	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620
smartmeter.0	2020-07-19 20:26:23.537	debug	(26076) ADD NEW DATA (125 + NEW 14)
[... cut ...]
smartmeter.0	2020-07-19 20:25:52.084	debug	(26076) MATCH-RESULT MESSAGE: "445a47000282526b7262016503ac4832620263b419007605ae52e00a62" -> null
smartmeter.0	2020-07-19 20:25:52.081	debug	(26076) ADD NEW DATA (15 + NEW 14)
smartmeter.0	2020-07-19 20:25:52.068	debug	(26076) MATCH-RESULT MESSAGE: "445a47000282526b7262016503ac48" -> null
smartmeter.0	2020-07-19 20:25:52.065	debug	(26076) ADD NEW DATA (0 + NEW 15)
smartmeter.0	2020-07-19 20:25:52.057	debug	(26076) SET MESSAGE TIMEOUT TIMER: 120000
smartmeter.0	2020-07-19 20:25:52.051	debug	(26076) SERIALPORT OPEN`

Apollon77, 2020-07-19:
Aber das Log zeigt ja einen validen Wert mit 5226.49 W .. oder !? Ich bräuchte ein Log wo es negativ ist bitte

Jetiman, 2020-07-19:
Unten ist einer mit = -59.96 W

Apollon77, 2020-07-19:
Ahhh übersehen :-)

Apollon77, 2020-07-19:
Aber die zeilen sind nicht aus dem Logfile oder? Die zeilen sind wieder abgeschnitten. Sorry, damit leider nicht nutzbar

Jetiman, 2020-07-19:
Wie meist du das mit abgeschnitten?
Was brauchst du genau und wie komme ich an die Daten?

Apollon77, 2020-07-19:
Die Zeilen sind nicht Vollständig. Du hast Sie bestimmt aus dem Admin kopiert oder ?

Bitte aus dem Logfile auf der Platte kopieren: /opt/iobroker/log/...

Apollon77, 2020-07-19:
Ok wenn Sie aus dem logfile kommen dann hat es die beim kopieren über deine Shell abgeschnitten ...

Beispiel: Viele Zeilen haben
smartmeter.0	2020-07-19 20:26:23.683	debug	(26076) MATCH-RESULT MESSAGE: "241b1b1b1b0101010176050a53e00a620062007263010176010102310b0a01445a47000282526b7262016503ac4851620263eb2c0076050b53e00a620062007263070177010b0a01445a47000282526b070100620

... heisst aber das die in wirklichkeit länger sind ... aber das fehlt

Jetiman, 2020-07-20:
Wie schaut es damit aus?

[log.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/4948743/log.txt)

Apollon77, 2020-07-23:
Besser :-)

Apollon77, 2020-07-23:
Schau mal was ich hier gefunden habe ... ich wusste doch das ich das schonmal auseinander genommen habe: https://github.com/Apollon77/smartmeter-obis/issues/75 

Hier ist scheinbar der Zähler Buggy in der Firmware ... kannst Du also bei deinem Energieanbieter reklamieren oder nimmst den adapter und versuchst mal ob bei einem negativen Wert ggf mittels der Umrechnungsformel unter https://github.com/Apollon77/smartmeter-obis/issues/75#issuecomment-581650736 und einem kleinen Javascriptchen fixen kannst.

Apollon77, 2020-07-23:
PS: Wenn das tut kann ich das noch in die Readme aufnehmen

Apollon77, 2020-07-23:
In deinem angehängten Log wäre das:

-22.12W --> 633.24W
-32.06W --> 623.3W

passt das?

Dein ganz oben genanntes -94.41W wäre dann nicht ganz dein Wert (aber kann ja leicht anders sein) sondern 560.95W

Jetiman, 2020-07-24:
Habe noch nicht richtig mit Java gearbeitet. Werde aber versuchen das umzusetzen.

Apollon77, 2020-07-24:
ANsonsten beschreib das Thema im Forum ind er "Skripen" Sektion, da wirst Du garantiert support bekommen :-)

closed 2020-08-02 by Apollon77

toofii, 2023-07-15:
Hey, gibt es hier neue Erkenntnisse? Habe mir auch kürzlich einen IR Kopf besorgt und bekomme bei gleichem Zähler den selben Fehler.

Apollon77, 2023-07-22:
Naja die Lösung und so um es zu beheben steht oben im Thread sonst wäre es nicht geschlossen. Buggy Firmware ist halt Buggy Firmware