Apollon77/ioBroker.smartmeter #144: Smartmeter Werte immer senden (nicht nur bei Update)

zehweh, 2020-08-27:
Hi,

ich baue gerade an folgendem Ablauf:

[iobroker / smartmeter] ---- mqtt ---- > [ node red / daten sammeln, aufbereiten ] --------> [ influxdb ]

Mein Problem ist, dass die Werte nur bei einem Update an node red weitergeschickt werden.

Gibt es eine Möglichkeit, abseits von einer Code-Änderung hier 
https://github.com/Apollon77/ioBroker.smartmeter/blob/master/smartmeter.js#L461-L479
die Daten bei jeder Smartmeter-Abfrage neu zu verschicken?

Vielen Dank

Apollon77, 2020-08-27:
Aktuell müsste man das als Feature einbauen. Bisher ist es "Absicht" weil ggf alle par Sekunden sehr viele Werte aktualisiert werden könnten was in den meisten Fällen unnötig ist.

Warum würdest Du es denn brauchen das zB auch die Seriennummer oder Monatssumme immer neu gesendet wird auch wenn nicht geändert?

zehweh, 2020-08-27:
Danke für die schnelle Antwort.

Ich würde gerne alle X Minuten die aktuellen Zählerwerte nach Influxdb schreiben. Dazu sammle ich die Daten der einzelnen Zähler und schreibe sie gebündelt.

So sollte es aussehen:
```
time                Einspeisung    Bezug 
----                ----           -----
1598528949207181218 1234           5678
1598528949201960536 1235           5678
1598528888751666306 1239           5678
```  

Wenn ich nur die Änderungen via mqtt bekomme, sieht es so aus:
```
time                Einspeisung    Bezug 
----                ----           -----
1598528949207181218 1234           5678
1598528949201960536 1235          
1598528888751666306 1239          
```  

Im Moment behelfe ich mir damit, den Smartmeter-Adapter jede Minute neu zu starten, da dann immer alles gesendet wird.

Mittlerweile habe ich festgestellt, dass das bei den ganzen anderen Adaptern wie z.B. modbus auch genauso läuft und nur Updates geliefert werden (was ja generell auch Sinn macht).
Also werde ich mir vermutlich im node red eine Logik bauen, die mir die Werte via contextStorage zwischen speichert und über nen trigger in die Influxdb schreibt, sofern von mqtt nichts kommt.

zehweh, 2020-08-27:
Sorry,

ich habe das Pferd von der falschen Seite aufgezäumt.
Wie ich eben festgestellt habe, kann man node-red als Adapter im iobroker installieren und von dort aus den iobroker abfragen.
Damit kann ich alles - so wie ich es oben beschrieben habe - umsetzen.

Vielen Dank

Apollon77, 2020-08-28:
ich lass es mal als

closed 2020-08-28 by Apollon77