Apollon77/ioBroker.smartmeter #149: Werte in influxdb

lschaeferdryden, 2020-09-02:
Hallo,
der Adapter läuft korrekt in meiner Instanz und die Werte werden auch regelmäßig aktualisiert.
Jetzt wollte ich einzelne Werte in die InfluxDB speichern. Beim Aufruf der Konfig für einen Wert aktiviere ich die InfluxDB Konfig und speichere die Einstellung.
Leider wird die Konfig allerdings nicht gespeichert und die Daten werden nicht an die InfluxDB geliefert. Beim erneuten Aufruf der Konfig ist Influx wieder deaktiviert.

Ist das Problem bereits bekannt?

Apollon77, 2020-09-02:
Geht bei mir super. Wenn ist das also scheinbar ein Admin issue.
Aber lass mal hier suchen: welcher admin, welcher js-controller?

lschaeferdryden, 2020-09-02:
Admin: 4.1.7
js-controller: aktuell (Wo kann ich die Version lesen?)

Kann es eventuell an den Doppelpunkten im Namen liegen? Im Log sehe ich aktuell keine Fehlermeldung.

Apollon77, 2020-09-02:
welche smartmeter version?
Versuch bitte mal admin downgrade auf 4.1.6

lschaeferdryden, 2020-09-02:
Smartmeter Version: 3.1.3
Downgrade hat leider auch keinen Erfolgt.

lschaeferdryden, 2020-09-02:
Bei anderen Objekten funktioniert die Sache mit der InfluxDB auch ohne Probleme.
Kann es an den Namen der Objekte liegen?

[image]

Apollon77, 2020-09-02:
wie gesagt: bei mir tut das alles und daher liegt es nicht an den State IDs.

controller version unter "hosts" im Admin

Apollon77, 2020-09-02:
wenn du es aktivierst steht irgend ein fehler in der browser konsole?

lschaeferdryden, 2020-09-02:
In der Konsole kommt eine Warnung. Sonst keine weitere Meldung. Kann es eventuell sein, dass die Objekte nicht korrekt angelegt wurden? 

[image]

Apollon77, 2020-09-02:
and you downgradded admin to 4.1.6.? And completely reloaded in browser??

lschaeferdryden, 2020-09-02:
Hab den Adapter noch einmal auf die Version 4.1.6 installiert und den Browser Cache geleert. Jetzt funktioniert es wieder. Scheint also ein Problem mit der neuen Admin Version zu sein.

Danke für die schnelle Unterstützung.

closed 2020-09-02 by lschaeferdryden