Apollon77/ioBroker.smartmeter #152: Three ignored OBIS String

j0scha1, 2020-09-11:
**Describe the bug**  
OBIS String 1-1:30D, 1-1:7D, 1-1:1D are ignored by the adapter.

**To Reproduce**  
Steps to reproduce the behavior:
1. Have an Elster AS1440 (in my case I have two, one for household electricity and one for the heating (heating pump)
  a. Also have a Master/Slave iobroker setup. The slave has two optical devices attached via USB (IR-Schreib-Lesekopf by Volkszaehler).
2. Install adapter as described with D0
3. All values of the electricity meter are shown by smartmeter.
4. Observe the only error regarding the aforementioned OBIS Strings. All other strings are reported correctly.

**Expected behavior**  
The OBIS Strings represent the total energy consumed as per 30 days, 7 days and day.

**Screenshots & Logfiles**  
iobroker log lines in short:

smartmeter.0 | 2020-09-11 10:57:24.008 | info | (2666) Received 256 values, 7 updated
-- | -- | -- | --
smartmeter.0 | 2020-09-11 10:57:23.863 | warn | (2666) Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:30D
smartmeter.0 | 2020-09-11 10:57:23.863 | info | (2666) Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:30D
smartmeter.0 | 2020-09-11 10:57:23.862 | warn | (2666) Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:7D
smartmeter.0 | 2020-09-11 10:57:23.861 | info | (2666) Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:7D
smartmeter.0 | 2020-09-11 10:57:23.860 | warn | (2666) Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:1D
smartmeter.0 | 2020-09-11 10:57:23.859 | info | (2666) Error: Error while parsing D0 content: ignore complete part Error: Invalid Obis String 1-1:1D

**Versions:**  

On both master and slave
 - Adapter version: 3.1.2
 - JS-Controller version: 3.1.6
 - Node version: 12.8.3
 - Operating system: Linux iobroker 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux
[iobroker.2020-09-11.log](https://github.com/Apollon77/ioBroker.smartmeter/files/5207890/iobroker.2020-09-11.log)

**Additional context**  
I tried to reinstall the adapter but the log shows the same outcome. 
Is it possible to just ignore certain OBIS strings, so that no errors are shown?

Apollon77, 2020-09-11:
Lets say it that way ... these OBIS strings are "out of specs" :-( I need to see what I can do

j0scha1, 2020-09-13:
Thanks for your replay and your awesome work. Looking forward to a solution, however it is just really a minor issue.

pjo12345, 2022-12-27:
Habe das Problem auch. Wird es Abhilfe geben?

Apollon77, 2022-12-27:
Am liebsten würde ich sagen "zwingt doch mal die Hersteller endlich Spec-Konform zu arbeiten und nicht immer wieder Bugs oder unspezifizierte Erweiterungen einzubauen", aber tja ... die würden sich nur zurücklehnen und sagen "mir doch egal" ... Mal schauen wie ich das ausbaden werde