Apollon77/ioBroker.smartmeter #155: TCP Input delays data. Delay increases over time

klassisch, 2020-09-20:
**Describe the bug**  
for more details, please refer https://forum.iobroker.net/topic/36851/smartmeter-buffering-dateneingang-via-tcp-zeitdilatation 

The Itron OpenWay 3.HZ-AC-D1-A1 "moderne Messeinrichtung", which generally works fine with the adapter, sends 52 data packages per minute, roughly 1 per second. 
I collect these packages and distribute them by a Ethernet converter, namely USR-TCP232-E2, https://www.pusr.com/products/dual-ttl-to-ethernet-modules-usr-tcp232-e2.html 
The adapter is set to "network, read only" and connects. All green.
But the data  recorded by history show a delay or dilatation in time, which increases the times between events.

**To Reproduce**  
Steps to reproduce the behavior:
for more detail, please refer https://forum.iobroker.net/topic/36851/smartmeter-buffering-dateneingang-via-tcp-zeitdilatation
1. adjust sampling interval ro something >1, e.g. 5
2. Click restart adapter
3. wait 1 minute
4. switch on elctrical load so see the step response, e.g. 2kW load (stove)
5 wait an additional 1 minute
6 switch off the load 
7 compare real time with recorded time  

**Expected behavior**  
The power step in the objects data should follow the load step istantanously.

**Screenshots & Logfiles**  
I compared the behaviour of the built in TCP client wiht an optional virtual com driver provided on the manufacturers website, https://www.pusr.com/support/downloads/usr-vcom-setup-software.html   . 

Using this virtual com driver to generate a COM port and setting the adapter to the generated COM port the adapter behaves as expected without any remarkable time delay.   

The virtual com driver SW delivers a config pages which includes a package counter as depicted here in the red box: 
[image]
This counter keeps continously counting, independant of the setting of the sampling interval of the adapter or of the adapter at all. It seems that all packeges are received and dropped if not used.

The TCP client of the adapter seems to handle it in different manner. It seems to pile up the packages in a buffer and pulling them out in the manner of a shift register or fifo when requested by the application.
Just a guess, I am not an network expert. 

**Versions:**  
 - Adapter version: 3.1.3
 - JS-Controller version: 3.1.6
 - Node version: v10.17.0
 - Operating system: Win 10 Pro

**Additional context**  
Btw. You can set  the meter "Itron OpenWay 3.HZ-AC-D1-A1" on the white list. The extraction and interpretation of data seems to be correct. No complains there.

Apollon77, 2020-09-21:
try 3.1.4 (latest this afternoon, github already now)

klassisch, 2020-09-21:
Thanks for your effort.
But sorry, the result remains unchanged.
[image]
I updated, uploaded the adapter and restarted the complete ioBroker. But the behavior ist still uncanged. With smpling interval of 5 seconds I still see the delays. With parameter 0 all is as it should be.

Apollon77, 2020-09-21:
Ahhhhhhhh ... yes I see my error ... need to fix ... ggrrmmpfff

Apollon77, 2020-09-21:
Versuch nochmal Github bzw latest dann morgen

klassisch, 2020-09-21:
Great, now it runs as expected. I tested with 5s, 10s, 30s and the current power was read at the next sampling point. No more delays, buffered data and so on.
So, I have to update my guide in https://forum.iobroker.net/topic/36936/guide-zwangsumstellung-auf-smartmeter-freut-euch-drauf 
Thanks a lot and have a nice day!

closed 2020-09-21 by klassisch

klassisch, 2020-09-21:
Just a small remark, which is neither worth an issue neither an update, but could be considered in the next content driven update: The adapter config menue refers SML as SmartMeterLanguage. Currently in other sources it is referenced as Smart Message Language, https://de.wikipedia.org/wiki/Smart_Message_Language ; https://wiki.volkszaehler.org/software/sml  
Sorry, that I did not oberve this earlier.

Apollon77, 2020-09-21:
Can you create an own issue please ... or provide a PR :-)
(https://github.com/Apollon77/ioBroker.smartmeter/blob/master/admin/words.js ... click the pencil top right ... edit as needed, scroll down, propose as change :-)