Apollon77/ioBroker.smartmeter #160: Smartmeter anderen Datenpunkt als Quelle

lolax80, 2020-09-30:
Hallo,

erst einmal vielen Danke für den tollen Adapter.
Ich habe Wemos D1 mit einem Lesekopf und Tasmota drauf. Der Iobroker und der Wemos befinden sich an zwei unterschiedlichen Orten.
Ich würde jetzt gerne die Werte welche Tasmota einliest (kWh) an den Smartmeter Adapter schicken wollen.

Könnte man es einen zusätzlichen Punkt bei Datenübertragung geben, wo man einen Objekt/ Datenpunkt aus dem Iobroker auswählen könnte? So könnte man direkt Werte in den Smartmeter einspielen.

Mit freundlichen Grüßen
Alex

MMeinhardt1, 2020-10-01:
Hi, 
ich hab’s noch nicht getestet aber der SmartMeter Adapter kann Daten über das Netzwerk empfangen. 
Und warum machst du das nicht über MQTT mit dem sonoff Adapter?

Apollon77, 2020-10-01:
Verstehe ich es korrekt das Du mit der Wemos schon die Daten parst? oder wie? Was ist dann der Sinn es noch zum Adapter zu schicken?

lolax80, 2020-10-01:
Hallo,

fande die Berechnungen des Adapters toll.
Werde es jetzt aber mit sourceanalytix auswerten.

Danke

Apollon77, 2020-10-01:
Was berechnet der Adapter denn? Der Adapter parst D0 und SML und stellt die Daten als States zur Verfügung. Ich glaube du verwechselst das mit einem anderen Adapter.

closed 2020-10-01 by Apollon77

lolax80, 2020-10-01:
Ok, dann habe ich wohl damals was anderes genutzt.

Sorry