Apollon77/ioBroker.smartmeter #165: DZG DVS74 SmartMeter Parameter um momentane Wirkleistung Phase 1-3 erweitern bzw. freie Parameterangabe für individuelles Auslesen von Parametern

jug85, 2020-11-01:
Der DZG DVS74 SmartMeter kann keine negative momentane Wirkleistung darstellen. Die Momentane Wirkleistung (OBIS: 1-0:16.7.0*255) zeigt keine negativen Werte an. Daher kann keine Einspeisung ermittelt werden. 

Die momentane Wirkleistung Phasen 1-3 (OBIS: 01-00:24.07.00.FF (PL1), 01-00:38.07.00*FF (PL2), 01-00:4C.07.00*FF (PL3)) enthalten die momentane Wirkleistung als Zahlenwert mit Vorzeichen. Hier könnte demnach eine Einspeisung ermittelt werden.
https://www.dzg.de/fileadmin/dzg/content/downloads/produkte-zaehler/dvs74/dzg_dvs74_handbuch.pdf
Seite 36
Feature Request: Freie Konfiguration für individuelles Auslesen von OBIS-Attributen als optionale Erweiterung. Alternativ Aufnahmen der momentanen Wirkleistung Phasen 1-3.

Apollon77, 2020-11-01:
Hm ... wenn die Daten im SML enthalten wären dann wären Sie in den Objekten. Also was ist genau das Thema?  Bitte mal Debug Log aktivieren und hier zeigen. Sind eiD aten da drin? Irgendwelche Warnungen vom Adapter? Ansonsten ist die Frag e wie diese "optionalen Erweiterungen" aktiviert werden können.+

Der Adapter kann nur das anzeigen was auch im SML ist

jug85, 2020-11-01:
Hi, hier sind die Daten aus dem Debug-Log: 

2020-11-01 19:49:15.276 - debug: smartmeter.0 (13480) SET MESSAGE TIMEOUT TIMER: 120000
--
2020-11-01 19:49:15.276 - debug: smartmeter.0 (13480) RESUME READING SERIALPORT
2020-11-01 19:49:15.277 - debug: smartmeter.0 (13480) SCHEDULE NEXT RUN IN 5s
2020-11-01 19:49:15.286 - debug: smartmeter.0 (13480) SERIALPORT CLOSE
2020-11-01 19:49:20.281 - debug: smartmeter.0 (13480) SERIALPORT OPEN
2020-11-01 19:49:20.288 - debug: smartmeter.0 (13480) SET MESSAGE TIMEOUT TIMER: 120000
2020-11-01 19:49:21.323 - debug: smartmeter.0 (13480) ADD NEW DATA (0 + NEW 1)
2020-11-01 19:49:21.325 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b" -> null
2020-11-01 19:49:21.329 - debug: smartmeter.0 (13480) ADD NEW DATA (1 + NEW 5)
2020-11-01 19:49:21.331 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b0101" -> null
2020-11-01 19:49:21.340 - debug: smartmeter.0 (13480) ADD NEW DATA (6 + NEW 4)
2020-11-01 19:49:21.341 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b010101017609" -> null
2020-11-01 19:49:21.344 - debug: smartmeter.0 (13480) ADD NEW DATA (10 + NEW 7)
2020-11-01 19:49:21.345 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e88" -> null
2020-11-01 19:49:21.348 - debug: smartmeter.0 (13480) ADD NEW DATA (17 + NEW 7)
2020-11-01 19:49:21.350 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263" -> null
2020-11-01 19:49:21.357 - debug: smartmeter.0 (13480) ADD NEW DATA (24 + NEW 7)
2020-11-01 19:49:21.360 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b3030" -> null
2020-11-01 19:49:21.375 - debug: smartmeter.0 (13480) ADD NEW DATA (31 + NEW 14)
2020-11-01 19:49:21.376 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffff" -> null
2020-11-01 19:49:21.382 - debug: smartmeter.0 (13480) ADD NEW DATA (45 + NEW 4)
2020-11-01 19:49:21.384 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffff" -> null
2020-11-01 19:49:21.399 - debug: smartmeter.0 (13480) ADD NEW DATA (49 + NEW 9)
2020-11-01 19:49:21.407 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a470001" -> null
2020-11-01 19:49:21.411 - debug: smartmeter.0 (13480) ADD NEW DATA (58 + NEW 18)
2020-11-01 19:49:21.412 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e88" -> null
2020-11-01 19:49:21.416 - debug: smartmeter.0 (13480) ADD NEW DATA (76 + NEW 3)
2020-11-01 19:49:21.417 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b6200" -> null
2020-11-01 19:49:21.421 - debug: smartmeter.0 (13480) ADD NEW DATA (79 + NEW 5)
2020-11-01 19:49:21.422 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b62006200726307" -> null
2020-11-01 19:49:21.426 - debug: smartmeter.0 (13480) ADD NEW DATA (84 + NEW 5)
2020-11-01 19:49:21.427 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b6200620072630701770b3030" -> null
2020-11-01 19:49:21.430 - debug: smartmeter.0 (13480) ADD NEW DATA (89 + NEW 3)
2020-11-01 19:49:21.432 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b6200620072630701770b3030323330" -> null
2020-11-01 19:49:21.451 - debug: smartmeter.0 (13480) ADD NEW DATA (92 + NEW 5)
2020-11-01 19:49:21.453 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b6200620072630701770b30303233303330393334" -> null
2020-11-01 19:49:21.457 - debug: smartmeter.0 (13480) ADD NEW DATA (97 + NEW 18)
2020-11-01 19:49:21.458 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b6200620072630701770b303032333033303933340b0901445a4700015f6c96070100620affff" -> null
2020-11-01 19:49:21.462 - debug: smartmeter.0 (13480) ADD NEW DATA (115 + NEW 6)
2020-11-01 19:49:21.463 - debug: smartmeter.0 (13480) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760980000000064e885a620062007263010176010b303032333033303933340bfff1ffffffffffffffff0b0901445a4700015f6c960101638ce500760980000000064e885b6200620072630701770b303032333033303933340b0901445a4700015f6c96070100620affff7262016508ba" -> null
2020-11-01 19:49:21.466 - debug: smartmeter.0 (13480) ADD NEW DATA (121 + NEW 4)
[... cut ...]
2020-11-01 19:49:21.588 - info: smartmeter.0 (13480) terminating
2020-11-01 19:49:21.590 - debug: smartmeter.0 (13480) Plugin sentry destroyed
2020-11-01 19:49:21.591 - info: smartmeter.0 (13480) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2020-11-01 19:49:21.607 - debug: smartmeter.0 (13480) connected set to false

Ich habe gerade gesehen, dass ich dem dzg DWSE 20.xxx verbaut habe. Leider habe ich kein Handbuch dazu im Netz gefunden. Kann es sein, dass der Zähler nicht mehr Infos liefert?

Apollon77, 2020-11-01:
Ja das ist die erste Annahme. Das log ist zu wenig

sbeier23, 2020-11-06:
Wäre das passend? Ich bekomme nämlich nur drei Werte vom Adapter aus meinem SmartMeter:

smartmeter.0 | 2020-11-06 22:56:39.164 | debug | (9483) SERIALPORT CLOSE
-- | -- | -- | --
smartmeter.0 | 2020-11-06 22:56:39.107 | debug | (9483) SCHEDULE NEXT RUN IN 300s
smartmeter.0 | 2020-11-06 22:56:39.102 | debug | (9483) RESUME READING SERIALPORT
smartmeter.0 | 2020-11-06 22:56:39.098 | debug | (9483) SET MESSAGE TIMEOUT TIMER: 120000
smartmeter.0 | 2020-11-06 22:56:39.092 | info | (9483) Received 3 values, 0 updated
smartmeter.0 | 2020-11-06 22:56:39.089 | debug | (9483) Data for 1-0:1_8_0__255 unchanged
smartmeter.0 | 2020-11-06 22:56:39.085 | debug | (9483) 1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 6987 kWh
smartmeter.0 | 2020-11-06 22:56:39.080 | debug | (9483) Data for 1-0:96_1_0__255 unchanged
smartmeter.0 | 2020-11-06 22:56:39.076 | debug | (9483) 1-0:96.1.0*255: Serialnumber = 0a01445a47000394dcb8
smartmeter.0 | 2020-11-06 22:56:39.072 | debug | (9483) Data for 1-0:96_50_1__1 unchanged
smartmeter.0 | 2020-11-06 22:56:39.067 | debug | (9483) 1-0:96.50.1*1: Wartungseintrag (T1) letztes Jahr = DZG
smartmeter.0 | 2020-11-06 22:56:39.061 | debug | (9483) STORE DATA
smartmeter.0 | 2020-11-06 22:56:39.055 | debug | (9483) SML MESSAGE: START SML-File START SmlMessage Transaction-ID: � (b7100809) Group-No: 0 Abort On Error: 0 Message-Body: SmlPublicOpenResponse Codepage: undefined Client-ID: un
smartmeter.0 | 2020-11-06 22:56:39.035 | debug | (9483) MSG IDX 354
smartmeter.0 | 2020-11-06 22:56:39.031 | debug | (9483) PAUSE READING SERIALPORT TO HANDLE MESSAGE
smartmeter.0 | 2020-11-06 22:56:39.026 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:39.022 | debug | (9483) ADD NEW DATA (370 + NEW 15)
smartmeter.0 | 2020-11-06 22:56:39.008 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:39.003 | debug | (9483) ADD NEW DATA (354 + NEW 16)
smartmeter.0 | 2020-11-06 22:56:38.988 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.984 | debug | (9483) ADD NEW DATA (340 + NEW 14)
smartmeter.0 | 2020-11-06 22:56:38.969 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.964 | debug | (9483) ADD NEW DATA (324 + NEW 16)
smartmeter.0 | 2020-11-06 22:56:38.949 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.944 | debug | (9483) ADD NEW DATA (294 + NEW 30)
smartmeter.0 | 2020-11-06 22:56:38.928 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.922 | debug | (9483) ADD NEW DATA (279 + NEW 15)
smartmeter.0 | 2020-11-06 22:56:38.904 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.899 | debug | (9483) ADD NEW DATA (247 + NEW 32)
smartmeter.0 | 2020-11-06 22:56:38.881 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.876 | debug | (9483) ADD NEW DATA (217 + NEW 30)
smartmeter.0 | 2020-11-06 22:56:38.856 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.850 | debug | (9483) ADD NEW DATA (202 + NEW 15)
smartmeter.0 | 2020-11-06 22:56:38.831 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.826 | debug | (9483) ADD NEW DATA (187 + NEW 15)
smartmeter.0 | 2020-11-06 22:56:38.804 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b37377070100603201010172620162006200520004445a
smartmeter.0 | 2020-11-06 22:56:38.796 | debug | (9483) ADD NEW DATA (177 + NEW 10)
smartmeter.0 | 2020-11-06 22:56:38.028 | debug | (9483) MATCH-RESULT MESSAGE: "0047000394dcb8726201650302b1b36202633707007605b5100809620062007263070177010b0a01445a47000394dcb8070100620affff726201650302b1b3737707010060320101 [... cut ...]
smartmeter.0 | 2020-11-06 22:51:35.829 | debug | (9483) CREATE SERIALPORT: 9600 8 1 none
smartmeter.0 | 2020-11-06 22:51:35.778 | debug | (9483) SmartmeterObis options: {"debug":2,"protocol":"SmlProtocol","transport":"SerialResponseTransport","requestInterval":"300","anotherQueryDelay":"1000","transportSerialPort":"/dev/ttyUSB0","transp
smartmeter.0 | 2020-11-06 22:51:35.316 | info | (9483) starting. Version 3.1.5 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.19.0, js-controller: 3.1.6

Apollon77, 2020-11-11:
Naja, dann kommt nicht mehr rraus. Siehe Forum, ggf kannst ne PIN von deinem Netzbtreiber bekommen mit der du mehr Daten freischalten kannst

closed 2020-11-11 by Apollon77