Apollon77/ioBroker.smartmeter #166: Zähler der Energie AG Oberösterreich

Negalein, 2020-11-01:
Hallo Ingo!

Wir waren schon im Forum per Chat deswegen in Kontakt.

Wir in Österreich (zumindest in OÖ mit der Energie AG) benötigen einen 32 stelligen Key um die Datenschnittstelle auslesen zu können.
Jetzt wäre es toll, wenn die irgendwie mit deinem Adapter auch funktionieren würde.

Derzeit bekomme ich die Werte mit Volkszähler und hole mir dann die Werte mit einem Script in ioBroker.

LG

Apollon77, 2020-11-01:
Ja das ist ein leicht anderes verschlüsseltes Protokoll ... steht auf der Liste wenn ich mal dazu Zeit finde

Negalein, 2021-01-11:
mit diesem Node-Red kommen viele Daten in ioBroker!

https://www.loxwiki.eu/pages/viewpage.action?pageId=82149444

ManuelF001, 2021-05-13:
Das wäre echt toll, wenn der Adapter das können würde.
Ist es die nächsten Monate geplant, dass es realisiert wird?

Apollon77, 2021-05-14:
>  Ist es die nächsten Monate geplant, dass es realisiert wird?

Alles eine Frage der Zeit und die ist gerade der stark limitierende Faktor ... kann nichts versprechen. Nehme aber gern PRs an

Negalein, 2021-11-28:
> EIn Daumen hoch" im ersten Post erhöht die sichtbarkeit

gibts bereits einen Fortschritt?

Apollon77, 2021-11-28:
Sorry, aber Zeit ist aktuell ein sehr großes Thema ... und einfach nicht da :-( (nicht nur für dieses interessante Thema sondern für gefühlt 20 weitere). Ich lade jeden Entwickler ein mitzuarbeiten. Alles (ggf ein Skript in JavaScript was die daten verarbeiteb kann oders so) hilft ... Ich habe bishe rnichts gefunden was eine Grundlage wäre und muss daher bei 0 anfangen bzw aus anderen Sprachen adaptern (weas bei crypto kram immer wenig spass macht).

Negalein, 2021-11-28:
> Alles (ggf ein Skript in JavaScript was die daten verarbeiteb kann oders so) hilft ... Ich habe bishe rnichts gefunden was eine Grundlage wäre und muss daher bei 0 anfangen bzw aus anderen Sprachen adaptern (weas bei crypto kram immer wenig spass macht).

ging es mit diesem Node Red?
https://loxwiki.atlassian.net/wiki/spaces/LOX/pages/1597866043/Amis+Z+hler+Netz+O+per+NodeRed+und+Raspberry

ManuelF001, 2021-11-29:
Dieses Nodered habe ich auch in Verwendung. Allerdings muss man die Konfiguration für aes nach jedem Update vom Nodered Adapter neu machen.

Apollon77, 2021-11-29:
Den Link hab ich ... aber am Ende um diese Node-red nodes am Ende in sinnvollen JavsScript code zurückzuübersetzen ist meeeh aufwand. Ich muss node-red machen und den Flow importieren, dann für jede node den javascript code raussuchen und alles sinnvoll in "Normales javascript "bringen. Auch nicht ohne und frisst alles zeit ... wenn das jemand machen kann ists für mich einfacher ... ist einfach so

martink173, 2022-01-15:
> Wir in Österreich (zumindest in OÖ mit der Energie AG) benötigen einen 32 stelligen Key um die Datenschnittstelle auslesen zu können. Jetzt wäre es toll, wenn die irgendwie mit deinem Adapter auch funktionieren würde.

Welche Zähler habt ihr in OÖ?

ManuelF001, 2022-01-15:
Das ist der Siemens TD-3511. (Netz OÖ)

noriyuki41, 2022-08-05:
Ich bin an dieser Erweiterung auch interessiert, gibt es schon eine Perspektive?

ManuelF001, 2022-08-10:
Könnte eine kleine Spende die Motivation steigern?

noriyuki41, 2022-08-11:
> Könnte eine kleine Spende die Motivation steigern?

Ich würde mich anschließen!

Apollon77, 2022-08-30:
Ich fürchte eine Spende erhöht nur bedingt die Anzahl der verfügbaren Stunden am Tag :-( Es ist am Ende die Kombination aus Zeit, Länge der Todo-Liste und einpriorisierung eines Thema nach "Anzahl interessenten/Nutzern" bzw priorität ...

Ranzig93, 2022-12-25:
Hallo,

falls noch wer ein Javascript benötigt zum Auslesen.

https://github.com/Ranzig93/TD3511-IoBroker-Javascript

ManuelF001, 2022-12-26:
@Ranzig93 
Super, dass sich hier was tut.

Würde man sich dadurch diesen Schritt ersparen?:
AES128 Library zur Entschlüsselung der Daten notwendig
1. JavaScript Paket installieren: npm install aes-js
• Infos: https://github.com/ricmoo/aes-js
2. JavaScript Paket verfügbar machen: 
• Editieren von .node-red/settings.js
• Unter functionGlobalContext: { folgendes einfügen: aesjs:require("aes-js")

Das war nämlich das, was bei jedem Update von Node Red weg war.

Ranzig93, 2022-12-26:
@ManuelF001 

Zu Node Red kann ich nichts sagen, da ich es nicht verwende.

Ja, nach einem Update vom Javascript Adapter sollten die NPM Module weiterhin verfügbar sein.

Über Palette verwalten? Da kann ich nämlich mit aes nicht finden.
In den Instanzeinstellungen von deiner Javascript Instanz musst du in das Feld "Zusätzliche NPM-Module" und "Aktivieren Sie die Syntaxhilfe für diese npm-Module:"folgende Module hinzufügen: aes-js, serialport
anschließend die Instanz neu starten, und schon sollte das Script funktionieren.

[image]

Apollon77, 2022-12-26:
ich schaue mir das die Tage mal an, Danke!

ManuelF001, 2022-12-26:
@Ranzig93 
Ah, jetzt kapiere ich es erst. Das wäre natürlich der Hammer, wenn das so problemlos und einfach funktionieren würde.
Ich werde das heute Abend gleich mal testen. Danke schonmal!

ManuelF001, 2022-12-26:
Getestet und es funktionierte sofort problemlos.
Vielen Dank!

ManuelF001, 2022-12-27:
Gibt es eigentlich einen Standard, ob Strombezug oder Stromeinspeisung mit negativem Vorzeichen dargestellt wird?
Mit dieser Lösung war es genau umgekehrt:
[C:\Users\Admin\Desktop\2022-12-27 19_28_13-objects - iobroker.jpg](https://loxwiki.atlassian.net/wiki/spaces/LOX/pages/1597866043/Amis+Z+hler+Netz+O+per+NodeRed+und+Raspberry)

Also Strombezug mit negativem Vorzeichen und Stromeinspeisung mit positivem Vorzeichen.

Ranzig93, 2022-12-28:
Dazu habe ich keine Norm auf die schnelle gefunden. 
Aber ich habe es im Code angepasst und das Vorzeichen geändert.

ManuelF001, 2022-12-29:
Bei mir läuft es seit gestern problemlos im Echtsystem. Vielen Dank!

australien340, 2023-01-19:
Kann man das auch auf das Burgenland und den Landis-Gyr E450 erweitern?
Dieser hat ja keine usbSerial, dafür einen Authentication Key.

noriyuki41, 2023-02-14:
> Bei mir läuft es seit gestern problemlos im Echtsystem. Vielen Dank!
@ManuelF001 Könntest du bitte deine Erkenntnisse in eine kleine Beschreibung packen was du genau wo/wie installiert oder eingestellt hast.
Danke im Voraus!
Stefan

ManuelF001, 2023-02-14:
Ja, komme aber erst am Donnerstag dazu.

Ranzig93, 2023-02-14:
> Kann man das auch auf das Burgenland und den Landis-Gyr E450 erweitern? Dieser hat ja keine usbSerial, dafür einen Authentication Key. Die Verschlüsselung basiert auf DLMS/COSEM Security Suite 0 nach HLS5 - was auch immer das heissen mag.

Leider unterscheidet sich das Protokoll/Verschlüsselung. Es gibt zwar Beschreibungen im Internet, aber ohne Hardware zum Testen ist das zu aufwendig.

ManuelF001, 2023-02-16:
> > Bei mir läuft es seit gestern problemlos im Echtsystem. Vielen Dank!
> > @ManuelF001 Könntest du bitte deine Erkenntnisse in eine kleine Beschreibung packen was du genau wo/wie installiert oder eingestellt hast.
> > Danke im Voraus!
> > Stefan

1.) In der Admin Übersicht vom ioBroker links auf Skripte klicken. Neues Skript erstellen und den Inhalt einfügen. Bei var key_128 = den richtigen key eintragen und ev. USB Schnittstelle korigieren.
2.) Bei Instanzen -> Javascript -> Einstellungen ->NPM Module aes-js, serialport einfügen
3.) Instanz neu starten

noriyuki41, 2023-02-21:
> 1.) In der Admin Übersicht vom ioBroker links auf Skripte klicken. Neues Skript erstellen und den Inhalt einfügen. Bei var key_128 = den richtigen key eintragen und ev. USB Schnittstelle korigieren. [...]

Herzlichen Dank, läuft bei mir jetzt auch auf dem iobroker Slave!

Apollon77, 2023-02-21:
Super das Irh wege gefunden habt wie es geht ... Sobald ich mit "Matter" einen Stand erreicht habe wo ich mal wieder auf andere größere Baustellen schauen kann kommt das dran.