Apollon77/ioBroker.smartmeter #168: No or too long answer from Serial Device after last request after nodejs update

hacki11, 2020-11-18:
After updating nodejs from 12.16 to 12.19.1 the smartmeter adapter can not receive data from my Weidmann USB Dongle.

Seems related, but i want to use serial like before
https://github.com/Apollon77/ioBroker.smartmeter/issues/117

Configuration
[image]

**Expected behavior**  
Smartmeter should fetch the values from dongle as before the update

**Screenshots & Logfiles**  
```
2020-11-18 22:42:55.389  - info: host.nuc instance system.adapter.smartmeter.0 started with pid 7731
2020-11-18 22:42:55.793  - debug: smartmeter.0 (7731) Redis Objects: Use Redis connection: 127.0.0.1:9001
2020-11-18 22:42:55.812  - debug: smartmeter.0 (7731) Objects client ready ... initialize now
2020-11-18 22:42:55.814  - debug: smartmeter.0 (7731) Objects create System PubSub Client
2020-11-18 22:42:55.815  - debug: smartmeter.0 (7731) Objects create User PubSub Client
2020-11-18 22:42:55.816  - debug: smartmeter.0 (7731) Objects client initialize lua scripts
2020-11-18 22:42:55.826  - debug: smartmeter.0 (7731) Objects connected to redis: 127.0.0.1:9001
2020-11-18 22:42:55.830  - debug: smartmeter.0 (7731) objectDB connected
2020-11-18 22:42:55.831  - debug: smartmeter.0 (7731) Redis States: Use Redis connection: 127.0.0.1:9000
2020-11-18 22:42:55.836  - debug: smartmeter.0 (7731) States create User PubSub Client
2020-11-18 22:42:55.837  - debug: smartmeter.0 (7731) States create System PubSub Client
2020-11-18 22:42:55.845  - debug: smartmeter.0 (7731) States connected to redis: 127.0.0.1:9000
2020-11-18 22:42:55.846  - debug: smartmeter.0 (7731) statesDB connected
2020-11-18 22:42:56.176  - debug: smartmeter.0 (7731) Plugin sentry Initialize Plugin (enabled=true)
2020-11-18 22:42:56.179  - info: smartmeter.0 (7731) Plugin sentry Sentry Plugin disabled for this process because sending of statistic data is disabled for the system
2020-11-18 22:42:56.185  - info: smartmeter.0 (7731) starting. Version 3.1.6 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.19.1, js-controller: 3.1.6
2020-11-18 22:42:56.200  - debug: smartmeter.0 (7731) SmartmeterObis options: {"debug":0,"protocol":"SmlProtocol","transport":"SerialResponseTransport","requestInterval":"20","anotherQueryDelay":"1000","transportSerialPort":"/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0JEFPE-if00-port0","transportSerialBaudrate":9600,"transportSerialDataBits":8,"transportSerialStopBits":1,"transportSerialParity":"none","transportSerialMessageTimeout":10000,"protocolSmlIgnoreInvalidCRC":false}
2020-11-18 22:42:56.210  - debug: smartmeter.0 (7731) connected set to false
2020-11-18 22:42:56.245  - silly: smartmeter.0 (7731) States system redis pmessage system.adapter.smartmeter.0.logLevel/system.adapter.smartmeter.0.logLevel:{"val":"silly","ack":true,"ts":1605735776202,"q":0,"from":"system.adapter.smartmeter.0","lc":1605733924968}
2020-11-18 22:43:06.207  - warn: smartmeter.0 (7731) No or too long answer from Serial Device after last request.
2020-11-18 22:43:06.209  - debug: smartmeter.0 (7731) Error: No or too long answer from Serial Device after last request.
2020-11-18 22:43:06.207  - warn: smartmeter.0 (7731) No or too long answer from Serial Device after last request.
2020-11-18 22:43:06.209  - debug: smartmeter.0 (7731) Error: No or too long answer from Serial Device after last request.
```

**Versions:**  
 - Adapter version: 3.1.6
 - JS-Controller version: 3.1.6
 - Node version: 12.19.1
 - Operating system: linux ubuntu

**Additional context**  
Configured the permissions, but that was not done prior the update. iobroker is part of the group dialout. i wrote a simple node serial port reader and run it as iobroker to check if that is working:
```
juergen@nuc:~/usbdump$ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Nov 14 14:00 /dev/ttyUSB0
juergen@nuc:~/usbdump$ ls -l /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0JEFPE-if00-port0
lrwxrwxrwx 1 root root 13 Nov 14 14:00 /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0JEFPE-if00-port0 -> ../../ttyUSB0
juergen@nuc:~/usbdump$ ls -l /dev/ttyUSB0
crw-rw---- 1 root dialout 188, 0 Nov 14 14:00 /dev/ttyUSB0
juergen@nuc:~/usbdump$ groups iobroker
iobroker : iobroker dialout
juergen@nuc:~/usbdump$ sudo -u iobroker node dump_usb.js
> �☺☺☺☺♦☺bbre☺☺v☺☺♥OAB♂
> ☺HLY☻☺ j☺☺c.�v♦☻bbre☺w☺♂
> ☺HLY☻☺ j☺☺�♦w☺`2☺☺☺☺☺☺♦HLY☺w☺`☺�☺☺☺☺♂
> ☺HLY☻☺ j☺w☺�e∟☺♦e♥OABb▲R�e♥�¶�☺w☺�e∟☺♦e♥OABb▲R�b☺w☺►�☺☺bS☺v☺w☺ �☺☺b#R��☺w☺4�☺☺b#R��☺w☺H�☺☺b#R��☺w☺▼�☺☺b!R�b◄☺w☺3�☺☺b!R�bd☺w☺G�☺☺b!R�b�☺w☺Q☺�☺☺Rbx☺w☺Q☻�☺☺Rb�☺w☺Q♦�☺☺Rc☺■☺w☺Q☼�☺☺Rc☺&☺w☺Q→�☺☺Rc☺=☺w☺♫�☺☺b,R�c☺�☺w☺☻☺☺☺☺   1.02.007☺w☺`Z☻☺☺☺☺☺♣A01A☺w☺`♣�☺☺☺☺e∟☺♦☺☺☺c��v♦♥bbre☻☺q☺c��→☻■n☺☺☺☺♦☺bbre☺☺v☺☺♥OAD♂
> ☺HLY☻☺ j☺☺c �v♦☻bbre☺w☺♂
> ☺HLY☻☺ j☺☺�♦w☺`2☺☺☺☺☺☺♦HLY☺w☺`☺�☺☺☺☺♂
^C
juergen@nuc:~/usbdump$ ps aux |grep smartmeter
iobroker  7731  0.3  0.6 645436 53816 ?        Sl   22:42   0:01 io.smartmeter.0
```

Do you have an idea where to look here?

Apollon77, 2020-11-19:
Please provident debug log from the adapter when it does not work.

hacki11, 2020-11-19:
What else debug log do you need? I set loglevel to silly an posted the output in the post above.

Apollon77, 2020-11-19:
Ok if this isthe putput then simply no data come in. No ideal. never had such an effect with a minor Nodejs update ...and forme it works with a comparable version of nodejs

Please run "iobroker fix" after each nodejs update!!

hacki11, 2020-11-19:
I switched back and forth and it is working again. Should have tried yesterday.
iobroker fix just prints the cli help.

closed 2020-11-19 by hacki11

Apollon77, 2020-11-19:
>  iobroker fix just prints the cli help.

This should not happen with js.conttoller 3.1.6 ... Else you could execute the fixer manually. Check in Forum for the command please

hacki11, 2020-11-19:
`curl -sL https://iobroker.net/fix.sh | bash -`
now iobroker fix works