Apollon77/ioBroker.smartmeter #174: D0 / Error: No or too long answer from Serial Device after last request.

dotcom84, 2020-12-27:
Hi,

A quick question: I have two USB IR-read-write-heads connected to my two Logarex power meters.
One works just fine, the other continously produces errors in the log (Error: No or too long answer from Serial Device after last request.). Both are configured in exactly the same way and it does not seem to be an hardware issue (I exchanged the read-write-heads and reconfigured both instances accordingly).

So I compared the minicom output for the two:

The working one seems to identify itself correctly:

> 
> /LOG5LK13BE803039
> 
> 1-0:96.1.0*255(001LOG0065093055)
> 1-0:1.8.0*255(006049.6981*kWh)
> 1-0:2.8.0*255(004188.9527*kWh)
> 1-0:0.2.0*255(ver.03,432F,20170504)                                                                                               
> 1-0:96.90.2*255(0F66)                                                                                                             
> 1-0:97.97.0*255(00000000)                                                                                                         
> !                                                                                                                                 
> /LOG5LK13BE803039                                                                                                                 
>                                                                                                                                   
> 1-0:96.1.0*255(001LOG0065093055)                                                                                                  
> 1-0:1.8.0*255(006049.6983*kWh)                                                                                                    
> 1-0:2.8.0*255(004188.9527*kWh)                                                                                                    
> 1-0:0.2.0*255(ver.03,432F,20170504)                                                                                               
> 1-0:96.90.2*255(0F66)                                                                                                             
> 1-0:97.97.0*255(00000000)                                                                                                         
> !                                                                                                                                 
> /LOG5LK13BE803039                                                                                                                 
>                                                                                                                                   
> 1-0:96.1.0*255(001LOG0065093055)                                                                                                  
> 1-0:1.8.0*255(006049.6985*kWh)                                                                                                    
> 1-0:2.8.0*255(004188.9527*kWh)                                                                                                    
> 1-0:0.2.0*255(ver.03,432F,20170504)                                                                                               
> 1-0:96.90.2*255(0F66)                                                                                                             
> 1-0:97.97.0*255(00000000)                                                                                                         
> !     

whereas the non-working one has a weird looking first line:

> úê^3jzó&?????4'nfÌ̚9                                                                                                              
>                                                                                                                                   
> 1-0:96.1.0*255(001LOG0064003291)                                                                                                  
> 1-0:1.8.0*255(007114.2625*kWh)                                                                                                    
> 1-0:1.8.1*255(003626.8814*kWh)                                                                                                    
> 1-0:1.8.2*255(003487.3811*kWh)                                                                                                    
> 1-0:2.8.0*255(000000.0000*kWh)                                                                                                    
> 1-0:0.2.0*255(ver.03,432F,20170504)                                                                                               
> 1-0:96.90.2*255(61BF)                                                                                                             
> 1-0:97.97.0*255(00000000)                                                                                                         
> !                                                                                                                                 
> úê^ú*¾ã?????¢Êjʦ¬Hxó³§N1.0*255(001LOG0064003291)                                                                               
> 1-0:1.8.0*255(007114.2625*kWh)                                                                                                    
> 1-0:1.8.1*255(003626.8814*kWh)                                                                                                    
> 1-0:1.8.2*255(003487.3811*kWh)                                                                                                    
> 1-0:2.8.0*255(000000.0000*kWh)                                                                                                    
> 1-0:0.2.0*255(ver.03,432F,20170504)                                                                                               
> 1-0:96.90.2*255(61BF)                                                                                                             
> 1-0:97.97.0*255(00000000)                                                                                                         
> !                                                                                                                                 
> 󳧙úúô3êzófèÑ0³049                                                                                                                 
>                                                                                                                                   
> 1-0:96.1.0*255(001LOG0064003291)                                                                                                  
> 1-0:1.8.0*255(007114.2626*kWh)                                                                                                    
> 1-0:1.8.1*255(003626.8814*kWh)                                                                                                    
> 1-0:1.8.2*255(003487.3812*kWh)                                                                                                    
> 1-0:2.8.0*255(000000.0000*kWh)                                                                                                    
> 1-0:0.2.0*255(ver.03,432F,20170504)                                                                                               
> 1-0:96.90.2*255(61BF)                                                                                                             
> 1-0:97.97.0*255(00000000)                                                                                                         
> !  
> 

Any ideas? Is there an option to ignore that first line and just consume the kWh values?

Apollon77, 2020-12-27:
So the question is what is the reson for these "crap" data at the beginning.

Because  it is only always the starting line it can only be a bug in the smartmeter which sends invalid stuff or it is the baudrate changeover. Please set the affected instance to debug log and let it run and post log.

Debug log = Instances - Export mode - column loglevel

dotcom84, 2020-12-28:
Sure, please find an excerp from the log file attached.
I quickly had to turn logging off again as the logfile grew to ~11MB within 30 secs...

[smartmeter_log.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/5746051/smartmeter_log.txt)

Apollon77, 2020-12-29:
I know that this is much ... I would need the "beginning" so when it starts ... can you try to grab that for me again? :-)

dotcom84, 2020-12-30:
Hi Ingo, sorry must have missed that detail... I've now attached the first couple of hundred lines from starting the adapter.

[smartmeter_log-2.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/5753121/smartmeter_log-2.txt)

Apollon77, 2021-01-31:
Reason is that the smartmeter do not seem to respond correctly ... are the settings correct? 

Is it really request/response or does the meter send from itself?

closed 2021-05-09 by stale[bot]