Apollon77/ioBroker.smartmeter #179: js-cotntroller 3.2.3 warnings

Ilovegym66, 2021-01-09:
with js-controller 3.2.3 are warnings:
smartmeter Adapter gibt ne Warnung:
smartmeter.0 2021-01-09 13:35:48.320 warn (2581) State "smartmeter.0.1-0:1_8_0__255.rawvalue" has no existing object, this might lead to an error in future versions

coyote1, 2021-01-12:
Same here:

`

smartmeter.0 | 2021-01-12 19:30:23.188 | warn | (17429) State "smartmeter.0.1-0:15_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
smartmeter.0 | 2021-01-12 19:30:23.187 | warn | (17429) State "smartmeter.0.1-0:1_8_1__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-01-12 19:30:23.187 | warn | (17429) State "smartmeter.0.1-0:1_8_0__255.rawvalue" has no existing object, this might lead to an error in future versions

`

Apollon77, 2021-01-13:
Please post a debug log that I see where it happens

Ilovegym66, 2021-01-13:
no problem 
`2021-01-13 22:54:01.491  - warn: smartmeter.0 (520) State "smartmeter.0.1-0:16_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
2021-01-13 22:54:03.956  - info: host.zeroberry stopInstance system.adapter.smartmeter.0 (force=false, process=true)
2021-01-13 22:54:04.029  - info: smartmeter.0 (520) Got terminate signal TERMINATE_YOURSELF
2021-01-13 22:54:04.033  - info: host.zeroberry stopInstance system.adapter.smartmeter.0 send kill signal
2021-01-13 22:54:04.059  - info: smartmeter.0 (520) terminating
2021-01-13 22:54:04.068  - info: smartmeter.0 (520) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2021-01-13 22:54:06.558  - info: host.zeroberry instance system.adapter.smartmeter.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2021-01-13 22:54:07.234  - info: host.zeroberry instance system.adapter.smartmeter.0 started with pid 1495
2021-01-13 22:54:18.941  - debug: smartmeter.0 (1495) Redis Objects: Use Redis connection: 192.168.3.41:6379
2021-01-13 22:54:19.256  - debug: smartmeter.0 (1495) Objects client ready ... initialize now
2021-01-13 22:54:19.280  - debug: smartmeter.0 (1495) Objects create System PubSub Client
2021-01-13 22:54:19.291  - debug: smartmeter.0 (1495) Objects create User PubSub Client
2021-01-13 22:54:19.302  - debug: smartmeter.0 (1495) Objects client initialize lua scripts
2021-01-13 22:54:19.429  - debug: smartmeter.0 (1495) Objects connected to redis: 192.168.3.41:6379
2021-01-13 22:54:19.502  - debug: smartmeter.0 (1495) objectDB connected
2021-01-13 22:54:19.521  - debug: smartmeter.0 (1495) Redis States: Use Redis connection: 192.168.3.41:6379
2021-01-13 22:54:19.595  - debug: smartmeter.0 (1495) States create System PubSub Client
2021-01-13 22:54:19.606  - debug: smartmeter.0 (1495) States create User PubSub Client
2021-01-13 22:54:19.686  - debug: smartmeter.0 (1495) States connected to redis: 192.168.3.41:6379
2021-01-13 22:54:19.697  - debug: smartmeter.0 (1495) statesDB connected
2021-01-13 22:54:20.266  - debug: smartmeter.0 (1495) Plugin sentry Initialize Plugin (enabled=true)
2021-01-13 22:54:22.021  - info: smartmeter.0 (1495) starting. Version 3.1.6 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.20.0, js-controller: 3.2.6-0
2021-01-13 22:54:22.225  - debug: smartmeter.0 (1495) SmartmeterObis options: {"debug":2,"protocol":"SmlProtocol","transport":"SerialRequestResponseTransport","requestInterval":"30","anotherQueryDelay":"1000","transportSerialPort":"/dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DN03WQ0V-if00-port0","transportSerialBaudrate":9600,"transportSerialDataBits":8,"transportSerialStopBits":1,"transportSerialParity":"none","transportSerialMessageTimeout":60000,"protocolSmlIgnoreInvalidCRC":false}
2021-01-13 22:54:22.250  - debug: smartmeter.0 (1495) CREATE SERIALPORT: 9600 8 1 none
2021-01-13 22:54:22.336  - debug: smartmeter.0 (1495) SERIALPORT OPEN
2021-01-13 22:54:22.454  - debug: smartmeter.0 (1495) NEW DATA
2021-01-13 22:54:22.469  - debug: smartmeter.0 (1495) MATCH-RESULT MESSAGE: "006e1b1b1b1b01010101760509dcd1a062006200726301017601" -> null
2021-01-13 22:54:22.478  - debug: smartmeter.0 (1495) SERIALPORT RESET BAUDRATE TO 9600
2021-01-13 22:54:22.506  - debug: smartmeter.0 (1495) NEW DATA
2021-01-13 22:54:22.512  - debug: smartmeter.0 (1495) MATCH-RESULT MESSAGE: "006e1b1b1b1b01010101760509dcd1a062006200726301017601010503499b360b090149534b00042e40220101634c9700760509dcd1a1620062007263070177010b090149534b00042e4022070100620affff72620165048b647e7777078181c78203ff010101010449534b0177070100000009ff010101010b090149534b00042e402201" -> null
2021-01-13 22:54:22.522  - debug: smartmeter.0 (1495) INITIAL MESSAGES TO SEND: 0
2021-01-13 22:54:22.528  - debug: smartmeter.0 (1495) DONE SEND 0
2021-01-13 22:54:22.534  - debug: smartmeter.0 (1495) SET MESSAGE TIMEOUT TIMER2: 60000
2021-01-13 22:54:22.572  - debug: smartmeter.0 (1495) NEW DATA
2021-01-13 22:54:22.578  - debug: smartmeter.0 (1495) MATCH-RESULT MESSAGE: "006e1b1b1b1b01010101760509dcd1a062006200726301017601010503499b360b090149534b00042e40220101634c9700760509dcd1a1620062007263070177010b090149534b00042e4022070100620affff72620165048b647e7777078181c78203ff010101010449534b0177070100000009ff010101010b090149534b00042e4022011e52ff590000000005d5b00a0177070100010801ff0101621e52ff59000000" -> null
2021-01-13 22:54:22.604  - debug: smartmeter.0 (1495) NEW DATA
2021-01-13 22:54:22.610  - debug: smartmeter.0 (1495) MATCH-RESULT MESSAGE: "006e1b1b1b1b01010101760509dcd1a062006200726301017601010503499b360b090149534b00042e40220101634c9700760509dcd1a1620062007263070177010b090149534b00042e4022070100620affff72620165048b647e7777078181c78203ff010101010449534b0177070100000009ff010101010b090149534b00042e4022011e52ff590000000005d5b00a0177070100010801ff0101621e52ff590000000005d5b00a0177070100010802ff0101621e52ff5900000000000000000177070100100700ff0101621b520055000002ea0177078181c78205ff01010101" -> null
2021-01-13 22:54:22.619  - debug: smartmeter.0 (1495) connected set to false
2021-01-13 22:54:22.630  - debug: smartmeter.0 (1495) NEW DATA
2021-01-13 22:54:22.635  - debug: smartmeter.0 (1495) MATCH-RESULT MESSAGE: "006e1b1b1b1b01010101760509dcd1a062006200726301017601010503499b360b090149534b00042e40220101634c9700760509dcd1a1620062007263070177010b090149534b00042e4022070100620affff72620165048b647e7777078181c78203ff010101010449534b0177070100000009ff010101010b090149534b00042e4022011e52ff590000000005d5b00a0177070100010801ff0101621e52ff590 [... cut ...]
2021-01-13 22:55:25.813  - debug: smartmeter.0 (1495) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760509dcd22462006200726301017601010503499b620b090149534b00042e40220101635ccd00760509dcd225620062007263070177010b090149534b00042e4022070100620affff72620165048b64bd7777078181c78203ff010101010449534b0177070100000009ff010101010b090149534b00042e40220177070100010800ff650000018201621e52ff590000000005d5b08e0177070100010801ff0101621e52ff590000000005d5b08e0177070100010802ff0101621e52ff5900000000000000000177070100100700ff0101621b520055000002ee0177078181c78205ff0101010183024ce8ba4f6976df2127f753b3cd1b94a95c3e8609ea4cf27ca90c33362f36d819b542e9" -> null
`

Apollon77, 2021-01-13:
fixed by 3.1.7

closed 2021-01-13 by Apollon77

coyote1, 2021-01-14:
Updatet to 3.1.7, but warnings are still there:

```smartmeter.0	2021-01-14 06:45:34.753	info	(8631) Received 7 values, 7 updated
smartmeter.0	2021-01-14 06:45:34.748	debug	(8631) Set State 129-129:199_130_5__255.value = 754282561329551fea230b2c6a10a8fb8b7fced004f7f654fcf548d05d780702f0a2a92ecf23db366d218de61e7f03f8
smartmeter.0	2021-01-14 06:45:34.745	warn	(8631) State "smartmeter.0.129-129:199_130_5__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.743	debug	(8631) Set State 129-129:199_130_5__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.741	debug	(8631) Create State 129-129:199_130_5__255.value
smartmeter.0	2021-01-14 06:45:34.739	debug	(8631) Create Channel 129-129:199_130_5__255 with name Forename/Public Key
smartmeter.0	2021-01-14 06:45:34.735	debug	(8631) 129-129:199.130.5*255: Forename/Public Key = 754282561329551fea230b2c6a10a8fb8b7fced004f7f654fcf548d05d780702f0a2a92ecf23db366d218de61e7f03f8
smartmeter.0	2021-01-14 06:45:34.722	debug	(8631) Set State 1-0:15_7_0__255.value = 343.7
smartmeter.0	2021-01-14 06:45:34.715	warn	(8631) State "smartmeter.0.1-0:15_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.712	debug	(8631) Set State 1-0:15_7_0__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.708	debug	(8631) Create State 1-0:15_7_0__255.value
smartmeter.0	2021-01-14 06:45:34.704	debug	(8631) Create Channel 1-0:15_7_0__255 with name Momentanwert (Total)
smartmeter.0	2021-01-14 06:45:34.702	debug	(8631) 1-0:15.7.0*255: Momentanwert (Total) = 343.7 W
smartmeter.0	2021-01-14 06:45:34.691	debug	(8631) Set State 1-0:1_8_2__255.value = 0
smartmeter.0	2021-01-14 06:45:34.677	warn	(8631) State "smartmeter.0.1-0:1_8_2__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.674	debug	(8631) Set State 1-0:1_8_2__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.662	debug	(8631) Create State 1-0:1_8_2__255.value
smartmeter.0	2021-01-14 06:45:34.661	debug	(8631) Create Channel 1-0:1_8_2__255 with name Zählerstand 1 Summe Wirkarbeit Bezug + (T2)
smartmeter.0	2021-01-14 06:45:34.660	debug	(8631) 1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 0 kWh
smartmeter.0	2021-01-14 06:45:34.651	debug	(8631) Set State 1-0:1_8_1__255.value = 43747.1089
smartmeter.0	2021-01-14 06:45:34.647	warn	(8631) State "smartmeter.0.1-0:1_8_1__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.644	debug	(8631) Set State 1-0:1_8_1__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.643	debug	(8631) Create State 1-0:1_8_1__255.value
smartmeter.0	2021-01-14 06:45:34.639	debug	(8631) Create Channel 1-0:1_8_1__255 with name Zählerstand 1 Summe Wirkarbeit Bezug + (T1)
smartmeter.0	2021-01-14 06:45:34.639	debug	(8631) 1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 43747.1089 kWh
smartmeter.0	2021-01-14 06:45:34.635	debug	(8631) Set State 1-0:1_8_0__255.value = 43747.1089
smartmeter.0	2021-01-14 06:45:34.630	warn	(8631) State "smartmeter.0.1-0:1_8_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.628	debug	(8631) Set State 1-0:1_8_0__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.627	debug	(8631) Create State 1-0:1_8_0__255.value
smartmeter.0	2021-01-14 06:45:34.626	debug	(8631) Create Channel 1-0:1_8_0__255 with name Zählerstand 1 Summe Wirkarbeit Bezug + (Total)
smartmeter.0	2021-01-14 06:45:34.625	debug	(8631) 1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 43747.1089 kWh
smartmeter.0	2021-01-14 06:45:34.623	debug	(8631) Set State 1-0:0_0_9__255.value = 06454d480102715b7470
smartmeter.0	2021-01-14 06:45:34.620	warn	(8631) State "smartmeter.0.1-0:0_0_9__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.609	debug	(8631) Set State 1-0:0_0_9__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.608	debug	(8631) Create State 1-0:0_0_9__255.value
smartmeter.0	2021-01-14 06:45:34.606	debug	(8631) Create Channel 1-0:0_0_9__255 with name Device ID
smartmeter.0	2021-01-14 06:45:34.605	debug	(8631) 1-0:0.0.9*255: Device ID = 06454d480102715b7470
smartmeter.0	2021-01-14 06:45:34.599	debug	(8631) Set State 129-129:199_130_3__255.value = EMH
smartmeter.0	2021-01-14 06:45:34.598	debug	(8631) SERIALPORT CLOSE
smartmeter.0	2021-01-14 06:45:34.594	warn	(8631) State "smartmeter.0.129-129:199_130_3__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0	2021-01-14 06:45:34.593	debug	(8631) connected set to true
smartmeter.0	2021-01-14 06:45:34.592	debug	(8631) Set State 129-129:199_130_3__255.rawvalue = undefined
smartmeter.0	2021-01-14 06:45:34.589	debug	(8631) Create State 129-129:199_130_3__255.value
smartmeter.0	2021-01-14 06:45:34.586	debug	(8631) SCHEDULE NEXT RUN IN 300s
smartmeter.0	2021-01-14 06:45:34.585	debug	(8631) RESUME READING SERIALPORT
smartmeter.0	2021-01-14 06:45:34.584	debug	(8631) SET MESSAGE TIMEOUT TIMER: 120000
smartmeter.0	2021-01-14 06:45:34.582	debug	(8631) Create Channel 129-129:199_130_3__255 with name Manufacturer ID
smartmeter.0	2021-01-14 06:45:34.582	debug	(8631) 129-129:199.130.3*255: Manufacturer ID = EMH
smartmeter.0	2021-01-14 06:45:34.580	debug	(8631) STORE DATA
[... cut ...]
smartmeter.0	2021-01-14 06:45:34.217	debug	(8631) SERIALPORT OPEN
smartmeter.0	2021-01-14 06:45:34.196	debug	(8631) CREATE SERIALPORT: 9600 8 1 none
smartmeter.0	2021-01-14 06:45:34.194	debug	(8631) SmartmeterObis options: {"debug":2,"protocol":"SmlProtocol","transport":"SerialResponseTransport","requestInterval":300,"anotherQueryDelay":1000,"transportSerialPort":"/dev/ttyUSB0","transportS
smartmeter.0	2021-01-14 06:45:34.164	info	(8631) starting. Version 3.1.7 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.20.1, js-controller: 3.2.6-0```

Apollon77, 2021-01-14:
thx, 3.1.8 :-)

LaForge112, 2021-01-22:
Hi Apollon77, 

Still there in Version 3.1.9 ?
js controller 3.2.8

Need another Debug?

Greets LaFo

`

smartmeter.0 | 2021-01-22 18:12:20.983 | warn | (24226) State "smartmeter.0.1-0:76_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
smartmeter.0 | 2021-01-22 18:12:20.976 | warn | (24226) State "smartmeter.0.1-0:56_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-01-22 18:12:20.974 | warn | (24226) State "smartmeter.0.1-0:36_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-01-22 18:12:20.973 | warn | (24226) State "smartmeter.0.1-0:16_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-01-22 18:12:20.971 | warn | (24226) State "smartmeter.0.1-0:1_8_1__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-01-22 18:12:20.967 | warn | (24226) State "smartmeter.0.1-0:1_8_0__255.rawvalue" has no existing object, this might lead to an error in future versions

`

Apollon77, 2021-01-22:
should not. Please provide debug log

LaForge112, 2021-01-23:
Hey Apollon77, thank you for your reply.

Here is the debug log:

[smartmeter.log](https://github.com/Apollon77/ioBroker.smartmeter/files/5860186/smartmeter.log)

Thank you for your work :)

Apollon77, 2021-01-23:
Ääähhm ... starting. Version 3.0.10 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.20.1, js-controller: 3.2.8

Der Fix ist seit der 3.1.8/9 drin ... da solltest Du bitte mal updaten :-) Ist noch im Latest

Von daher ist doe Frage ob das log mit der 3.1.9 leer ist :-)

LaForge112, 2021-01-23:
Thats weird :)

[image]

Apollon77, 2021-01-23:
Pot Multihost and wrong host selected in admin?