Apollon77/ioBroker.smartmeter #180: Netzwerkgerät Daten auslesen

creecool, 2021-01-17:
**Describe the bug**  
Ich erhalte keine Daten über den Adapter.

Adapter Status: Verbunden mit smartmeter: falsch

Infos im IOB Log:
Connected tcp Socket 

No or too long answer from socket after last Request 

Der IOB kann eine Verbindung zur EDL Netzwerkschnittstelle herstellen. (Test via Telnet über Port 5000) 

Es handelt sich dabei um das EDL-Cockpit von co.met 

http://www.edl-cockpit.de

**Screenshots & Logfiles**  

So sieht der Datenstream des Gerätes aus. 

[image]

**Versions:**  
 - Adapter version: 3.1.6 und 3.1.8 getestet

Apollon77, 2021-01-18:
Bitte Adapter auf debug log stellen und senden.

Effektiv hätte ich erwartet das der Adapter die Daten liesst und erkennt.

creecool, 2021-01-19:
> 
> 
> Bitte Adapter auf debug log stellen und senden.
> 
> Effektiv hätte ich erwartet das der Adapter die Daten liesst und erkennt.

smartmeter.0 | 2021-01-19 23:27:19.646 | warn | (32028) No or too long answer from Socket after last request.
-- | -- | -- | --
smartmeter.0 | 2021-01-19 23:27:19.644 | info | (32028) Error: No or too long answer from Socket after last request.
smartmeter.0 | 2021-01-19 23:25:19.644 | info | (32028) CONNECTED TO TCP SOCKET
smartmeter.0 | 2021-01-19 23:20:19.640 | warn | (32028) No or too long answer from Socket after last request.
smartmeter.0 | 2021-01-19 23:20:19.636 | info | (32028) Error: No or too long answer from Socket after last request.
smartmeter.0 | 2021-01-19 23:18:19.644 | info | (32028) CONNECTED TO TCP SOCKET
smartmeter.0 | 2021-01-19 23:18:19.561 | info | (32028) starting. Version 3.1.8 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.20.1, js-controller: 3.1.6
smartmeter.0 | 2021-01-19 23:18:19.540 | info | (32028) Plugin sentry Sentry Plugin disabled for this process because sending of statistic data is disabled for the system

creecool, 2021-01-19:
> 
> 
> Bitte Adapter auf debug log stellen und senden.
> 
> Effektiv hätte ich erwartet das der Adapter die Daten liesst und erkennt.

Hi, reichen dir diese Infos?

Zeit	
	
			
smartmeter.0	2021-01-19 23:48:33.444	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:33.442	debug	(29307) ADD NEW DATA (6377 + NEW 799)
smartmeter.0	2021-01-19 23:48:33.411	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:33.409	debug	(29307) ADD NEW DATA (6279 + NEW 98)
smartmeter.0	2021-01-19 23:48:31.671	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:31.670	debug	(29307) ADD NEW DATA (5480 + NEW 799)
smartmeter.0	2021-01-19 23:48:31.625	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:31.625	debug	(29307) ADD NEW DATA (5382 + NEW 98)
smartmeter.0	2021-01-19 23:48:29.916	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:29.914	debug	(29307) ADD NEW DATA (4583 + NEW 799)
smartmeter.0	2021-01-19 23:48:29.884	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:29.881	debug	(29307) ADD NEW DATA (4485 + NEW 98)
smartmeter.0	2021-01-19 23:48:28.129	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:28.128	debug	(29307) ADD NEW DATA (3686 + NEW 799)
smartmeter.0	2021-01-19 23:48:28.096	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:28.095	debug	(29307) ADD NEW DATA (3588 + NEW 98)
smartmeter.0	2021-01-19 23:48:26.461	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:26.460	debug	(29307) ADD NEW DATA (2789 + NEW 799)
smartmeter.0	2021-01-19 23:48:26.428	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:26.427	debug	(29307) ADD NEW DATA (2691 + NEW 98)
smartmeter.0	2021-01-19 23:48:24.834	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:24.833	debug	(29307) ADD NEW DATA (1892 + NEW 799)
smartmeter.0	2021-01-19 23:48:24.800	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:24.800	debug	(29307) ADD NEW DATA (1794 + NEW 98)
smartmeter.0	2021-01-19 23:48:23.007	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:23.006	debug	(29307) ADD NEW DATA (995 + NEW 799)
smartmeter.0	2021-01-19 23:48:22.974	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:22.973	debug	(29307) ADD NEW DATA (897 + NEW 98)
smartmeter.0	2021-01-19 23:48:22.084	debug	(29307) connected set to false
smartmeter.0	2021-01-19 23:48:22.055	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:22.054	debug	(29307) ADD NEW DATA (98 + NEW 799)
smartmeter.0	2021-01-19 23:48:22.025	debug	(29307) MATCH-RESULT MESSAGE: "6d61633d30303a30383a44433a31353a38423a45460d0a626f6f743d310d0a66773d30322e30300d0a74696d653d313236303536373939320d0a7a70623d444530303030303030303030303030303030303030303
smartmeter.0	2021-01-19 23:48:22.021	debug	(29307) ADD NEW DATA (0 + NEW 98)
[... cut ...]
smartmeter.0	2021-01-19 23:47:24.224	debug	(26351) SCHEDULE NEXT RUN IN 300s
smartmeter.0	2021-01-19 23:47:24.223	debug	(26351) Transport Reset!! Restart = true
smartmeter.0	2021-01-19 23:47:24.223	debug	(26351) Error: No or too long answer from Socket after last request.
smartmeter.0	2021-01-19 23:47:24.222	warn	(26351) No or too long answer from Socket after last request.
smartmeter.0	2021-01-19 23:47:24.221	debug	(26351) Error: No or too long answer from Socket after last request.
smartmeter.0	2021-01-19 23:47:24.220	debug	(26351) MESSAGE TIMEOUT TRIGGERED

Apollon77, 2021-01-20:
Hey, es wird nicht 50 als port angegeben, sonderndu holst das log aus Admin und da werden Zeilen auf eine länge limitisert, am ende ist daher das log halb-helfend. Am besten immer ins Logfile unter /opt/iobroker/log/... schauen und genau das brauche ich auch.an sieht auf jeden Fall mal das irgendwelche Daten reinkommen.

Wie ist denn das -  werden die Daten gepusht auf dem tcp sobald es neue gibt oder bekommt jede neue connection ein Datenpaket.

Interressanterweise sieht es so aus als ob die Daten in hex kommen?

Wenn du eintelnet da hin machst, siehst Du dann die Daten die du oben im screenshot hast auch in der ausgabe?

creecool, 2021-01-20:
Wie ist denn das - werden die Daten gepusht auf dem tcp sobald es neue gibt oder bekommt jede neue connection ein Datenpaket.  --> Ja Daten kommen automatisch ca. jede Sekunde sobald einen Verbndung z.B. per Telenet aufbebaut wurde und offen bleibt. Es geht auch immer nur einen V

[iobroker.2021-01-20.log](https://github.com/Apollon77/ioBroker.smartmeter/files/5844866/iobroker.2021-01-20.log)

erbindung gleichzeitig. Es greift  keine weitere Schnitstelle/Skript zu wenn ich deinen Adapter aktiv habe. 

Interressanterweise sieht es so aus als ob die Daten in hex kommen? --> ja kommen in HEX an. 

Wenn du ein telnet da hin machst, siehst Du dann die Daten die du oben im screenshot hast auch in der ausgabe?

[telnet_log.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/5844881/telnet_log.txt)

creecool, 2021-01-23:
There are no options to change the data stream protocol for me.

Apollon77, 2021-01-23:
Ok, so I need to think about any maybe need to define a new "transport" where the data are converted

creecool, 2021-01-23:
I've already thought about it myself and currently it's going through some detours for me. If you want I would send you my script.
How can I get it to you? Wouldn't like to post it right here.

Apollon77, 2021-01-23:
I do a feature right now ... comes later or tomorrow

Apollon77, 2021-01-23:
Ok, please update from GitHub, version stay the same for now ... You should have a new option in admin interface of adapter to define the "encoding" ... set that to ascii or utf-8 (what is it??) and try please. send full log from disk (not admin!)

creecool, 2021-01-24:
> 
> 
> Ok, please update from GitHub, version stay the same for now ... You should have a new option in admin interface of adapter to define the "encoding" ... set that to ascii or utf-8 (what is it??) and try please. send full log from disk (not admin!)

HI, It´s working - TTTTHHHHAAANNNKKKSSS

[image]

LOG:

[LOG_Smartmether.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/5862433/LOG_Smartmether.txt)

Apollon77, 2021-01-24:
fixed in 3.2.0

closed 2021-01-24 by Apollon77