Apollon77/ioBroker.smartmeter #181: Stromzähler elster A1500-W141-521 optische Schnittstelle Verbindungsprobleme EN 62056-21

Fendt2000, 2021-01-18:
Hallo,
ich hab ein Problem mit der Kommunikation mit meinem Stromzähler vom Typ: elster A1500-W141-521.
vielleich liegt es auch nur an meinen Einstellungen. Laut Datenblatt kommuniziet er mit dem Protokoll nach EN 62056-2.
Schnittstelle und Lesekopf erkennt der Adapter, doch leider baut er keine verbindung auf.

Ich hab Zwei Einstellungen versucht:
Versuch 1: D0 Datenprotokoll (19:07 uhr)
[image]
Versuch 2: JSON Format (19:10 uhr) 

[image]

hier das Logfile dazu: 
[LOG_IOBroker.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/5832337/LOG_IOBroker.txt)

Wenn ich über das Programm "Emlog Testprogramm" von der Firma Weidmann mich mit dem IR-Lesekopf verbinde antwortet der Stromzähler auch.
[image]
die Übertragung geht nur bei 300 Baudrate 7E1 und dauert gute 3 Minuten 
Das Protokoll lautet: OBIS ASCII 

ich hoffe mit meinen Information könnt ihr mir weiter helfen.

mein Ziel ist es, den aktuellen Verbrauch im Zylus von max 5 s auszulesen, um eventuell große Verbraucher zu schalten um den Verbrauch zu minimieren.

ich hoffe es kann mir jemand weiterhelfen, sodass ich eine Verbindung aufbauen kann und die Übertragungsgeschwindigkeit zu erhöhen.

( ich bin ein Anfänger in sachen IOBroker)

Apollon77, 2021-01-18:
Naja du schreibst das das Abfragen 3 Minuten dauert ... und setzt ein message timeout von 1,5s ... passt net so ganz ;-))

Effkeit hast Du die Baudrate überschrieben mit 300 baud damit macht er ALLES mit der laaahmen baudrate. nimm den Wert mal raus ... dann sollte es schneller gehtn

Apollon77, 2021-01-18:
Und D0 ist korrekt

closed 2021-01-18 by Apollon77

Fendt2000, 2021-01-19:
Hallo Apollon77,
vielen Dank für deine Hilfe, hat funktioniert!
hab den Übertragungsintervall auf 20s gebracht.
bin sehr zufrieden :) der Adapter ist mega, 
Danke!!

Fendt2000, 2021-02-08:
Hallo Apollon77,
jetzt muss ich nochmals stören :) ich habe mir jetzt mal alle empfangenen Objekte angeschaut, aber leider bekomme ich keine aktuellen Werte von dem Stromzähler. Laut meinem Netzbetreiber brauch ich die Werte 31.7.0/51.7.0/71.7.0 um den aktuellen Stromverbrauch von jeder Phase zu bekommen.
Wie kann ich diese Werte über den Adapter erhalten?

Apollon77, 2021-02-08:
Es kommen die Werte die kommen ... da kann der Adapter nichts tun - oder musst Du einen bestimmten D0 Abfrage-Befehl absetzen? Was kommt denn raus beim Adapter an Daten?

Fendt2000, 2021-02-09:
ich hätte mir erst das Debug Log file anschauen müssen :( da steht drin das mein Zähler den Mode E braucht und ich soll mich an dich wenden :) 
aktuell bekomme ich 283 Werte.
hier mein Log File 
[LOG_IOBroker_210208.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/5952168/LOG_IOBroker_210208.txt)

kannst du mir dabei behilflich sein?

Apollon77, 2021-02-09:
Und in den 283 Werten ist nicht das drin was du suchst? ALso ja ich sehe da kein 31.xxx oder 51.xxxx oder so.

Die Meldung zu "Mode E" heisst nur das es Mode E Zähler gibt die quasi ein eigenes Datenformat zurückgeben können, da ist bei dir aber nicht der Fall.Dasist Standard. Aber auch im D0 ist kein 31./51/71 zeug drin. Am besten sende doch mal das Ergebnis hier an deinen netzbetreiber (vllrt einfach das ganze log) udnd ann muss der sagen was anders sein soll