Apollon77/ioBroker.smartmeter #186: Support for sending individual data to IR for Landis+Gyr E350 EDL21

kayser, 2021-01-22:
For integrating my smart meter I need to send data to the IR sensor. I have a Landis+Gyr E350 EDL21 and a IR reader/writer connected via USB to my raspi iobroker.

Is there a way to configure the request bytes I have to send to the meter?
I have to wake it up and then send my data request...

Here is a python example which works:

    ser = serial.Serial(serialPort, baud, serial.SEVENBITS, serial.PARITY_EVEN,serial.STOPBITS_TWO)

    sendtxta = "\x2F\x3F\x21\x0D\x0A"
    ser.write(sendtxta.encode())

    time.sleep(2.0)

    sendtxtb = "\x06\x30\x30\x30\x0D\x0A"
    ser.write(sendtxtb.encode())

    data = {}

    while True:
        response=ser.readline()
        #print(response)
        str_hex = " ".join(hex(n) for n in response)
        if (str_hex == "0x21 0xd 0xa"):
            ser.close()
            break

        if '('.encode() in response:
            str_split = response.split('('.encode())
            str_value = str_split[1].split(')'.encode())[0]
            data[str_split[0]] = str_value

kayser, 2021-01-22:
Landis+Gyr E350 EDL21 works! Here is the configuration:

[image]

add to supported devices as you please.

Apollon77, 2021-01-23:
The reacl characters to send would had been easiert ... your wakeup should be "/?!", or ?! And this is D0 standard and sent by the adapter automaticaly if not overwritten, After that you send "000" which normally is done when you enter a number in "anzahl wake up characters".
And please remove the mode override you have set there.

I also had a Landis&Gyr "Fernwärme" device and it works that way. If it is that also set" ersatz OBIS Medium" to "Wärme"

closed 2021-01-23 by Apollon77