Apollon77/ioBroker.smartmeter #191: warning in the logfile

AliBabaBinLaden, 2021-02-19:
after the update of the js-controler to 3.2.16 i have a lot of warnings at hthe logfile. What is the meaning of that messages and how can i fix it?

smartmeter.0 | 2021-02-19 20:49:18.897 | warn | (600) State "smartmeter.0.1-0:76_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
-- | -- | -- | --
smartmeter.0 | 2021-02-19 20:49:18.895 | warn | (600) State "smartmeter.0.1-0:56_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-02-19 20:49:18.889 | warn | (600) State "smartmeter.0.1-0:36_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-02-19 20:49:18.887 | warn | (600) State "smartmeter.0.1-0:16_7_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-02-19 20:49:18.884 | warn | (600) State "smartmeter.0.1-0:1_8_1__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-02-19 20:49:18.880 | warn | (600) State "smartmeter.0.1-0:1_8_0__255.rawvalue" has no existing object, this might lead to an error in future versions
smartmeter.0 | 2021-02-19 20:49:18.869 | info | (600) Received 11 values, 6 updated

Apollon77, 2021-02-19:
Also with 3.2.0 of smartmeter?

AliBabaBinLaden, 2021-02-20:
Yes smartmeter has the Version
3.2.0

Apollon77, 2021-02-21:
Please check this again - also updated on correct host? ( Select host in Admin on top). This needs to be fixed.

If you re really shure I need a full debug log please

AliBabaBinLaden, 2021-02-22:
ok. i found the problem. i updated the adapter only on the main system but not at the sub system. Now all´is fine.

closed 2021-02-22 by AliBabaBinLaden