Apollon77/ioBroker.smartmeter #192: Daten werden nicht erkannt (Adapter bleibt gelb)

mlapp, 2021-02-21:
Hallo , ich habe einen EBZ DD3 Zähler und versuche diesen mit einem IR-Schreib-Lesekopf auszulesen. 
Auf der Console des Raspi funktioniert das auch prima.

/EBZ5DD32R06ETA_107

1-0:0.0.0*255(1EBZ0100618430)
1-0:96.1.0*255(1EBZ0100618430)
1-0:1.8.0*255(003699.65305306*kWh)
1-0:2.8.0*255(004748.76702794*kWh)
1-0:16.7.0*255(001002.92*W)
1-0:36.7.0*255(000510.99*W)
1-0:56.7.0*255(000108.64*W)
1-0:76.7.0*255(000383.29*W)
1-0:96.5.0*255(001C0104)
0-0:96.8.0*255(010183FE)
!
/EBZ5DD32R06ETA_107

1-0:0.0.0*255(1EBZ0100618430)
1-0:96.1.0*255(1EBZ0100618430)
1-0:1.8.0*255(003699.65333221*kWh)
1-0:2.8.0*255(004748.76702794*kWh)
1-0:16.7.0*255(001004.92*W)
1-0:36.7.0*255(000515.28*W)
1-0:56.7.0*255(000108.86*W)
1-0:76.7.0*255(000380.78*W)
1-0:96.5.0*255(001C0104)
0-0:96.8.0*255(010183FF)
!

Der Zähler läuft mit 7,E,1 und wenn ich die einstelle kommen obige Daten auf der Console an (mit cat).
Gefühlt sind die Werte absolut brauchbar. Allerdings werden sie vom Adapter anscheinend nicht erkannt.
Dabei ist es egal ob ich Binärdaten, Ascii oder UTF-8 einstelle. 
Ist echt schade, hab heute nen halben Tag dran rumgebastelt, meine Raspi mit der Raspimatic auf pivCCU umzustellen, weil der direkt am Zählerschrank steht und meine Syno den Lesekopf nicht will.

ioBroker läuft jetzt im Multihost-Mode mit zwei Raspis als Slave (einer am Erzeugungszähler und der Zweite am Hauszähler)
Kann es evtl. daran liegen? Erkannt werden aber beide Slaves problemlos und der Erste funktioniert auch super.

Apollon77, 2021-02-21:
Naja wenn Du über binär oder Ascii "oder so nachdenst ... Bitte Einstellungen screenshot zeigen!
Du musst in jedem Fall "Seriell" nehmen ( denke am ehesten "nur lesen") und D0 als protokoll

mlapp, 2021-02-21:
Stimmt hast Recht.
Auf D0 wäre ich im Leben nicht gekommen, weil im Handbuch des Zählers von "Das Protokoll ist in SML Version 1.03 und 1.04 ausgeführt" gesprochen wird.
Habs auf D0 geändert und die Werte sind da. Absolut perfekt.
Dank dir für den Tipp :-)

closed 2021-02-21 by Apollon77