Apollon77/ioBroker.smartmeter #193: State "smartmeter.1.1-0:2_8_0__255.rawvalue" has no existing object, this might lead to an error in future versions

FISO291, 2021-02-25:
**Describe the bug**  
I get this error message continuously in the log.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to Log
2. /
3. /
4. /

**Expected behavior**  
No error message in log

**Screenshots & Logfiles**  
/

**Versions:**  
 - Adapter version: 3.2.0
 - JS-Controller version: 3.2.16
 - Node version: v10.22.0
 - Operating system: Linux

Apollon77, 2021-02-25:
did you updated the adapter really on the right host?? Is really 3.2.0 started?

FISO291, 2021-02-25:
Wir können auch in Deutsch weiterschreiben :)

Hab grad den Adapter (hab 2 Instanzen auf dem Slave Raspberry Pi laufen; steht im Keller beim Sicherungskasten. Der Master steht im EG) neu gestartet.
Hab eben auch mal den Slave gebootet, aber die Fehlermeldung bzw Warnung kommt weiterhin.

Am Iskra Stromzähler sind übrigens die Leseköpfe von Weidemann installiert.

Folgende Version ist bei mir auf dem Slave installiert:

Verfügbare Version:3.2.0
Installierte Version:3.2.0
Installierte Instanzen:2 

Ich kann dir auch leider nicht sagen seit Wann dieser Fehler (Version) auftritt.... Es werden nur 7 Tage lang Logs angelegt...

Apollon77, 2021-02-25:
Und im Log wenn smartmeter startet steht WIRKLICH 3.2.0?

Nicht das du das update nur auf dem master installiert hast ... Bei updates von slave adaptern musst du im Admin oben den host wählen!

FISO291, 2021-02-25:
starting. Version 3.1.5 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v10.23.0, js-controller: 3.2.16

Oh man, du hattest vollkommen recht. Daran hab ich überhaupt nicht gedacht, dachte immer, dass beide Systeme so aktualisiert werden. Wieder man was gelernt! 

Ich bin dir sehr dankbar Apollon77! Geht nun und dieser Issue kann somit geschlossen werden :)

Apollon77, 2021-02-25:
Cool, super das es "nur" das war :-)

closed 2021-02-25 by Apollon77