Apollon77/ioBroker.smartmeter #200: Support meters with DLMS encrypted data

james2810, 2021-03-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Support of meters of kind Sagemcom T210-D using DLMS protocol with encryption

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Native support of DLMS encrypted data (by providing required key as configuration parameter)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
--------

**Additional context**
Add any other context or screenshots about the feature request here.

Apollon77, 2021-04-11:
I need to see if that makes sense here .. in fact DLMS is more neanr to mbus ... but we will see

zloe, 2022-01-25:
I would also support this request. Alternatively the encryption could ofc. be done "outside" of this adapter. But having it all in one place "feels" better.

Apollon77, 2022-03-06:
Additional infos: https://github.com/Apollon77/ioBroker.smartmeter/issues/298#issuecomment-1060032938