Apollon77/ioBroker.smartmeter #206: (1729) Error: No or too long answer from Serial Device after last request.

dentro-git, 2021-04-08:
**Describe the bug**  
I have just installed the ELV USB-IEC Adapter and wanted to integrate it into my IOBroker.
It was not working even after many tries. Details below... I pretested the USB-IEC Head with a Windows Programm which checked writing and reading to the sensor itself and conection to the smart Meter afterwards, all good.

**Now the Linux and IOBroker Story which was not that positive :**

The Sensor is working and receiving its data throuw /dev/ttyUSB0
[image]

[image]

Configuration like mentioned in the Howto of the Smartmeter:
[image]

I also tried D0 as Protocol in fact of this information in the Datasheet: And I have the ODZ1 Version
[image]

**SML:**
After that test I added the Adapter in the IOBroker and configured it like this:
[image]

Logs like:
`
--
2021-04-08 13:07:04.133 - warn: smartmeter.0 (1729) No or too long answer from Serial Device after last request.
2021-04-08 13:07:04.135 - debug: smartmeter.0 (1729) Error: No or too long answer from Serial Device after last request.
2021-04-08 13:07:44.154 - warn: smartmeter.0 (1729) No or too long answer from Serial Device after last request.
2021-04-08 13:07:44.155 - debug: smartmeter.0 (1729) Error: No or too long answer from Serial Device after last request.
2021-04-08 13:07:50.866 - info: unifi.0 (1228) Update done
2021-04-08 13:08:24.173 - warn: smartmeter.0 (1729) No or too long answer from Serial Device after last request.
2021-04-08 13:08:24.174 - debug: smartmeter.0 (1729) Error: No or too long answer from Serial Device after last request.
2021-04-08 13:08:51.216 - info: unifi.0 (1228) Update done
2021-04-08 13:09:04.188 - warn: smartmeter.0 (1729) No or too long answer from Serial Device after last request.
2021-04-08 13:09:04.189 - debug: smartmeter.0 (1729) Error: No or too long answer from Serial Device after last request.
2021-04-08 13:09:21.327 - silly: smartmeter.0 (1729) States system redis pmessage io.messagebox.system.adapter.smartmeter.0/io.messagebox.system.adapter.smartmeter.0:{"command":"listUart","message":null,"from":"system.adapter.admin.0","callback":{"message":null,"id":32,"ack":false,"time":1617880161325},"_id":46594435}
2021-04-08 13:09:21.330 - debug: smartmeter.0 (1729) Message received = {"command":"listUart","message":null,"from":"system.adapter.admin.0","callback":{"message":null,"id":32,"ack":false,"time":1617880161325},"_id":46594435}
2021-04-08 13:09:21.592 - info: smartmeter.0 (1729) List of port: [{"manufacturer":"Silicon Labs","serialNumber":"8cc78ef24da0ea1198ce5bcbfe9faf31","pnpId":"usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_8cc78ef24da0ea1198ce5bcbfe9faf31-if00-port0","vendorId":"10c4","productId":"ea60","path":"/dev/ttyUSB0"},{"path":"/dev/ttyAMA0"}]
2021-04-08 13:09:21.594 - debug: smartmeter.0 (1729) sendTo "listUart" to system.adapter.admin.0 from system.adapter.smartmeter.0
2021-04-08 13:09:27.749 - silly: smartmeter.0 (1729) States system redis pmessage *.logging/system.adapter.admin.0.logging:{"val":false,"ack":true,"ts":1617880167735,"q":0,"from":"system.adapter.admin.0","lc":1617880167735}
2021-04-08 13:09:27.751 - debug: smartmeter.0 (1729) system.adapter.admin.0: logging false
2021-04-08 13:09:36.773 - silly: smartmeter.0 (1729) States system redis pmessage *.logging/system.adapter.admin.0.logging:{"val":true,"ack":true,"ts":1617880176766,"q":0,"from":"system.adapter.admin.0","lc":1617880176766}
2021-04-08 13:09:36.774 - debug: smartmeter.0 (1729) system.adapter.admin.0: logging true
2021-04-08 13:09:44.205 - warn: smartmeter.0 (1729) No or too long answer from Serial Device after last request.
2021-04-08 13:09:44.206 - debug: smartmeter.0 (1729) Error: No or too long answer from Serial Device after last request.

`

**D0**
[image]

`Log

smartmeter.0 | 2021-04-08 13:34:49.377 | debug | (1882) system.adapter.admin.0: logging true
-- | -- | -- | --
smartmeter.0 | 2021-04-08 13:34:49.376 | silly | (1882) States system redis pmessage *.logging/system.adapter.admin.0.logging:{"val":true,"ack":true,"ts":1617881689367,"q":0,"from":"system.adapter.admin.0","lc":1617881689367}
smartmeter.0 | 2021-04-08 13:34:22.855 | debug | (1882) Error: No or too long answer from Serial Device after last request.
smartmeter.0 | 2021-04-08 13:34:22.854 | warn | (1882) No or too long answer from Serial Device after last request.
smartmeter.0 | 2021-04-08 13:33:42.838 | debug | (1882) Error: No or too long answer from Serial Device after last request.
smartmeter.0 | 2021-04-08 13:33:42.836 | warn | (1882) No or too long answer from Serial Device after last request.
smartmeter.0 | 2021-04-08 13:33:02.813 | debug | (1882) Error: No or too long answer from Serial Device after last request.
smartmeter.0 | 2021-04-08 13:33:02.812 | warn | (1882) No or too long answer from Serial Device after last request.
smartmeter.0 | 2021-04-08 13:32:56.753 | debug | (1882) system.adapter.admin.0: logging false
smartmeter.0 | 2021-04-08 13:32:56.751 | silly | (1882) States system redis pmessage *.logging/system.adapter.admin.0.logging:{"val":false,"ack":true,"ts":1617881576738,"q":0,"from":"system.adapter.admin.0","lc":1617881576738}
smartmeter.0 | 2021-04-08 13:32:52.823 | debug | (1882) connected set to false
smartmeter.0 | 2021-04-08 13:32:52.796 | silly | (1882) States system redis pmessage system.adapter.smartmeter.0.logLevel/system.adapter.smartmeter.0.logLevel:{"val":"silly","ack":true,"ts":1617881572785,"q":0,"from":"system.adapter.smartmeter.0","
smartmeter.0 | 2021-04-08 13:32:52.778 | debug | (1882) SmartmeterObis options: {"debug":0,"protocol":"D0Protocol","transport":"SerialResponseTransport","requestInterval":"30","anotherQueryDelay":"1000","transportSerialPort":"/dev/serial/by-id/usb-
smartmeter.0 | 2021-04-08 13:32:52.742 | info | (1882) starting. Version 3.2.0 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.22.1, js-controller: 3.2.16
smartmeter.0 | 2021-04-08 13:32:52.532 | debug | (1882) Plugin sentry Initialize Plugin (enabled=true)
smartmeter.0 | 2021-04-08 13:32:52.295 | debug | (1882) statesDB connected
smartmeter.0 | 2021-04-08 13:32:52.294 | debug | (1882) States connected to redis: 127.0.0.1:9000
smartmeter.0 | 2021-04-08 13:32:52.283 | debug | (1882) States create User PubSub Client
smartmeter.0 | 2021-04-08 13:32:52.282 | debug | (1882) States create System PubSub Client
smartmeter.0 | 2021-04-08 13:32:52.272 | debug | (1882) Redis States: Use Redis connection: 127.0.0.1:9000
smartmeter.0 | 2021-04-08 13:32:52.269 | debug | (1882) objectDB connected
smartmeter.0 | 2021-04-08 13:32:52.263 | silly | (1882) redis psubscribe cfg.o.system.user.*
smartmeter.0 | 2021-04-08 13:32:52.260 | debug | (1882) Objects connected to redis: 127.0.0.1:9001
smartmeter.0 | 2021-04-08 13:32:52.234 | debug | (1882) Objects client initialize lua scripts
smartmeter.0 | 2021-04-08 13:32:52.233 | debug | (1882) Objects create User PubSub Client
smartmeter.0 | 2021-04-08 13:32:52.232 | debug | (1882) Objects create System PubSub Client
smartmeter.0 | 2021-04-08 13:32:52.229 | debug | (1882) Objects client ready ... initialize now
smartmeter.0 | 2021-04-08 13:32:52.185 | debug | (1882) Redis Objects: Use Redis connection: 127.0.0.1:9001
`

**Versions:**  
 - Adapter version: 3.2.0
 - JS-Controller version:  3.2.16
 - Node version: v12.22.1
 - Operating system: Linux raspberrypi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux

[eBZ_Betriebsanleitung_DD3.pdf](https://github.com/Apollon77/ioBroker.smartmeter/files/6278292/eBZ_Betriebsanleitung_DD3.pdf)

Apollon77, 2021-04-08:
Ok, the data format is D0 in any case. did you had that "cat" command on ssh running while trying with adapter? If yes this has blocked the port ... please do not do that. honestly I somehow miss logging lines, but that setting is looking good in general. Can you please try again

dentro-git, 2021-04-09:
@Apollon77 : Thanks a lot for the hint. I've tried so often with the wrong data format that I messed something up when testing D0. I deleted the smartmeter adapter, restarted the Pi and tried it from scratch. And it directly worked with this settings. Just if someone has the same issue. That is more or less "factory default", Now I just have to get the code from my Providor to get detailed information over the Bus.

[image]

closed 2021-04-09 by dentro-git