Apollon77/ioBroker.smartmeter #208: Error Parsing SML message

joeflower, 2021-04-13:
Grundsätzlich funktioniert alles, aber alle paar Minuten kommt diese Fehlermeldung. Ich habe verucht den String zu zerlegen, leider ohne großen Erfolg:

`2021-04-13 10:38:45.583  - info:  smartmeter.0 (1492) Error: Error while parsing SML message: Error:  Unknown TL-Field 0x1b for OctetString [Offset: 131]!: Message:  1b1b1b1b01010101760508dd12b3620062007265000001017601010765425a4444330b090145425a01000e0740010163c3f500760508dd12b46200620072650000070177010b090145425a01000e07400172620165017a2f067677078181c78203ff010101010445425a0177070100000009ff010101010b090145421b1b1b1b1b1b1b1b
2021-04-13 10:38:45.585  - warn:  smartmeter.0 (1492) Error while parsing SML message: Error: Unknown  TL-Field 0x1b for OctetString [Offset: 131]!: Message:  1b1b1b1b01010101760508dd12b3620062007265000001017601010765425a4444330b090145425a01000e0740010163c3f500760508dd12b46200620072650000070177010b090145425a01000e07400172620165017a2f067677078181c78203ff010101010445425a0177070100000009ff010101010b090145421b1b1b1b1b1b1b1b
2021-04-13 10:39:15.596  - info: smartmeter.0 (1492) CONNECTED TO TCP SOCKET
2021-04-13 10:39:16.348  - info: smartmeter.0 (1492) Received 6 values, 0 updated`

ODER anderes Beispiel:

`2021-04-13 10:36:34.338 - info: smartmeter.0 (1492) Received 6 values, 0 updated
2021-04-13 10:37:08.116 - info: smartmeter.0 (1492) Error: Error while parsing SML message: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 44. Received 45: Message: 1b1b1b1b01010101760508dd1073620062007265000001017601010765425a4444330b09011b1b1b1b1b1b1b1b
2021-04-13 10:37:08.118 - warn: smartmeter.0 (1492) Error while parsing SML message: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 44. Received 45: Message: 1b1b1b1b01010101760508dd1073620062007265000001017601010765425a4444330b09011b1b1b1b1b1b1b1b`

Mein Aufbau: Volkszähler Kopf an ESP8266 mit ESPeasy und mit ser2net an iobroker.smartmeter
Mein Stromzähler eBZ DDR3 2R06 DTA - SMZ1

joeflower, 2021-04-13:
und noch eine kürzere:
`smartmeter.0 | 2021-04-13 10:55:42.377 | warn | (1492) Error while parsing SML message: Error: Wrong TL-Field (1) for  Unsigned!: Message: 1b1b1b1b01010101760508dd2a89621b1b1b1b1b1b1b1b

smartmeter.0 | 2021-04-13 10:55:42.375 | info | (1492) Error: Error while parsing SML message: Error: Wrong TL-Field  (1) for Unsigned!: Message:  1b1b1b1b01010101760508dd2a89621b1b1b1b1b1b1b1b`

Apollon77, 2021-04-13:
Dann Bitte Sitz des Kopfes prüfen und das die Datenübertragung passt (ggf USB Kabelweiter weg von den Stromkabels legen und ggf aktiven Switch zur Masse/Spassungsverbesserung nutzen.

Das erste (lange) sieht wie ein gekipptes Bit und damit falsche Daten aus - kann durch das kommen was ich eben geschrieben habe.

Diese mega kurzen Nachrichten sind komplett ungültig - keine ahnung wie die entstehen können. Normalerweise sind Die Datenpakete vom Zähler immer die gleiche länge.

Also da passt was nicht.

Zu ESPEasy und ser2net kann ich nichts sagen, ggf. im Forum melden Bitte

joeflower, 2021-04-13:
OK Danke, werde deine Hinweise befolgen und weiter beobachten.

closed 2021-04-15 by joeflower