Apollon77/ioBroker.smartmeter #210: No or too long answer from Serial Device after last request

jkbiz, 2021-04-21:
Hallo zusammen,

Ich habe ioBroker und eine Instanz smartmeter auf einen RaspberryPi installiert. Daran angeschlossen ist ein Weidmann IR Schreib/Lesekopf USB (Optokopf).

Bei dem Stromzähler handelt es sich um ein Easymeter Q3M.
Im Handbuch steht bei den Datenschnittstellen dazu folgendes: 
(MSB- und INFO-DSS)
Telegramm Protokoll nach SML 1.04
Baudrate 9600 Bit/s
Byte Format (8,N,1)
 
Den Adapter habe ich so konfiguriert. Hier fehlt leider SML 1.0.4
[image]

Es hat bisher gut funktioniert (07.04 - 19.04), doch seit gestern bekomme ich keine Werte in den ioBroker übermittelt. Ich habe mir den debug aktiviert und habe festgestellt, dass die Ausgabe nicht endet und damit Meldung immer länger wird. 
Hinter der ersten Abgefragten Ausgabe wird immer wieder was angehängt (gelb, grün, cyan, ilia, rot). Ich habe das logfile mal komprimiert hochgeladen. Ich würde mich über eine Lösung freuen.

[image]

2021-04-21 13:14:28.910  - [32minfo[39m: smartmeter.0 (1596) starting. Version 3.2.0 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.22.0, js-controller: 3.2.16
2021-04-21 13:14:29.140  - [34mdebug[39m: smartmeter.0 (1596) SmartmeterObis options: {"debug":2,"protocol":"SmlProtocol","transport":"SerialResponseTransport","requestInterval":"300","anotherQueryDelay":"1000","transportSerialPort":"/dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_D309V00U-if00-port0","transportSerialBaudrate":9600,"transportSerialDataBits":8,"transportSerialStopBits":1,"transportSerialParity":"none","transportSerialMessageTimeout":250000,"protocolSmlIgnoreInvalidCRC":true,"obisFallbackMedium":1}
2021-04-21 13:14:29.165  - [34mdebug[39m: smartmeter.0 (1596) CREATE SERIALPORT: 9600 8 1 none
2021-04-21 13:14:29.248  - [34mdebug[39m: smartmeter.0 (1596) SERIALPORT OPEN
2021-04-21 13:14:29.374  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (0 + NEW 2)
2021-04-21 13:14:29.385  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085" -> null
2021-04-21 13:14:29.393  - [34mdebug[39m: smartmeter.0 (1596) SET MESSAGE TIMEOUT TIMER: 250000
2021-04-21 13:14:29.459  - [34mdebug[39m: smartmeter.0 (1596) connected set to false
2021-04-21 13:14:29.757  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (2 + NEW 4)
2021-04-21 13:14:29.763  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085fefefe1e" -> null
2021-04-21 13:14:29.779  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (6 + NEW 13)
2021-04-21 13:14:29.783  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085fefefe1e343c28d83d3fffbff8fef08fca" -> null
2021-04-21 13:14:29.799  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (19 + NEW 13)
2021-04-21 13:14:29.803  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085fefefe1e343c28d83d3fffbff8fef08fca3780f380fb5ef02040c3feff08" -> null
2021-04-21 13:14:29.823  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (32 + NEW 14)
2021-04-21 13:14:29.827  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085fefefe1e343c28d83d3fffbff8fef08fca3780f380fb5ef02040c3feff0845fbff3c8fcfdbde3cfc6a9de6f0" -> null
2021-04-21 13:14:29.843  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (46 + NEW 25)
2021-04-21 13:14:29.848  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085fefefe1e343c28d83d3fffbff8fef08fca3780f380fb5ef02040c3feff0845fbff3c8fcfdbde3cfc6a9de6f0fefee0e03e7c80ffa1fed7f8fef08fdb3780f380fbfe78d0e0" -> null
2021-04-21 13:14:29.862  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (71 + NEW 14)
......
2021-04-21 13:18:39.090  - [34mdebug[39m: smartmeter.0 (1596) ADD NEW DATA (51009 + NEW 67)
2021-04-21 13:18:39.137  - [34mdebug[39m: smartmeter.0 (1596) MATCH-RESULT MESSAGE: "0085fefefe1e343c28d83d3fffbff8fef08fca3780f380fb5ef02040c3feff0845fbff3c8fcfdbde3cfc6a9de6f0fefee0e03e7c80ffa1fed7f8fef08fdb3780f380fbfe78d0e0......fb8900f380fb630281fee0fe7b103020e0feb4fef636ccba" -> null
2021-04-21 13:18:39.399  - [34mdebug[39m: smartmeter.0 (1596) MESSAGE TIMEOUT TRIGGERED
2021-04-21 13:18:39.403  - [34mdebug[39m: smartmeter.0 (1596) Error: No or too long answer from Serial Device after last request.
2021-04-21 13:18:39.414  - [33mwarn[39m: smartmeter.0 (1596) No or too long answer from Serial Device after last request.
2021-04-21 13:18:39.416  - [34mdebug[39m: smartmeter.0 (1596) Error: No or too long answer from Serial Device after last request.
2021-04-21 13:18:39.419  - [34mdebug[39m: smartmeter.0 (1596) Transport Reset!! Restart = true
2021-04-21 13:18:39.440  - [34mdebug[39m: smartmeter.0 (1596) SERIALPORT CLOSE
2021-04-21 13:18:39.446  - [34mdebug[39m: smartmeter.0 (1596) SCHEDULE NEXT RUN IN 300s
2021-04-21 13:19:33.059  - [32minfo[39m: smartmeter.0 (1596) Got terminate signal TERMINATE_YOURSELF
2021-04-21 13:19:33.083  - [32minfo[39m: smartmeter.0 (1596) terminating
2021-04-21 13:19:33.091  - [34mdebug[39m: smartmeter.0 (1596) Plugin sentry destroyed
2021-04-21 13:19:33.095  - [32minfo[39m: smartmeter.0 (1596) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2021-04-21 13:19:51.360  - [32minfo[39m: smartmeter.0 (1636) starting. Version 3.2.0 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v12.22.0, js-controller: 3.2.16

Besten Dank und viele Grüße

[iobroker.2021-04-21.zip](https://github.com/Apollon77/ioBroker.smartmeter/files/6350446/iobroker.2021-04-21.zip)

jkbiz, 2021-04-21:
Hier ist noch ein wenig Ausgabe via Screen und minicom

[image]

[image]

Apollon77, 2021-04-21:
Naja eine wartezeit von nur 1s ist etwas kurz. Bedenke es ist serielle Kommunikation!! Bitte hochsetzen und nochmal versuchen, mindestens 5-10s

Apollon77, 2021-04-21:
PS: ALso im Standard ist die serielle antwortzeit 120s btw ... und aus gutem grund :-)

jkbiz, 2021-04-21:
Danke für den Tipp, dass habe ich mir aber auch schon gedacht und 250 Sekunden eingetragen um das logfile zu erstellen.. leider habe ich hier das gleiche Fehlerbild.

Apollon77, 2021-04-22:
Dann bitte neues Debug log. Generell auch SItz des kopfes prüfen, der binaäe auszug oben lässt keine SML "Marker" erkennen. Da muss 1b 1b 1b 1b ... kommen am Start und Ende einer Nachricht ... Das fehlt

jkbiz, 2021-04-22:
Danke. Ich habe den Lesekopf umgesetzt und jetzt funktioniert es.

closed 2021-04-22 by Apollon77