Apollon77/ioBroker.smartmeter #214: SmartMUC ZDUE sagemcom fka. Dr.Neuhaus

LocalNic, 2021-04-30:
I try to connect to a SmartMuc that is installed by my EnergyProvider and the MUC is connected to all primary Energy sources. 
This Device is collecting data in so called Mirrors or Data Collectors. 
here a Link to the Device: https://www.sagemcom.com/V02/fileadmin/user_upload/Energy/Dr.Neuhaus/Support/ZDUE-MUC/Doc_communs/ZDUE-MUC_Anwenderhandbuch_V2_5.pdf

I also have access to the Device via the Propriatary software and also tried to connect to the device by a Java Source. 
I tried the jSQL by OpenMuc and got a connection but wasn't able to understand the requested SML Fragments. 
https://www.openmuc.org/sml/

The reason why I reach out to you is that this Device is available by IP via a User (Ethernet) interface and habe Authentication ( User and Password ) configured. So I suspect that it might be easy to adapt your Module for IObroker to support credentials as well. 

Hope this is a nobrainer for you and not huge effort. 

Thanks in advance

Apollon77, 2021-04-30:
I need way more information on how to get the SML ... if it is then official SML thats no issue, but I need exact info on getting it.

The  document states on page 193 that SML can be requested via TCP ... but no info on any user/password

Can you maybe access that directly with telnet?

LocalNic, 2021-04-30:
I have used jSML in the past to get directly OBIS numbers from the current Consumption direct from the muc. 
I had tried to create my own openhab so to say .. I can provide you the source lines that I used to utilize the jSML Library and this works. 

```
		OctetString user = new OctetString(this.user.getBytes());
		OctetString password = new OctetString(this.password.getBytes());

		OctetString clientID = new OctetString(hexToByteArray(this.clientId));
		OctetString serverID = new OctetString(hexToByteArray(this.serverId));
		
		SML_PublicOpenReq openRequest = new SML_PublicOpenReq(null, clientID, reqFileId, null, user, password, null);
		SML_MessageBody smlMessageBody = new SML_MessageBody(SML_MessageBody.OpenRequest, openRequest);
		SML_Message openRequestMessage = new SML_Message(transactionId, groupNum, abortOnError, smlMessageBody);
```
in the library https://www.openmuc.org/sml/download/ of jsml is also the source for the SML_PublicOpenReq class. 
So I suppose it is just an additional field in the SML OpenReqest 
I'm not so deep into programming, and the rest of the code looks like a mess but It was hard for me to find out the the Strings need to be transfered into the OctetString. ;)

LocalNic, 2021-04-30:
That is more as I have expected ;) I can definatly support you by testing it with my Device. 
As Far as I know were those MUC the standard in a Pilot of a German Energy Provider so there might be a lot more that have this device already build in their Houses. 

The MUC collects from external RF-M-BUS meters the data and collects it in an internal memory. 
To use these Collectors you just do a request to provide the actual Value of the meter or accessing the internal Collection. 

As I am not so familiar with the OBIS Numbers I did a reingeneering of a TCPSession where I gather the actual Values of the meters.
I just realize that the MUC here only provides Values for Power, I think that for Water and Gas the Batteries run dry. 

I can provide you with the relevant IDs, as I said I have allready read out the actual values. 
But to be completely open I just have no clue how this SML and OBIS stuff works. :) 

Thanks for your support ...

Apollon77, 2021-04-30:
In fact it could help to see a SML message created by that jSML library so that I can compare it to what my creates - and alsoa copy of an response SML, then I can do a dry run too and most likely it is easier then. YOu can send that to me as email to iobroker@fischer-ka.de

Freighter, 2021-12-28:
Sorry, I had lost this topic out of scope and just get faced by that topic again, may I be able to send you the SML trace anyway? 
As we discussed, it might be an easy thing to implement.

Apollon77, 2021-12-28:
Have fun ... Attach here if possible ...

LocalNic, 2022-01-05:
Here the export from the SML Req/Res hope it shows the Info where to put the User and Password ;) 

[sml with auth.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/7814910/sml.with.auth.txt)

LocalNic, 2022-01-13:
Hi Apollon77, 
I found some old code where I try to get some Data from the MUC .. using the jSML from openmuc Project 

I allready know a lot about my numbers so here is what I did in the past : 
in Pseudo code 
1.  generate a connection 
        sml_tConnection = sml_tSAP.connectTo(InetAddress.getByAddress(openmucBytes), port, 50);
2.  create a sml message with Open and close request
3. in between I tried two things 
a . generating a getListReq where the Client ID is my MAC and Server ID is the server string from my "EMH"
    		SmlGetListReq getListRequest = new SmlGetListReq(clientID, serverID, user, password, null);
		smlMessageBody = new SmlMessageBody(EMessageBody.GET_LIST_REQUEST.id(), getListRequest);
and 
b. generating a getProfileListRequest knowing the collector for measurement entries
            	OctetString smlTreePath = new OctetString(hexToByteArray("8181c78617FF"));
		SmlTreePath treePath = new SmlTreePath(new OctetString[] { smlTreePath });
		SmlGetProfileListReq getProfileListRequest = new SmlGetProfileListReq( serverID, user, password, null, null, null, treePath, null, null);
		smlMessageBody = new SmlMessageBody(EMessageBody.GET_PROFILE_LIST_REQUEST.id(), getProfileListRequest);

the List REsponse is the actual value and the Profile REsponse is a list of the last 30 min with secondly measurements 

I would be happy if you can use this to integrate into your code that a lot of people can ask to get direct access tu the MUC they have allready in their homes :)

Freighter, 2022-09-06:
just to my curiousity, is there any update ? 
Like providing username and passwords with that Module ?

Apollon77, 2022-09-06:
Did not had any time for it so far. Thumbs up in first post help to see how many users are interested ...