Apollon77/ioBroker.smartmeter #224: Holley DT541 nicht mehr erreichbar seit letztem Update

Schnietz, 2021-05-21:
Hi,
Habe einen IR Lesekopf an einem Holley DTZ541 seit dem 21.04.
Seit dem 10.05 also dem letzten Update(3.2.1) zeigt dieser nichts mehr an.

Apollon77, 2021-05-21:
Und was sagt dein Log? Die 3.2.1 hat an sich keine relevanten Änderungen

Schnietz, 2021-05-22:
Ich bin leider noch nicht so bewandert das Log vollständig aus IObroker zu bekommen. Sobald ich es runterlade fehlen die Einträge für den Smartmeter.(Adapter auf Slave installiert) Was im Log steht ist folgendes
(653) No or too long answer from Serial Device after last request.

Apollon77, 2021-05-22:
Dann bitte Sitz des kopfes prüfen und so weiter, ggf mal rechner rebooten. Dann kommen keine Daten an. Ggf genauer ins log schauen. Liegt auch auf Platte unter /opt/iobroker/log

closed 2021-05-22 by Apollon77