Apollon77/ioBroker.smartmeter #245: ED300L  varying error messages about TL-Fields

drnicolas, 2021-08-04:
I have a ED300L meter with optical interface.
The IR-head is connected to an ESPEasy which sends the data via Ser2Net to iobroker

In iobroker, I get varying error messages similar to this:
2021-08-04 14:32:08.483 - warn: smartmeter.0 (1548) Error while parsing SML message: Error: Wrong TL-Field (1) for Unsigned!: Message:
1b1b1b1b010101017607000a09b78e02620062007263010176010107000a09032f560b0901454d48000057c66e010163de73007607000a09b78e03620062007263070177010b0901454d48000057c66e070100620affff72620165090356ef7a77078181c78203ff0101010104454d480177070100000009ff010101010b0901454d48000057c66e0177070100010800ff6401018201621e52ff5600244de5f10177070100020800ff6401018201621e52ff5600000006db0177070100010801ff0101621e52ff5600244de5f10177070100020801ff0101621e52ff5600000006db0177070100010802ff0101621e52ff5600000000000177070100020802ff1b1b1b1b01010101

The SML-messages looks fine for me, but it is not decoded. Changing to ASCII, Base64 or UTF does not change anything

What can I do?

Apollon77, 2021-08-04:
Is the error always the same? is the message length the same? This question wants to make sure that not the head is placed incorrectly or the usb cable is not good protected and so bit's are swapping

drnicolas, 2021-08-05:
Der Fehler betrifft wohl jedes Paket, aber der TL Fehler scheint zu variieren.
Auf dem oszilloskop sieht alles super aus was am ESP8266 ankommt.

Dr. med. Nicolas Nickisch

Apollon77, 2021-08-05:
Wenn der "TL Fehler" immer ein anderer bzw woanders ist dann ist es (aus bisheriger erfahrung) entweder das Kabel und damit der Übertragsungsweg oder ein nicht korrekt sitzender kopf. Versuch den mal "mikrometerweise" (scherz) zu verschieben ...

drnicolas, 2021-08-05:
Ja, das könnte die Ursache sein. Aktuell ist das Kabel ungeschirmt; wenn auch die Signale gut aussehen. Ein Problem könnte auch der ESP selber sein. Ich kriege unter ESPEasy einfach die HWSerial nicht zum Laufen. Signale bekomme ich überhaupt nur mit SoftSerial – dann aber auf demselben Pin.

Apollon77, 2021-08-05:
Ungeschirmt neben "Stromführendem Equipment" blöde Idee :-) Ich tope das einfach hier und da bits kippen. Und da die Library die hier genutzt wird nicht nur wie andere einzelne Stellen sucht und rausschnippelt sondern das SML komplett parst fliegt das dann auf die nase ...

AM besten mach die möglichen Fehlerfälle weg und meld dich nochmal wenn es dann immer noch nicht tut

closed 2021-08-05 by Apollon77

drnicolas, 2021-08-05:
Für einen ersten Test einfach ein Telefon-Flachkabel geschanppt. Ca. 4m. lang ☹
Schande. Ich werde es probieren