Apollon77/ioBroker.smartmeter #247: Ram usage is quite high !

xeno27, 2021-08-16:
Hi,

I have an Rasperry Pi 4 (8GB ram) and two smartmeter adapter via USB installed (Hichi USB). I installed the smartmeter plugin for IoBroker and see that the ram usage increase more and more. i have to restart the Adapter to fix this.

At the moment every adapter took more than 200Mb of ram !
i have the version 3.2.1

[image]

Apollon77, 2021-08-16:
Hm ... I never experienced that. mine just need 70MB per instance.

So we need to find out whats up on your system and why it is that way.

So lets start with logs. is the 30s interval working and always reporing data? Or are ther ein between some intervals that do not do that?

xeno27, 2021-08-16:
i think there are some intervals there are not reporting data (seen in the log).
should i need to set this interval to an higher number ?
[iobroker.2021-08-16.log](https://github.com/Apollon77/ioBroker.smartmeter/files/6991428/iobroker.2021-08-16.log)

Apollon77, 2021-08-16:
Hm ... for me it looks good. each instance has two lines per minute ... thats as it should be for 30s delay (becasue next 30s starts after a mesage was received) ...

Hm ... interssting. Please try to use "10s" data query interval ... (for one instance) ... is ram use lower then?

xeno27, 2021-08-16:
Ok i will try (but it will take about one week to see that it increase).

xeno27, 2021-08-17:
So after less than 24H (about 18H) we have more than 100Mb ram usage (witch is quite much ?)
smartmeter 0 has 10sec interval
smartmeter 1 has 30sec interval
[image]

Apollon77, 2021-08-17:
Strange ... this do not happen on my system. WHich node.js version is used?

xeno27, 2021-08-17:
I use 12.22.1
[image]

Apollon77, 2021-08-17:
Could be a question if an upgrade to nodejs 14 helps ... in fact the adapter handles a lot of data with all the serial stuff and handling, but also all data structures are normally free'd by adapter (at least as I'm aware). SO it is up to the nodejs process/GarbageCollector when he free's up stuff ...

xeno27, 2021-08-17:
Ok that make Sense, is there an Good Upgrade Guide ? Wich i can Trust ?

Apollon77, 2021-08-17:
A german one in ioBroker forum https://forum.iobroker.net/topic/44566/how-to-node-js-f%C3%BCr-iobroker-richtig-updaten-2021-edition

xeno27, 2021-08-24:
after updating to NODE.JS: V14.17.5 and I-broker 5.1.23 the ram usage is after one week around 70Mb. 
Thanks !

closed 2021-08-24 by xeno27

Apollon77, 2021-08-24:
Interesting ... thank you