Apollon77/ioBroker.smartmeter #253: Error while parsing D0 content / Error: Invalid Obis

Quarksack, 2021-09-17:
Hallo Apollon,

ich haben ein  Problem.
Der Adapter bombt mir das Log mit _Error while parsing D0 content: ignore complete part Error: Invalid Obis_ immer wieder zu.

Den Sitz des Lesekopfs habe ich geprüft. Der Sitzt soweit ich das sehen kann korrekt. 

Ausgelesen wird ein Siemens TD3511. Es kommen auch Daten an, aber die Fehler machen mich schon stutzig.

Anbei ein Debuglog des Adapters und ein Screenshot der Adaptereinstellung.

Viele Grüße und Danke für deine Mühen.

Quarksack

[log.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/7183342/log.txt)
[settings](https://user-images.githubusercontent.com/90892930/133738805-b8b11ecd-cf5c-413d-88ae-963fc20251d7.JPG)

_Originally posted by @Quarksack in https://github.com/Apollon77/ioBroker.smartmeter/issues/32#issuecomment-921549167_

Apollon77, 2021-09-17:
As said it is not the adapzter it are simply invalid OBIS IDs returned in the data

Invalid Obis String C.60.5.1

Quarksack, 2021-09-17:
Ok. Can I disable the messages or get the OBIS IDs filtered out? So that the log is not cluttered?

Apollon77, 2021-09-17:
You could set loglevel to error

closed 2022-03-27 by stale[bot]