Apollon77/ioBroker.smartmeter #254: fehlerhafte SerialPort-Zuordnung bei mehr als einer Instanz

Xander0823, 2021-09-17:
Hallo,

ich nutze Version 3.2.1 mit 2 USB-IR-Adaptern für jeweils einen ISKRA MT631.

Für jeden Zähler läuft eine eigene Instanz (0 und 1).

Bei der Auswahl der seriellen Schnittstelle über das DropDown-Menü überschreibt die Auswahl in der 2. Instanz den Wert der 1. Instanz. Beide Instanzen zeigen dann die gleichen Messwerte.

[image]

**Zähler 1 gespeichert**

[image]

**Zähler 2 gespeichert**

[image]

**Zähler 1 nun mit falscher Schnittstelle**

Wählt man "Benutzerdefinierter Name" und vergibt für Instanz 0 "/dev/ttyUSB0" und für Instanz 1 "/dev/ttyUSB1" funktioniert alles wie gewünscht,

Gruß Alex

Apollon77, 2021-09-17:
Das ist komisch und habe ich So noch nicht gesehen. Bitte Beide Instanzen auf Debug Log stellen und dann Admin aufrufen. Da ist eine Logzeile mit den seriellen ports. Bitte von beiden instanzen hier posten

Xander0823, 2021-09-17:
[image]

[image]

Apollon77, 2021-09-17:
Und jeztz bitte die Ausgabe von "ls -la /dev/serial/by-id/"

Xander0823, 2021-09-18:
total 0
drwxr-xr-x 2 root root 60 Sep 17 18:30 .
drwxr-xr-x 4 root root 80 Sep 17 18:30 ..
lrwxrwxrwx 1 root root 13 Sep 17 18:30 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0

Xander0823, 2021-09-18:
**ls -la /dev/serial/by-path/**
total 0
drwxr-xr-x 2 root root 80 Sep 17 18:30 .
drwxr-xr-x 4 root root 80 Sep 17 18:30 ..
lrwxrwxrwx 1 root root 13 Sep 17 18:30 platform-3f980000.usb-usb-0:1.1.2:1.0-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 Sep 17 18:30 platform-3f980000.usb-usb-0:1.1.3:1.0-port0 -> ../../ttyUSB1

Xander0823, 2021-09-18:
[https://wiki.fhem.de/wiki/Mehrere_USB-Ger%C3%A4te_einbinden#Problemf.C3.A4lle:_by-id](url)

Apollon77, 2021-09-18:
ALso liegt es am seriaal device ... 

Meins sieht so aus (auch gleiche köpfe)

```
ingof@io3:~$ ls -la /dev/serial/by-id/
total 0
drwxr-xr-x 2 root root 100 Aug 13 18:55 .
drwxr-xr-x 4 root root  80 Aug 13 18:55 ..
lrwxrwxrwx 1 root root  13 Aug 13 18:55 usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root  13 Aug 13 18:55 usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_010625FA-if00-port0 -> ../../ttyUSB1
lrwxrwxrwx 1 root root  13 Aug 13 18:55 usb-Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller_01065F3B-if00-port0 -> ../../ttyUSB2
```

Du hast einen Workaround und dafür gibts das "Custom" auch.

closed 2021-09-18 by Apollon77