Apollon77/ioBroker.smartmeter #279: iobroker smartmeter adapter installation error

MSal-RPI4, 2022-01-03:
**Describe the bug**  
After upgrade of the rpi4b from buster to bullseye the update of the entire iobroker system went well ... except reactivating the smartmeter adapter installation, which operated successfully for more than a year.  After several attempts of reactivating the instance the adapter was uninstalled and reinstallation does not seem to be possible anymore (always finishing with errors) ?!

**Screenshots & Logfiles**  
Logfile attached.

**Versions:**  
 - Adapter version: 3.2.1
 - JS-Controller version: 3.3.22
 - Node version: 14.18.2
 - Operating system: RPI Debian 11

[2022-01-03T13_31_05_679Z-debug.log](https://github.com/Apollon77/ioBroker.smartmeter/files/7802529/2022-01-03T13_31_05_679Z-debug.log)

Apollon77, 2022-01-03:
It seems that the installation of the serialport library is failing somehow ... this is a better topic for the iobroker forum to get support ... it is not a failure of the adapter.

In fact the question what exactly does that "buster to bullseye" mean ... I assume also nodejs version got updated?

In fact you could try to remove /opt/iobroker/node_modules/@serialport and /opt/iobroker/node_modules/serialport completely and try again then to install the adapter, else I need more logs and details

MSal-RPI4, 2022-01-04:
Thanks for your quick reply.  Sorry, "Buster to Bullseye" update was
referring to the RPI Debian OS update and indeed nodejs also has been
updated to the most recent version.

Removed ***@***.***
<https://github.com/serialport> dir (the other did not exist), but
unfortunately no change ... adapter still does not install.  Additional
error logging has been pasted below.  However, not sure what other logging
I could provide ?!

Thanks and bets regards
Michael

$ iobroker add smartmeter --host iobroker-master

NPM version: 6.14.15

npm install ***@***.*** --loglevel error --prefix
"/opt/iobroker" (System call)

make: g++: No such file or directory

make: *** [bindings.target.mk:112:
Release/obj.target/bindings/src/serialport.o] Error 127

gyp ERR! build error

gyp ERR! stack Error: `make` failed with exit code: 2gyp ERR! stack at
ChildProcess.onExit
(/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)gyp
ERR! stack at ChildProcess.emit (events.js:400:28)gyp ERR! stack at
Process.ChildProcess._handle.onexit (internal/child_process.js:282:12)gyp
ERR! System Linux 5.10.87-v7l+

gyp ERR! command "/usr/bin/node"
"/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js"
"rebuild"gyp ERR! cwd ***@***.***/bindingsgyp
ERR! node -v v14.18.2gyp ERR! node-gyp -v v5.1.0gyp ERR! not ok

npm ERR! code ELIFECYCLE

npm ERR! errno 1

npm ERR! @***@***.*** install: `prebuild-install --tag-prefix
@serialport/bindings@ || node-gyp rebuild`

npm ERR! Exit status 1npm ERR! npm ERR! Failed at the
@***@***.*** install script.npm ERR! This is probably not a
problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:npm ERR!
/home/iobroker/.npm/_logs/2022-01-03T23_34_54_767Z-debug.log

host.iobroker-master Cannot install ***@***.***: 1
ERROR: Process exited with code 25

MSal-RPI4, 2022-01-04:
> Thanks and best regards

Please find below some additional information about the system setup of the
rpi4.

Thanks and best regards
Michael

***@***.***:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
***@***.***:~ $ nodejs -v && node -v && npm -v
v14.18.2
v14.18.2
6.14.15

MSal-RPI4, 2022-01-04:
Hi,

wie gesagt bitte ins Forum damit. Ich bin kein Linux Experte.

Ein "iobroker fix" sollte das alles fixen, weil jetzt findet er icht mal 
den g++ compiler ... wenn das nicht tut müssen andere Leute ran die sich 
mit Linux auskennen

closed 2022-02-20 by Apollon77