Apollon77/ioBroker.smartmeter #280: Can't get Elster AS1440 work with smartmeter

janory, 2022-01-06:
### Problem
I can't get my `Elster AS1440` work with `smartmeter`.
The iobroker log gives a `No or too long answer from Serial Device after last request.` warning and the `Connected to device or service` status remains red on the UI.

I am using an [IR Schreib/Lesekopf USB](https://shop.weidmann-elektronik.de/index.php?page=product&info=24) which is working fine with my other (Landis & Gyr) electricity meter and it's also working with the `Elster AS1440`, if I configure my Optokopf from the console with the following command:
```bash
stty -F /dev/ttyUSB0 300 -parodd cs7 -cstopb parenb -ixoff -crtscts -hupcl -ixon -opost -onlcr -isig -icanon -iexten -echo -echoe -echoctl -echoke
```

and trigger the electricity meter like this: `echo -ne '/?!\r\n' > /dev/ttyUSB0`, I am getting some values:
```bash
pi@raspberrypi:~ $ cat /dev/ttyUSB0
/ELS4\@V9.36
1-1:F.F(00000000)
1-1:F.F.2(00000060)
1-1:0.0.0(20400260)
1-1:0.0.1(00000000)
1-1:0.0.2(00000000)
1-1:0.9.1(203703)
1-1:0.9.2(060122)
1-1:0.3.0(1000.00)
1-1:0.3.3(500.00)
1-1:0.1.0(16)
1-1:1.2.0(0110.34)
1-1:1.2.0*16(0110.34)
1-1:1.2.0*15(0099.05)
1-1:1.2.0*14(0087.32)
1-1:1.2.0*13(0080.01)
1-1:1.2.0*12(0077.20)
1-1:1.2.0*11(0070.67)
1-1:1.2.0*10(0063.75)
1-1:1.2.0*09(0055.54)
1-1:1.2.0*08(0052.95)
1-1:1.2.0*07(0049.65)
1-1:1.2.0*06(0046.36)
1-1:1.2.0*05(0037.59)
1-1:1.2.0*04(0028.37)
1-1:1.2.0*03(0018.29)
1-1:1.2.0*02(0009.69)
1-1:1.2.1(0108.29)
1-1:1.2.1*16(0108.29)
1-1:1.2.1*15(0097.00)
1-1:1.2.1*14(0086.57)
1-1:1.2.1*13(0079.26)
1-1:1.2.1*12(0076.79)
1-1:1.2.1*11(0070.26)
1-1:1.2.1*10(0063.34)
1-1:1.2.1*09(0055.13)
1-1:1.2.1*08(0052.75)
1-1:1.2.1*07(0049.49)
1-1:1.2.1*06(0046.24)
...
```

### Adapter settings
Based on [volkszaehler](https://wiki.volkszaehler.org/hardware/channels/meters/power/edl-ehz/elster_as1440#elster_as1440) configured like this:
[image]

Apollon77, 2022-01-07:
Remove that what you put into the Sign on command because we already send the default call and with this addition you break it ... also remove thebaudrate changeover overwrite ... best remove all custom stuff beside the serial device settings AND set the transfer to "bidirectional" (because as you say IF you send X then you receive Y" ... ;-) so "read only" can not work

closed 2022-01-07 by Apollon77

janory, 2022-01-09:
@Apollon77 worked like a charm!
Thanks for the quick help and your work on ioBroker. 💪 
I've sent you a small donation. 🎉  😉

Apollon77, 2022-01-09:
Thank you very much and happy smart homing