Apollon77/ioBroker.smartmeter #292: SerialPort is not a constructor

arteck, 2022-02-21:
[image]

version 3.3.1

TbsJah, 2022-02-21:
same for me

Ilovegym66, 2022-02-21:
same here without ser2net, Raspi Zero with Node 14 and NPM 6.14.16

Apollon77, 2022-02-21:
fixed in upcoming 3.3.2 https://github.com/Apollon77/ioBroker.smartmeter/actions/runs/1874984749

Apollon77, 2022-02-21:
Can you please verify that also the "port search" works?

TbsJah, 2022-02-21:
new version it is not working for me
path is missing

[image]

Apollon77, 2022-02-21:
Ok, 3.3.3 ... next try

closed 2022-02-21 by Apollon77

TbsJah, 2022-02-21:
great job - now works like clockwork

amg666, 2022-02-21:
Master-Slave system, iob 4.0.15 and node.js 14.19.0.
3.3.3 works well, thank you
`<html><body>

host.iobroker | 2022-02-21 11:27:32.665 | info | iobroker exit 0
-- | -- | -- | --
host.iobroker | 2022-02-21 11:27:31.636 | info | iobroker  upload [0] smartmeter.admin  /opt/iobroker/node_modules/iobroker.smartmeter/admin/words.js words.js  application/javascript
host.iobroker | 2022-02-21 11:27:31.628 | info | iobroker upload [1] smartmeter.admin /opt/iobroker/node_modules/iobroker.smartmeter/admin/tooltip.css tooltip.css text/css
host.iobroker | 2022-02-21 11:27:31.619 | info | iobroker  upload [2] smartmeter.admin  /opt/iobroker/node_modules/iobroker.smartmeter/admin/smartmeter.png  smartmeter.png image/png
host.iobroker | 2022-02-21 11:27:31.605 | info | iobroker  upload [3] smartmeter.admin  /opt/iobroker/node_modules/iobroker.smartmeter/admin/smartmeter.jpg  smartmeter.jpg image/jpeg
host.iobroker | 2022-02-21 11:27:31.592 | info | iobroker  upload [4] smartmeter.admin  /opt/iobroker/node_modules/iobroker.smartmeter/admin/questionmark.png  questionmark.png image/png
host.iobroker | 2022-02-21 11:27:31.407 | info | iobroker upload [5] smartmeter.admin /opt/iobroker/node_modules/iobroker.smartmeter/admin/index.html index.html text/html
host.iobroker | 2022-02-21 11:27:31.220 | info | iobroker 168 packages are looking for funding  run `npm fund` for details
host.iobroker | 2022-02-21 11:27:17.796 | info | iobroker + iobroker.smartmeter@3.3.3removed 2 packages and updated 2 packages in 48.752s
host.iobroker | 2022-02-21 11:26:27.623 | info | iobroker NPM version: 6.14.16Installing iobroker.smartmeter@3.3.3... (System call)
host.iobroker | 2022-02-21 11:26:26.987 | info | iobroker Update smartmeter from @3.3.1 to @3.3.3
host.iobroker | 2022-02-21 11:26:25.028 | info | iobroker upgrade smartmeter@3.3.3
host.iobroker | 2022-02-21 11:25:29.332 | info | Updating repository "beta" under "http://download.iobroker.net/sources-dist-latest.json"

</body>
</html>`