Apollon77/ioBroker.smartmeter #298: Support DLMS encrypted with GUEK (no auth)

Idefix1977, 2022-03-02:
Do you plan to add support for:
Support for CustomerIFC P1 at Sagecom T210-D, Sagecom S210, Kaifa M110, Kaifa MA309 which are used in Austria. The IFC is an MBUS Push telegram which is send every 5 seconds.
There are MBUS Slave USB Serial Adapters for a few dollars available. The Baudrate is 2400Baud.
The data is send as an encrypted (by a GUEK (global unicast encryption key), authentication is disabled) DLMS/COSEM frame. DLMS-Securitylevel = SecuritySuite0

OpenSource Decrypter: (Python)
https://github.com/ric-geek/DLMS_APDU-Encrypter-Decrypter

OpenSource DLMS Translator:
Gurux GXDLMSDirector

If not, do you mind, if i try to implement it myself based on your adapter?

Thanks,
Elias.

Apollon77, 2022-03-03:
duplicate #200

Apollon77, 2022-03-03:
@Idefix1977 The topics are the following:
* First of all (and here "time" is a constraint) I did not had time to reimplement the decryption part into Javascript, so yes this could help alot
* Then the second issue is that this adapter completely bases on OBIS data structures ... but DLMS is more mbus based, so data just have an order by the OBIS data are missing, so we would need to add a mapping

Additionally there is https://github.com/lvogt/ioBroker.wireless-mbus in the meantime where I asked several times if someone can try if maybe this adapter already works for it ... maybe you can try it ... because then this is more "The way to go")

closed 2022-03-03 by Apollon77

Idefix1977, 2022-03-06:
@Apollon77  Thanks for the duplicate-hint ...seems that there are already other people interested in this issue, whats great.
Yes, time is also a constriant for me but i want so solve this issue even if it will take months :-)
The WMBus is something different. Thus a WMBus enabled device/meter is needed. Then you can use a WMBus Stick to receive this data. But in austria all smartmeters just have a physical wired MBUS connection (called user API P0). 
This IFC sends a MBUS Frame every 5 seconds, which is a push notification and can easily be reveived with a 5-10USD MBUS USB Slave Receiver. The MBUS Frame can easily be decoded. Then you will get an encrypted DLMS/COESM Frame. After decryption you will have a DLMS Frame which is the application layer of the communication stack. When you parse this frame, you will receive a collection of OBIS object which represent the menu entries of the smart meter. For all this there are several open source tools which i already got working for my smart meter. I attached you a decoded XML showing the received OBIS. I think it would fit nicely into your adapter because you already have the logic for receivin/reading data from a serial port and you already handle OBIS data. So we just have to add decryption and interpretation of DLMS as application layer.
[image]

Apollon77, 2022-03-06:
Then we "just" need to find a way to parse that in JavaScript ...

tomsn1978, 2024-01-26:
> @Apollon77 感谢您的重复提示...似乎已经有其他人对这个问题感兴趣了，太好了。是的，时间对我来说也是一个因素，但我想解决这个问题，即使需要几个月的时间：-） WMBus 是不同的。因此，需要一个支持 WMBus 的设备/仪表。然后，您可以使用 WMBus Stick 来接收此数据。但在奥地利，所有智能电表都只有一个物理有线MBUS连接（称为用户API P0）。该 IFC 每 5 秒发送一次 MBUS 帧，这是一个推送通知，可以使用 5-10USD MBUS USB 从接收器轻松显示。MBUS帧可以很容易地解码。然后，您将获得一个加密的 DLMS/COESM 帧。解密后，您将拥有一个DLMS帧，它是通信堆栈的应用层。当您解析此帧时，您将收到一组 OBIS 对象，这些对象代表智能电表的菜单项。对于所有这些，我已经为我的智能电表提供了几个开源工具。我附上了一个解码的 XML，显示了收到的 OBIS。我认为它非常适合您的适配器，因为您已经拥有从串行端口接收/读取数据的逻辑，并且您已经处理了 OBIS 数据。因此，我们只需要添加DLMS的解密和解释作为应用层即可。 [image] [ [image] ](https://user-images.githubusercontent.com/24613777/156940783-2519fa57-65a1-46ea-8e3c-3bb204c4240f.gif) [ ](https://user-images.githubusercontent.com/24613777/156940783-2519fa57-65a1-46ea-8e3c-3bb204c4240f.gif)

Hi I am interested in your XML translation tool can I get it