Apollon77/ioBroker.smartmeter #306: Holley DTZ251 - trotz Pin nur 4 Werte und keine Vorzeichen

Tarzipan, 2022-04-17:
Hi,

ich habe einen Holley DTZ241 verbaut bekommen. Habe den Pin eingegeben, die Einstellung "Info on" aktiviert und die Pinabfrage deaktiviert. Im Standardfall bekomme ich aber erstmal nur 4 Werte über den Adapter angezeigt.

[image]

Eigentlich würde das soweit reichen aber der Zähler gibt anscheinend keine Stromrichtung zurück. Ich habe eine kleine PV-Anlage und ich hätte eigentlich ein negativen Wert bei einer Einspeisung erwartet aber leider macht er das nicht.

Nach etwas suchen bin ich über folgenden Eintrag gestolpert.
https://forum.fhem.de/index.php/topic,51948.msg1172755.html?PHPSESSID=dtpv0oqp1m3vl7kt4urrifjeg1#msg1172755

Anscheinend hat der User dort das gleiche Problem und Holley gibt an, dass die Stromrichtung in einer Nachricht codiert ist.

Ist es möglich diese Stromrichtung auszulesen und als Objekt auszugeben?

Vielen Dank.

Apollon77, 2022-04-17:
Der Adapter stellt alle Daten zur verfügung die im SML geliefert werden. Bitte mal Adapter auf Debug Loglevel stellen dann siehst Du genau was im geparsten SML drin ist und genau das existiert in den States ... Sag was fehlt und ich prüfe es gern mit dem Debug log ... aber sonst gehe ich erstmal davon aus das alles da ist was an Daten kommt

Tarzipan, 2022-04-19:
Hi,

hier ist mein Log. Es geht um smartmeter.2

2022-04-19 08:42:56.089 - info: host.raspberrypi "system.adapter.smartmeter.2" enabled
--
2022-04-19 08:42:56.160 - info: host.raspberrypi instance system.adapter.smartmeter.2 started with pid 5758
2022-04-19 08:42:58.664 - info: smartmeter.1 (899) Received 10 values, 4 updated
2022-04-19 08:42:59.245 - debug: smartmeter.2 (5758) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-04-19 08:42:59.308 - debug: smartmeter.2 (5758) Objects client ready ... initialize now
2022-04-19 08:42:59.312 - debug: smartmeter.2 (5758) Objects create System PubSub Client
2022-04-19 08:42:59.315 - debug: smartmeter.2 (5758) Objects create User PubSub Client
2022-04-19 08:42:59.344 - debug: smartmeter.2 (5758) Objects client initialize lua scripts
2022-04-19 08:42:59.356 - debug: smartmeter.2 (5758) Objects connected to redis: 127.0.0.1:9001
2022-04-19 08:42:59.393 - debug: smartmeter.2 (5758) Redis States: Use Redis connection: 127.0.0.1:9000
2022-04-19 08:42:59.406 - debug: smartmeter.2 (5758) States create System PubSub Client
2022-04-19 08:42:59.408 - debug: smartmeter.2 (5758) States create User PubSub Client
2022-04-19 08:42:59.425 - debug: smartmeter.2 (5758) States connected to redis: 127.0.0.1:9000
2022-04-19 08:42:59.476 - debug: smartmeter.2 (5758) Plugin sentry Initialize Plugin (enabled=true)
2022-04-19 08:43:00.027 - info: smartmeter.2 (5758) starting. Version 3.3.4 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v14.19.1, js-controller: 4.0.21
2022-04-19 08:43:00.082 - debug: smartmeter.2 (5758) SmartmeterObis options: {"debug":2,"protocol":"SmlProtocol","transport":"SerialResponseTransport","requestInterval":"5","anotherQueryDelay":"1000","transportSerialPort":"/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0037-if00-port0","transportSerialBaudrate":9600,"transportSerialMessageTimeout":null,"protocolSmlIgnoreInvalidCRC":false}
2022-04-19 08:43:00.087 - debug: smartmeter.2 (5758) CREATE SERIALPORT: 9600 8 1 none
2022-04-19 08:43:00.100 - debug: smartmeter.2 (5758) SERIALPORT OPEN
2022-04-19 08:43:00.129 - debug: smartmeter.2 (5758) SET MESSAGE TIMEOUT TIMER: 120000
2022-04-19 08:43:00.144 - debug: smartmeter.2 (5758) connected set to false
2022-04-19 08:43:00.567 - debug: smartmeter.2 (5758) ADD NEW DATA (0 + NEW 1)
2022-04-19 08:43:00.571 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b" -> null
2022-04-19 08:43:00.574 - debug: smartmeter.2 (5758) ADD NEW DATA (1 + NEW 7)
2022-04-19 08:43:00.577 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b01010101" -> null
2022-04-19 08:43:00.580 - debug: smartmeter.2 (5758) ADD NEW DATA (8 + NEW 6)
2022-04-19 08:43:00.582 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760400000162" -> null
2022-04-19 08:43:00.586 - debug: smartmeter.2 (5758) ADD NEW DATA (14 + NEW 4)
2022-04-19 08:43:00.587 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072" -> null
2022-04-19 08:43:00.592 - debug: smartmeter.2 (5758) ADD NEW DATA (18 + NEW 6)
2022-04-19 08:43:00.594 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176" -> null
2022-04-19 08:43:00.598 - debug: smartmeter.2 (5758) ADD NEW DATA (24 + NEW 5)
2022-04-19 08:43:00.599 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b01010101760400000162006200726500000101760101070000" -> null
2022-04-19 08:43:00.603 - debug: smartmeter.2 (5758) ADD NEW DATA (29 + NEW 6)
2022-04-19 08:43:00.604 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a" -> null
2022-04-19 08:43:00.608 - debug: smartmeter.2 (5758) ADD NEW DATA (35 + NEW 4)
2022-04-19 08:43:00.610 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a01484c59" -> null
2022-04-19 08:43:00.613 - debug: smartmeter.2 (5758) ADD NEW DATA (39 + NEW 5)
2022-04-19 08:43:00.615 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a01484c5902000e7387" -> null
2022-04-19 08:43:00.619 - debug: smartmeter.2 (5758) ADD NEW DATA (44 + NEW 4)
2022-04-19 08:43:00.620 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a01484c5902000e738772620165" -> null
2022-04-19 08:43:00.624 - debug: smartmeter.2 (5758) ADD NEW DATA (48 + NEW 6)
2022-04-19 08:43:00.625 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a01484c5902000e73877262016500139dc50163" -> null
2022-04-19 08:43:00.629 - debug: smartmeter.2 (5758) ADD NEW DATA (54 + NEW 5)
2022-04-19 08:43:00.630 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a01484c5902000e73877262016500139dc50163a071007604" -> null
2022-04-19 08:43:00.634 - debug: smartmeter.2 (5758) ADD NEW DATA (59 + NEW 4)
2022-04-19 08:43:00.636 - debug: smartmeter.2 (5758) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000139dc50b0a01484c5902000e73877262016500139dc50163a07100760400000262" -> null
2022-04-19 08:43:00.639 - debug: smartmeter.2 (5758) ADD NEW DATA (63 + NEW 5)
[... cut ...]
2022-04-19 08:43:04.645 - debug: smartmeter.2 (5758) Plugin sentry destroyed
2022-04-19 08:43:04.647 - info: smartmeter.2 (5758) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2022-04-19 08:43:04.667 - debug: smartmeter.2 (5758) connected set to false
2022-04-19 08:43:05.224 - info: host.raspberrypi instance system.adapter.smartmeter.2 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)

Der Wert in "1-0:16.7.0*255" mit 425 ist schon Einspeisung

Ich hoffe das hilft weiter.

mfg

Apollon77, 2022-04-19:
Naja die Frage isto b es DIR weiterhilft ... weil Du siehst hier genau welche Daten vom gerät kommen. Sag mir was falsch im Adapter ist und was anders sein muss. Ich weiss nicht woran ich irgendwas erkennen könnte - was davon passt denn wie zu dem von FHEM zusammen?

Tarzipan, 2022-04-19:
In dem FHEM Forum schreibt der User, dass im "... die Stromrichtung im Statuswort des Zählerstands im bit 11 steckt...."
und hat die entsprechende Stelle in den Rohdaten markiert.
Die Frage ist, wo bzw. ob die Daten auch in dem Adapter vorkommen. (Das kann ich, weil ich es nicht programmiert habe, erstmal nicht erkennen). Wenn diese Stromrichtung mitgeliefert wird, wäre halt die Frage ob man das dann als extra Objekt ausgeben kann, damit ich im Nachgang damit was steuern kann. Natürlich könnte man auch einfach in dem Objekt 16.7.0 den Wert negativ machen.

Vielen dank.

Apollon77, 2022-04-19:
Naja ok damit wäre der Feature request neben dem "value" auch das Statusword als State bereitzustellen ... Rausparsen musst Du es dann immer noch selbst (mit nem kleinen Javascript-chen oder so) weil das Statuswort meines Wissens nach nicht "Zählerhersteller-übergreifend definiert ist" - alsokann Bit 11 bei jedem was anders bedeuten.

Allerdings scheint ein Status nur zb bei 1-0:1.8.0255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 83.735 kWh zu stehen. Wäre hier 1d4904 ... also Binär 11 10101 0010 01000 00100
In dem wäre das 11. Bit 0 ... macht das sinn?

Vllt nochmal ein Debug wenn Strom kommt und er was einspeist?

Tarzipan, 2022-04-19:
Ich glaube das ist die Stelle.
Ich habe jetzt Bezug und habe, wie der aus dem FHEM Forum, den Wert "1c0104".

Ich würde nochmal bis morgen abwarten und schauen ob dieser Statuswert sich wieder auf "1d4904" ändert aber wenn er das tut, wie würde ich denn da mit einem JS script dran gehen können?
Oder wäre es vielleicht für dich möglich, den Adapter um ein Feature zu erweitern, dass wenn jemand solch einen Zähler hat der Adapter bei "1d4904 " ein true und bei "1c0104" ein false in ein Einspeisungs-Objekt setzt? Oder vielleicht einfach der 16.7.0 Wert negiert wird?

2022-04-19 18:52:45.090 - info: host.raspberrypi "system.adapter.smartmeter.2" enabled
2022-04-19 18:52:45.161 - info: host.raspberrypi instance system.adapter.smartmeter.2 started with pid 900
2022-04-19 18:52:47.370 - info: smartmeter.1 (808) Received 10 values, 5 updated
2022-04-19 18:52:48.269 - debug: smartmeter.2 (900) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-04-19 18:52:48.338 - debug: smartmeter.2 (900) Objects client ready ... initialize now
2022-04-19 18:52:48.342 - debug: smartmeter.2 (900) Objects create System PubSub Client
2022-04-19 18:52:48.344 - debug: smartmeter.2 (900) Objects create User PubSub Client
2022-04-19 18:52:48.374 - debug: smartmeter.2 (900) Objects client initialize lua scripts
2022-04-19 18:52:48.386 - debug: smartmeter.2 (900) Objects connected to redis: 127.0.0.1:9001
2022-04-19 18:52:48.422 - debug: smartmeter.2 (900) Redis States: Use Redis connection: 127.0.0.1:9000
2022-04-19 18:52:48.435 - debug: smartmeter.2 (900) States create System PubSub Client
2022-04-19 18:52:48.437 - debug: smartmeter.2 (900) States create User PubSub Client
2022-04-19 18:52:48.453 - debug: smartmeter.2 (900) States connected to redis: 127.0.0.1:9000
2022-04-19 18:52:48.505 - debug: smartmeter.2 (900) Plugin sentry Initialize Plugin (enabled=true)
2022-04-19 18:52:49.092 - info: smartmeter.2 (900) starting. Version 3.3.4 in /opt/iobroker/node_modules/iobroker.smartmeter, node: v14.19.1, js-controller: 4.0.21
2022-04-19 18:52:49.143 - debug: smartmeter.2 (900) SmartmeterObis options: {"debug":2,"protocol":"SmlProtocol","transport":"SerialResponseTransport","requestInterval":"5","anotherQueryDelay":"1000","transportSerialPort":"/dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0037-if00-port0","transportSerialBaudrate":9600,"transportSerialMessageTimeout":null,"protocolSmlIgnoreInvalidCRC":false}
2022-04-19 18:52:49.148 - debug: smartmeter.2 (900) CREATE SERIALPORT: 9600 8 1 none
2022-04-19 18:52:49.161 - debug: smartmeter.2 (900) SERIALPORT OPEN
2022-04-19 18:52:49.189 - debug: smartmeter.2 (900) SET MESSAGE TIMEOUT TIMER: 120000
2022-04-19 18:52:49.205 - debug: smartmeter.2 (900) connected set to false
2022-04-19 18:52:49.793 - debug: smartmeter.2 (900) ADD NEW DATA (0 + NEW 3)
2022-04-19 18:52:49.798 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b" -> null
2022-04-19 18:52:49.802 - debug: smartmeter.2 (900) ADD NEW DATA (3 + NEW 9)
2022-04-19 18:52:49.805 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000" -> null
2022-04-19 18:52:49.810 - debug: smartmeter.2 (900) ADD NEW DATA (12 + NEW 6)
2022-04-19 18:52:49.812 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072" -> null
2022-04-19 18:52:49.817 - debug: smartmeter.2 (900) ADD NEW DATA (18 + NEW 6)
2022-04-19 18:52:49.819 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176" -> null
2022-04-19 18:52:49.824 - debug: smartmeter.2 (900) ADD NEW DATA (24 + NEW 7)
2022-04-19 18:52:49.826 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b010101017604000001620062007265000001017601010700000014" -> null
2022-04-19 18:52:49.831 - debug: smartmeter.2 (900) ADD NEW DATA (31 + NEW 7)
2022-04-19 18:52:49.833 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000142cb30b0a01484c" -> null
2022-04-19 18:52:49.838 - debug: smartmeter.2 (900) ADD NEW DATA (38 + NEW 6)
2022-04-19 18:52:49.840 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000142cb30b0a01484c5902000e7387" -> null
2022-04-19 18:52:49.845 - debug: smartmeter.2 (900) ADD NEW DATA (44 + NEW 6)
2022-04-19 18:52:49.847 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000142cb30b0a01484c5902000e7387726201650014" -> null
2022-04-19 18:52:49.852 - debug: smartmeter.2 (900) ADD NEW DATA (50 + NEW 6)
2022-04-19 18:52:49.854 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000142cb30b0a01484c5902000e73877262016500142cb3016393b1" -> null
2022-04-19 18:52:49.859 - debug: smartmeter.2 (900) ADD NEW DATA (56 + NEW 7)
2022-04-19 18:52:49.861 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000142cb30b0a01484c5902000e73877262016500142cb3016393b100760400000262" -> null
2022-04-19 18:52:49.865 - debug: smartmeter.2 (900) ADD NEW DATA (63 + NEW 6)
2022-04-19 18:52:49.867 - debug: smartmeter.2 (900) MATCH-RESULT MESSAGE: "1b1b1b1b0101010176040000016200620072650000010176010107000000142cb30b0a01484c5902000e73877262016500142cb3016393b100760400000262006200726500" -> null
2022-04-19 18:52:49.870 - debug: smartmeter.2 (900) ADD NEW DATA (69 + NEW 6)
[... cut ...]
2022-04-19 18:52:52.936 - info: smartmeter.2 (900) terminating
2022-04-19 18:52:52.938 - debug: smartmeter.2 (900) Plugin sentry destroyed
2022-04-19 18:52:52.940 - info: smartmeter.2 (900) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2022-04-19 18:52:52.953 - debug: smartmeter.2 (900) connected set to false
2022-04-19 18:52:53.527 - info: host.raspberrypi instance system.adapter.smartmeter.2 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2022-04-19 18:52:58.302 - info: smartmeter.1 (808) Received 10 values, 3 updated

Apollon77, 2022-04-20:
* 1d4904 ... also Binär 0111010100 1 00100000100
* 1c0104 ...  also Binär 1110000000 0 0100000100
naja ok es ist Bit 11 ... aber von links (so rum zählt man an sich nicht, aber egal)

Tarzipan, 2022-04-22:
ah okay, das macht dann Sinn.

Tarzipan, 2022-07-16:
Ist denn eigentlich geplant, dass diese Erweiterung umgesetzt wird?

Apollon77, 2022-07-17:
Naja, wäre es sonst mit "enhancement" getaggt wenn es nicht angedacht wäre?

Tarzipan, 2022-07-18:
Wahrscheinlich nicht. :)

Wollte aus gegebenen Anlass aber vorsichtig fragen ob daran gerade was gemacht wird.
Ich würde das ja selbst umsetzen, aber ich hab noch keinen ioBroker Adapter gemacht und keine Ahnung wie ich das Testweise auf mein Pi bekomme, geschweige von dem Releasekram auf github.
Falls du einen Tester brauchst, könnte ich das natürlich machen.

Vielen Dank im Voraus