Apollon77/ioBroker.smartmeter #307: Sync data fetch timestamp with other instance of this adapter

Loenne83, 2022-04-17:
**Problem:**
When using two instances of this adapter, they might be not simultaneously fetch data, even if this would be interesting, because they both get power data. When calculating differences of these values, the time difference leads to inaccurate data.

**Possible solution:**
Get an option in the instance setting, where the instance to be synced to can be named (i.e. "smartmeter.1"). Alternatively, defining a cron timeplan for each instance would do the job either, I suppose.

Apollon77, 2022-04-17:
Syncing between instances will not be that easiely possible, honestly ... then I more like the idea in #308 because with a "refresh" state that you can trigger you can achieve exactly this behaviour here by just setting both states (one per instance) at the same time Thats why I close this one

closed 2022-04-17 by Apollon77

Loenne83, 2022-04-19:
since I'm not deep into the adapter structure it was just an idea. I agree with your opinion, but what about the cron option? On the other hand, this would indirectly also be possible with the solution in #308, correct?

Apollon77, 2022-04-19:
yes, also possible - controlled by a script of you