Apollon77/ioBroker.smartmeter #323: [Feature Request]  EEBus Support

stilicrafter, 2022-06-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I am frustrated that EEBus is not supported

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Integrate EEBus 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Seartched for standalone EEBus, but didnt find any plugin

**Additional context**
Add any other context or screenshots about the feature request here.
It looks like EEBus and EBus are different systems, but EEBus should be similar to ModBus TCP

Apollon77, 2022-06-23:
is eebus data OBIS based?

Apollon77, 2022-06-24:
I checked the specs and I think an own adapter makes more sense, also because no OBIS is used according to the specs

closed 2022-06-24 by Apollon77

stilicrafter, 2022-06-24:
No Problem.
I opened the request to add it as a new Adapter.

Thank you