Apollon77/ioBroker.smartmeter #337: Missing/wrong data from eZB DDS3

double-a, 2022-10-12:
Just plugged in a Hitchi USB logger I connected to my smartmeter and successfully receive data on the raspberry.

Values for 1-0:1.8.0*255 are shown as kWh whereas the hardware documentation specifies accuracy of micro-Wh.
Values for power e.g. 1-0:16.7.0*255 and 1-0:36.7.0*255 are missing.
Values for voltage e.g. 1-0:32.7.0*255 are missing.

Any way to debug/improve that?

HW documentation can be found [here](https://www.celle-uelzennetz.de/fileadmin/data/netze_documents/1_Kunde/Messwesen/Betriebsanleitung_eBZ_DD3_moderne_Messeinrichtung.pdf), page 10.

Apollon77, 2022-10-13:
When it is missing then the  parts are "out of OBIS specificiations", meaning that the meter manufactorer defined own stuff. And because this is most likely not standardized I think adding it to the official list is not the best option. So if you know what it is, set the object names yourself for now ... no idea if I want to add "unofficial types" because any other smartmeter manufactorer can use them for anthing else :-(

double-a, 2022-11-16:
Thanks. I was able to dig into the settings of the device and switch it to provide full output and now the values are available.

closed 2022-11-16 by double-a