Apollon77/ioBroker.smartmeter #350: Supported smartmeter / Information about data format as retrieved from http url

mar1ged, 2022-12-04:
First things first: you might want to add the Itron Openway 3.HZ-AC-D1-A1 to the list of supported smartmeters. My provider "Netze BW" of EnBW installed it and as far as I can tell it works just fine (with an Hichi USB reader):

[image]

Now to the longer story. I installed ioBroker and the smartmeter adapter on an old Raspberry Pi 1. I decided to do this because I wanted to recycle the old hardware, and learn something (especially getting to run NodeJS 18 running on the device was "interesting")
But one can see that the hardware is a little bit "overwhelmed" and especially after a restart of the device it takes a lot of time until everything is up and running.

So I though about installing something lightweight on the Raspberry Pi 1 that reads from the smartmeter, have this expose the data via http (or tcp ?) and install the smartmeter adapter on my main iobroker server.
But I haven't found information which http or tcp format is expected by the adapter. Can you give me a hint ?

Apollon77, 2022-12-04:
HTTP only one format is supported - the format of the EFR Smart Grid Hub (JSON format)

TCP is "smae formats as Serial" (so D0 or SML), but via TCP without baudrate switchover and such stuff

mar1ged, 2022-12-05:
Thank you @Apollon77 . Does this mean I could just grab the output of dev ttyUsb and export it over a socket without the need of any conversions ?

Apollon77, 2022-12-05:
Honestly ... this all highly depends on your smartmeter and siuch ... best use iobroker forum and discuss options with users there.

mar1ged, 2022-12-05:
OK, I will do so. Or just give it a try.

closed 2022-12-05 by mar1ged