Apollon77/ioBroker.smartmeter #353: DWS7612.1T and SML 1.05 -  parse Error

kct-networx, 2022-12-16:
Hi folks,
my new Smartmeter is a DGZ DWS7612.1T (profile 1.8.0). It's [Manual](https://www.dzg.de/fileadmin/dzg/content/downloads/produkte-zaehler/dvs76/DZG_DVS76_Handbuch_210118.pdf) tells on page 8,  it is using 9600-8-N-1 and SML 1.05.
When I connect this device to smartmeter, i end up with this error:
`Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!`

I am not sure if this has to to do with SML 1.05 of the mMe, because smartmeter allows only to select 1.03/1.04 for SML?!?
What do you think?

I will send the logs via email.

Best regards
Markus

Apollon77, 2022-12-16:
Need to check - also question is if the error is always at the same position and same error type .. if no then ,kost likely the IR head is not coprrectly adjusted or signal not good enough...

kct-networx, 2022-12-16:
the error occurs always. The position is OK. I can get a good output using `cat /dev/ttyusb0 | od - tx1`
It comes every few seconds and is about the same length.
I would say that is stable.
It is an original Hichi IR-reading unit.

If it helps, I can provide much longer log output to compare.
I don't wanted to send you megabytes of logs, while the content is always the same... 😉

Cheers
Markus

Apollon77, 2022-12-16:
In fact the problem is that there is a too much content in two places in the data stream.

1b1b1b1b0101010176053ef31f00620062007263010176010102310b0a01445a470003b41ffe726201640aa698620263a6b80076053ff31f00620062007263070177010b0a01445a470003b41ffe070100620affffffff726201640aa6987377070100603201010172620162006200520004445a470177070100600100ff**ff**017262016200620052000b0a01445a470003b41ffe0177070100010800ff**ff**641c01047262016200621e52036266010101635bf600760540f31f006200620072630201710163ac82001b1b1b1b1a002ed3

The field before defined a length of 7 characters fo the ID octet string

* 07 (length) ... 0100600100ff (Object Name)
* ff ... this is simply too much and WRONG
* 01 (optional field Status)
* ...

If I remove the two "too much data" teh it can be parsed.

In my eyes this is a firmware but of the meter because it simply delivery invalid data.

I do not think that it is becaus eog SML 1.0.5 because if they would have added something like this it would be a 2.0 because breaking change. Additionally I can not find any place where I can find the SML 1.0.5 specs and differences.

DGZ is known to have Firmware bugs in their SML :-(

kct-networx, 2022-12-16:
Thank you very much for your detailed explanation.
Very helpful! 

Seems like DGZ engineers are unable to do their job.... and I got stucked with this unusable SML.

I'm not a developer, but if I understand you correctly, there is no way to shift around this faulty behavior via software?

Thank you for your research and findings.
Best regards
Markus

Apollon77, 2022-12-16:
"no way" is a bit hard, but in fact I need to think how ... because in fact ff01 would also be a value value (but meaning that 241 byte of data are coming next and unexpected for that case.

I would in any case contact them and request a fix or replacement if possible (if you purchased that)

I need to think about how to do that ... I also contacted an other dev to look into it that also had a contact at that company ... But yes nothing "directly solvable". Will need some time

kct-networx, 2022-12-28:
Sorry for my late response. Christmas time came really unexpected (as always).... 😉

I did not bought this SML.
My energy network provider decided to replace the old Ferraris meter with that one.
I've contacted my provider, but I'm not sure if and how they react.

For now I switched over to Tasmota with SML feature, because it is able to handle the DWS7612.
I guess Tasmota is not really parsing the whole SML and only tries to fish for valid data.

But for me and for now it is the only workaround with this smartmeter.

If I get a response from my provider, I'll let you know.

Thank you for support!

Best regards
Markus

closed 2023-09-17 by stale[bot]