Apollon77/ioBroker.smartmeter #369: Zahlensprünge in der 3 oder 4 Nachkommastelle

micklafisch, 2023-03-20:
Hallo,
ich lesen einen Holley DTS541 Zähler mittels Optokopf (EHZ001K) und einem USR W600 Wandler aus.

Bei den Zählerwerten 1.8.0 sowie 2.8.0 entstehen Zahlensprünge in der 3. bzw. 4. Nachkommastelle auf. Es ist pro Abfrage unterschiedlich, ich kann über den Historie Adapter sehen, dass es wie ein Rundungsfehler aussieht (Adapter Version 3.3.4). 

Kommt das vom Adapter oder gibt der Zähler ggf. falsche RAW-Werte aus? Könnte man im Adapter ein Limit für die Nachkommastellen einsetzen? 4 Nachkommastellen ist schon sehr sehr detailliert beim Stromzähler.

[image]

Apollon77, 2023-03-20:
Ok dann bitte erstmal Details: Ist das D0 oder SML als Protokoll? Wenn D0 - wie sieht der rawValue Eintrag aus? AM besten genrrell mal Debug log aktivierne und hier als File anhängen, dan mal das vergelcihern kan ( am beste mit dem Auszug  vom logging das man weiss wann Zeitlich was "komisch" war.

> 4 Nachkommastellen ist schon sehr sehr detailliert beim Stromzähler.

Naja mein Wähmemengenzähler hat MWh als Basiseinheit, da sind 4 Nachkommastellen nicht zuviel :-)

micklafisch, 2023-03-21:
Guten Morgen,

es geht bei meinem Zähler um das SML Protokoll. Ein Auszug aus dem Debuglog wann es auftritt sowie einem Auszug der History Adapter sind anbei. Hoffe das hilft bei der Ursachenforschung.

Ja, sehe ich ein. bei MWh sind 4 Stellen sinnig =)

[image]
[iobroker log.txt](https://github.com/Apollon77/ioBroker.smartmeter/files/11026034/iobroker.log.txt)

closed 2023-09-17 by stale[bot]