Apollon77/ioBroker.smartmeter #373: Incoming data is not parsed correctly

Maintraxx, 2023-05-02:
**Describe the bug**  
I tried to read from an EMH LZQJ-XC. I'm using 300 Baud 7E1. When i try to read out the data with putty, it works successfully. But the smartmeter adapter failes to parse the data:

2023-05-02 11:36:08.117  - [34mdebug[39m: smartmeter.0 (1125) MESSAGE TIMEOUT TRIGGERED
2023-05-02 11:36:08.120  - [34mdebug[39m: smartmeter.0 (1125) Error: No or too long answer from Serial Device after last request.
2023-05-02 11:36:08.123  - [33mwarn[39m: smartmeter.0 (1125) No or too long answer from Serial Device after last request.
2023-05-02 11:36:08.124  - [34mdebug[39m: smartmeter.0 (1125) Error: No or too long answer from Serial Device after last request.
2023-05-02 11:36:08.125  - [34mdebug[39m: smartmeter.0 (1125) Transport Reset!! Restart = true
2023-05-02 11:36:08.126  - [34mdebug[39m: smartmeter.0 (1125) STOP
2023-05-02 11:36:08.127  - [34mdebug[39m: smartmeter.0 (1125) CLEAR MESSAGE TIMEOUT TIMER
2023-05-02 11:36:08.132  - [34mdebug[39m: smartmeter.0 (1125) SERIALPORT CLOSE
2023-05-02 11:36:08.133  - [34mdebug[39m: smartmeter.0 (1125) SERIALPORT REMOVE LISTENERS
2023-05-02 11:36:08.136  - [34mdebug[39m: smartmeter.0 (1125) CURRENT SIGNON MESSAGE COUNTER 1 OF 1
2023-05-02 11:36:08.137  - [34mdebug[39m: smartmeter.0 (1125) DELETE rest of data (0) because new Message start with a new Request Message
2023-05-02 11:36:08.138  - [34mdebug[39m: smartmeter.0 (1125) SCHEDULE NEXT RUN IN 60000s
2023-05-02 11:37:08.141  - [34mdebug[39m: smartmeter.0 (1125) CREATE SERIALPORT: 300 7 1 even
2023-05-02 11:37:08.145  - [34mdebug[39m: smartmeter.0 (1125) SERIALPORT OPEN
2023-05-02 11:37:08.148  - [34mdebug[39m: smartmeter.0 (1125) SERIALPORT RESET BAUDRATE TO 300
2023-05-02 11:37:08.151  - [34mdebug[39m: smartmeter.0 (1125) INITIAL MESSAGES TO SEND: 2
2023-05-02 11:37:08.152  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 0 IN GETNEXTMESSAGE
2023-05-02 11:37:08.155  - [34mdebug[39m: smartmeter.0 (1125) TO SEND 2: 
2023-05-02 11:37:08.155  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 1 IN GETNEXTMESSAGE
2023-05-02 11:37:08.156  - [34mdebug[39m: smartmeter.0 (1125) TO SEND 1: /4!

2023-05-02 11:37:08.330  - [34mdebug[39m: smartmeter.0 (1125) DONE SEND 1
2023-05-02 11:37:08.582  - [34mdebug[39m: smartmeter.0 (1125) DONE SEND 0
2023-05-02 11:37:08.584  - [34mdebug[39m: smartmeter.0 (1125) SET MESSAGE TIMEOUT TIMER2: 60000
2023-05-02 11:37:09.696  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:09.699  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 2 IN CHECKMESSAGE
2023-05-02 11:37:09.700  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT SIGNON: "/EMH4\\@01LZQJL00" -> null
2023-05-02 11:37:10.228  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:10.230  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 2 IN CHECKMESSAGE
2023-05-02 11:37:10.231  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT SIGNON: "/EMH4\\@01LZQJL0013A\r\n\u00020.0.0(1141" -> ["/EMH4\\@01LZQJL0013A\r\n","/EMH4\\@01LZQJL0013A"]
2023-05-02 11:37:10.232  - [34mdebug[39m: smartmeter.0 (1125) PAUSE READING SERIALPORT TO HANDLE MESSAGE
2023-05-02 11:37:10.233  - [34mdebug[39m: smartmeter.0 (1125) CLEAR MESSAGE TIMEOUT TIMER
2023-05-02 11:37:10.235  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 2 IN HANDLEMESSAGE
2023-05-02 11:37:10.237  - [34mdebug[39m: smartmeter.0 (1125) D0 Protocol E (@) not fully supported! Let us try it :-) Please contact the developer if you have problems
2023-05-02 11:37:10.238  - [34mdebug[39m: smartmeter.0 (1125) LEFT AFTER HANDLE-MESSAGE 11
2023-05-02 11:37:10.239  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 3 IN ISPROCESSCOMPLETE: false
2023-05-02 11:37:10.240  - [34mdebug[39m: smartmeter.0 (1125) ALL MESSAGES TO SEND: 2
2023-05-02 11:37:10.493  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 3 IN GETNEXTMESSAGE
2023-05-02 11:37:10.496  - [34mdebug[39m: smartmeter.0 (1125) TO SEND 2: 040

2023-05-02 11:37:10.703  - [34mdebug[39m: smartmeter.0 (1125) DONE SEND 2
2023-05-02 11:37:10.954  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN GETNEXTMESSAGE
2023-05-02 11:37:10.955  - [34mdebug[39m: smartmeter.0 (1125) BAUD CHANGEOVER TO 4800 baud
2023-05-02 11:37:10.959  - [34mdebug[39m: smartmeter.0 (1125) BAUD CHANGEOVER DONE
2023-05-02 11:37:10.960  - [34mdebug[39m: smartmeter.0 (1125) TO SEND 1: 
2023-05-02 11:37:10.960  - [34mdebug[39m: smartmeter.0 (1125) DONE SEND 0
2023-05-02 11:37:10.962  - [34mdebug[39m: smartmeter.0 (1125) RESUME READING SERIALPORT IN FINALIZE
2023-05-02 11:37:10.962  - [34mdebug[39m: smartmeter.0 (1125) SET MESSAGE TIMEOUT TIMER: 60000
2023-05-02 11:37:10.963  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN ISPROCESSCOMPLETE: false
2023-05-02 11:37:10.964  - [34mdebug[39m: smartmeter.0 (1125) REMAINING DATA AFTER MESSAGE HANDLING: 0.0.0(1141
2023-05-02 11:37:10.965  - [34mdebug[39m: smartmeter.0 (1125) END FINALIZE
2023-05-02 11:37:10.966  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:10.967  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:10.968  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{14" -> null
2023-05-02 11:37:10.978  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:10.980  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:10.982  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000" -> null
2023-05-02 11:37:10.989  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:10.990  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:10.995  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.005  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.006  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:11.007  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.021  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.023  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:11.023  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.038  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
[... 10 lines cut ...]
2023-05-02 11:37:11.096  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.111  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.112  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:11.112  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.128  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.129  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:11.129  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.138  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.139  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:11.139  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.161  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.162  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
[... cut ...] 0\u0000\u0000\u0000\u0000\u0000\u0000\u0000" -> null
2023-05-02 11:37:11.228  - [34mdebug[39m: smartmeter.0 (1125) NEW DATA
2023-05-02 11:37:11.229  - [34mdebug[39m: smartmeter.0 (1125) CURRENT PROCESS STEP 4 IN CHECKMESSAGE
2023-05-02 11:37:11.229  - [34mdebug[39m: smartmeter.0 (1125) MATCH-RESULT DATA: "\u00020.0.0(11415114)\r\n0???m{1415114)\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000" -> null

Apollon77, 2023-05-02:
Honestly ... the data look very strange. There are invalid characters in the data, so for now I assume that the IR head it not fit correctly or ur have data transfer issues ... try to check that the cable goes away very fast from the meter. Additionally try an active USB hub, check the grounding and such. Maybe better check in Forum for support.

Apollon77, 2023-05-02:
It does baudrate change eigther because you configured it that way or because the protocol data require it. Because it seems you already played on the settings I have no idea what is the reason.
Bit good if it works for you

closed 2023-05-02 by Apollon77