Apollon77/ioBroker.smartmeter #375: Unterscheidung Einspeisung/Bezug EMH eHZ-H

chemieka, 2023-05-27:
Bei einem EMH eHZ-H wird beim Momentan Wert kein Unterschied gemacht ob Importiert oder Exportiert wird da der Zähler das Vorzeichen nicht ändert. 
Ich kann leider nicht die Binären Daten sehen aber bei einem Export sollte ein A2 am Ende stehen. Beschrieben wird das hier:
https://lists.volkszaehler.org/pipermail/volkszaehler-dev/2017-April/005591.html
Könnte das auch in Smartmeter umgesetzt werden?
[image]

Apollon77, 2023-05-27:
Bitte Debug log anhängen das ich Daten sehe. ggf ist es Duplikat von #306 ?? Bitte da mal reinschauen

chemieka, 2023-05-27:
2023-05-27 09:37:22.103 - debug: smartmeter.0 (26727) STORE DATA
2023-05-27 09:37:22.106 - debug: smartmeter.0 (26727) 129-129:199.130.3*255: Manufacturer ID = EMH
2023-05-27 09:37:22.107 - debug: smartmeter.0 (26727) Create Channel 129-129:199_130_3__255 with name Manufacturer ID
2023-05-27 09:37:22.108 - debug: smartmeter.0 (26727) SET MESSAGE TIMEOUT TIMER: 120000
2023-05-27 09:37:22.109 - debug: smartmeter.0 (26727) RESUME READING SERIALPORT
2023-05-27 09:37:22.111 - debug: smartmeter.0 (26727) SCHEDULE NEXT RUN IN 120s
2023-05-27 09:37:22.118 - debug: smartmeter.0 (26727) SERIALPORT CLOSE
2023-05-27 09:37:22.128 - debug: smartmeter.0 (26727) Create State 129-129:199_130_3__255.value
2023-05-27 09:37:22.132 - debug: smartmeter.0 (26727) Set State 129-129:199_130_3__255.value = EMH
2023-05-27 09:37:22.136 - debug: smartmeter.0 (26727) connected set to true
2023-05-27 09:37:22.181 - debug: smartmeter.0 (26727) 1-0:0.0.9*255: Device ID = 080534d02d6f1a49
2023-05-27 09:37:22.181 - debug: smartmeter.0 (26727) Create Channel 1-0:0_0_9__255 with name Device ID
2023-05-27 09:37:22.184 - debug: smartmeter.0 (26727) Create State 1-0:0_0_9__255.value
2023-05-27 09:37:22.187 - debug: smartmeter.0 (26727) Set State 1-0:0_0_9__255.value = 080534d02d6f1a49
2023-05-27 09:37:22.193 - debug: smartmeter.0 (26727) 1-0:1.8.0*255: Zählerstand 1 Summe Wirkarbeit Bezug + (Total) = 43144.6847 kWh
2023-05-27 09:37:22.194 - debug: smartmeter.0 (26727) Create Channel 1-0:1_8_0__255 with name Zählerstand 1 Summe Wirkarbeit Bezug + (Total)
2023-05-27 09:37:22.197 - debug: smartmeter.0 (26727) Create State 1-0:1_8_0__255.value
2023-05-27 09:37:22.199 - debug: smartmeter.0 (26727) Set State 1-0:1_8_0__255.value = 43144.6847
2023-05-27 09:37:22.205 - debug: smartmeter.0 (26727) 1-0:1.8.1*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T1) = 43144.6847 kWh
2023-05-27 09:37:22.206 - debug: smartmeter.0 (26727) Create Channel 1-0:1_8_1__255 with name Zählerstand 1 Summe Wirkarbeit Bezug + (T1)
2023-05-27 09:37:22.209 - debug: smartmeter.0 (26727) Create State 1-0:1_8_1__255.value
2023-05-27 09:37:22.211 - debug: smartmeter.0 (26727) Set State 1-0:1_8_1__255.value = 43144.6847
2023-05-27 09:37:22.217 - debug: smartmeter.0 (26727) 1-0:1.8.2*255: Zählerstand 1 Summe Wirkarbeit Bezug + (T2) = 0 kWh
2023-05-27 09:37:22.218 - debug: smartmeter.0 (26727) Create Channel 1-0:1_8_2__255 with name Zählerstand 1 Summe Wirkarbeit Bezug + (T2)
2023-05-27 09:37:22.220 - debug: smartmeter.0 (26727) Create State 1-0:1_8_2__255.value
2023-05-27 09:37:22.223 - debug: smartmeter.0 (26727) Set State 1-0:1_8_2__255.value = 0
2023-05-27 09:37:22.273 - debug: smartmeter.0 (26727) 1-0:15.7.0*255: Momentanwert (Total) = 2.6 W
2023-05-27 09:37:22.274 - debug: smartmeter.0 (26727) Create Channel 1-0:15_7_0__255 with name Momentanwert (Total)
2023-05-27 09:37:22.277 - debug: smartmeter.0 (26727) Create State 1-0:15_7_0__255.value
2023-05-27 09:37:22.279 - debug: smartmeter.0 (26727) Set State 1-0:15_7_0__255.value = 2.6
2023-05-27 09:37:22.286 - debug: smartmeter.0 (26727) 129-129:199.130.5*255: Forename/Public Key = 23f8fa3f2aa692a2324a19d62a5b7979e74d8672a58ee23c0c3176071ee09f35b89f3f7e49f429cd1761bba610302be6
2023-05-27 09:37:22.286 - debug: smartmeter.0 (26727) Create Channel 129-129:199_130_5__255 with name Forename/Public Key
2023-05-27 09:37:22.290 - debug: smartmeter.0 (26727) Create State 129-129:199_130_5__255.value
2023-05-27 09:37:22.293 - debug: smartmeter.0 (26727) Set State 129-129:199_130_5__255.value = 23f8fa3f2aa692a2324a19d62a5b7979e74d8672a58ee23c0c3176071ee09f35b89f3f7e49f429cd1761bba610302be6
2023-05-27 09:37:22.301 - info: smartmeter.0 (26727) Received 7 values, 7 updated

chemieka, 2023-05-27:
Ich kann da nichts verwertbares sehen. Die Daten sehen auch nicht so aus wie die Binären aus dem Link obwohl der gleiche Zähler. #306 scheint ein anderes Problem.

Apollon77, 2023-06-08:
naja ein echt vollständiges debug log inklusive der final geparsten message wäre schon cool... hast ein bissl zu rigoroa ausgeschnitten

closed 2025-04-26 by stale[bot]